zsmartsystems/com.zsmartsystems.zigbee #972: EZSP v8 and protocol additions/deletions

dtodor, 2020-01-22:
It seems that the frame format for EZSP v8 has been already implemented. There are, however, additional changes to the protocol, most notably the deletions from the EzspDecisionId enum. Are these going to be addressed in the near future?

I guess with the change to the enums and structs, these have to be versioned now. Is this correct?

cdjackson, 2020-01-22:
I've not checked to see if there are really any changes here. In the past, these sort of changes were renaming or adding new features, bitmasks or commands, and I've generally added them and left it up to the user to decide what features they use (ie the user is left to manage if the firmware version is consistent with the options/commands they are sending). Generally, this is a necessary approach - the user needs to understand the feature set they support.

However, if you are now saying that there are changes that are not backward compatible, then I agree that we will have to deal with this - ie if a structure has changed format, or if a command has removed (or added) parameters, then we'd have to look at how that is handled. I'm just not sure if that has happened - can you provide an exampled of this?

dtodor, 2020-01-22:
For instance, inside the `ZigBeeDongleEzsp` constructor you are using:
`
stackPolicies.put(EzspPolicyId.EZSP_TRUST_CENTER_POLICY, EzspDecisionId.EZSP_ALLOW_PRECONFIGURED_KEY_JOINS);
`
According to the documentation,`EzspDecisionId.EZSP_ALLOW_PRECONFIGURED_KEY_JOINS` has been removed in v8.

cdjackson, 2020-01-22:
I've not looked at this change - I guess they have completely changed the way this is configured so this is not really a case of changing the API, but a case of understanding how now to configure these functions.

Do you know how this has been replaced?

dtodor, 2020-01-22:
I will look into it in detail and let you know. The coming days are pretty busy, so on Saturday the earliest.

sraillard, 2020-01-24:
Regarding how new this new EZSP version 8 is, would you recommend to stay with the version 6.6.x for the EmberZNet SDK and keep using EZSP version 7?

cdjackson, 2020-01-24:
Normally when EZSP is updated the system is backward compatible - I suspect it's mostly the same here. I've not noticed any problem, but I've not tested extensively with Ember 6.7 and quite clearly this policy has been removed.

My guess is it possibly doesn't matter and the default will work fine, but if you don't have the time to check your system yourself, then I'd suggest to stick with 6.6 for now.

If you do update though, then feedback here is welcome if you find any issues :)

cdjackson, 2020-03-22:
It looks to me like they have replaced these decision IDs with a decision bitmask. However the latest UG100 doesn't describe where this new bitmask is used - I could guess that it might be used in the ```EZSP_TRUST_CENTER_POLICY``` policy as a replacement for the previous parameter in this policy, but the document doesn't currently define this.

I've asked for clarification from SiLabs.

From -:
[image]

Changed to -:
[image]

cdjackson, 2020-03-24:
Silabs clarified with the following code snippet from their application framework -:

```
static EmberStatus setJoinPolicy(EmberJoinDecision decision)
{
  EzspDecisionBitmask policy = EZSP_DECISION_BITMASK_DEFAULT_CONFIGURATION;
  uint8_t textId = 0;

  if (decision == EMBER_USE_PRECONFIGURED_KEY) {
    policy = (EZSP_DECISION_ALLOW_JOINS | EZSP_DECISION_ALLOW_UNSECURED_REJOINS);
    textId = 1;
  } else if (decision == EMBER_SEND_KEY_IN_THE_CLEAR) {
    policy = (EZSP_DECISION_ALLOW_JOINS | EZSP_DECISION_ALLOW_UNSECURED_REJOINS | EZSP_DECISION_SEND_KEY_IN_CLEAR);
    textId = 2;
  }
  return emberAfSetEzspPolicy(EZSP_TRUST_CENTER_POLICY,
                              policy,
                              "Trust Center Policy",
                              ezspJoinPolicyText[textId]);
}
```

Thus these new bitmask values are still set using the ```EZSP_TRUST_CENTER_POLICY```. I will take a look at how to implement this.

PR #1013 (2020-03-30): Ezsp8 protocol updates
This updates the EZSP Decision IDs and adds the ```EzspDecisionBitmask```.

The setting of the join decision now checks the EZSP version and sets the decision ID appropriately.  The methods to set the decision ID have changed to use an integer which allows the bitmask to be supported.  Externally this doesn't impact framework users as the EZSP implementation handles this change.

Closes #972.
closed 2020-03-30 by cdjackson