zsmartsystems/com.zsmartsystems.zigbee #975: The Zigbee Discovery extension stops periodic discovery of a repeater device?

tejas-n, 2020-01-28:
We've been experiencing strange issues where the discovery extension stops doing the periodic discovery of Zigbee repeaters(experienced it on both Netvox and Climax based repeaters). Do you think it might be a bug or is it something which we aren't configuring properly?
In the logs below, **00137A0000065FD9** is the repeater and the rest devices are emergency pendants. Thanks.

```
10:48:54.797 I/System.out: 166978 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer  - 00137A0000065FD9: NWK Discovery add node 276F
10:48:54.801 I/System.out: 166982 [NotificationService-thread-15] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: created discoverer
10:48:54.801 I/System.out: 166983 [NotificationService-thread-15] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: start discovery
10:48:54.801 I/System.out: 166983 [NotificationService-thread-15] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: starting new tasks [NODE_DESCRIPTOR, NWK_ADDRESS, ACTIVE_ENDPOINTS, POWER_DESCRIPTOR]
10:48:54.802 I/System.out: 166984 [NotificationService-thread-15] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: scheduled [NODE_DESCRIPTOR, NWK_ADDRESS, ACTIVE_ENDPOINTS, POWER_DESCRIPTOR]
10:48:54.934 I/System.out: 167116 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer  - NWK Discovery for 276F IeeeAddressRequest returned IeeeAddressResponse [276F/0 -> 0000/0, cluster=8001, TID=2B, status=SUCCESS, ieeeAddrRemoteDev=00137A0000065FD9, nwkAddrRemoteDev=10095, startIndex=0, nwkAddrAssocDevList=[18003, 9670]]
10:48:54.934 I/System.out: 167116 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer  - NWK Discovery for 276F ending node discovery. Success=true.
10:48:55.382 I/System.out: 167564 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: running NODE_DESCRIPTOR
10:48:55.426 I/System.out: 167606 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: NodeDescriptorResponse returned NodeDescriptorResponse [276F/0 -> 0000/0, cluster=8002, TID=2C, status=SUCCESS, nwkAddrOfInterest=276F, nodeDescriptor=NodeDescriptor [apsFlags=0, bufferSize=82, complexDescriptorAvailable=false, manufacturerCode=109F, logicalType=ROUTER, serverCapabilities=[], incomingTransferSize=82, outgoingTransferSize=82, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=0]]
10:48:55.427 I/System.out: 167608 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: request NODE_DESCRIPTOR successful. Advancing to NWK_ADDRESS.
10:48:55.427 I/System.out: 167609 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: running NWK_ADDRESS
10:48:57.228 I/System.out: 169409 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: NetworkAddressRequest returned NetworkAddressResponse [276F/0 -> 0000/0, cluster=8000, TID=2D, status=SUCCESS, ieeeAddrRemoteDev=00137A0000065FD9, nwkAddrRemoteDev=10095, startIndex=null, nwkAddrAssocDevList=[]]
10:48:57.228 I/System.out: 169410 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: request NWK_ADDRESS successful. Advancing to ACTIVE_ENDPOINTS.
10:48:57.229 I/System.out: 169410 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: running ACTIVE_ENDPOINTS
10:48:57.262 I/System.out: 169444 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: ActiveEndpointsResponse returned CommandResult [SUCCESS, ActiveEndpointsResponse [276F/0 -> 0000/0, cluster=8005, TID=2E, status=SUCCESS, nwkAddrOfInterest=276F, activeEpList=[1]]]
10:49:01.182 I/System.out: 173363 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: SimpleDescriptorResponse returned SimpleDescriptorResponse [276F/0 -> 0000/0, cluster=8004, TID=2F, status=SUCCESS, nwkAddrOfInterest=276F, length=18, simpleDescriptor=SimpleDescriptor [endpoint=1, profileId=0104, deviceId=0008, deviceVersion=0, inputClusterList=[0000, 0001, 0003, 0015, 0B05], outputClusterList=[]]]
10:49:01.186 I/System.out: 173367 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: request ACTIVE_ENDPOINTS successful. Advancing to POWER_DESCRIPTOR.
10:49:01.187 I/System.out: 173368 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: running POWER_DESCRIPTOR
10:49:01.251 I/System.out: 173433 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: PowerDescriptorResponse returned PowerDescriptorResponse [276F/0 -> 0000/0, cluster=8003, TID=30, status=SUCCESS, nwkAddrOfInterest=276F, powerDescriptor=PowerDescriptor [currentPowerMode=RECEIVER_ON_PERIODICALLY, availablePowerSources=[RECHARGABLE_BATTERY, MAINS], currentPowerSource=RECHARGABLE_BATTERY, powerLevel=MEDIUM]]
10:49:01.252 I/System.out: 173433 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: request POWER_DESCRIPTOR successful. Advancing to null.
10:49:01.253 I/System.out: 173434 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: complete
10:49:54.987 I/System.out: 227167 [NotificationService-thread-11] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer  - 00137A00000154FB: Device announce received. NWK=25C6
10:49:54.991 I/System.out: 227169 [NotificationService-thread-11] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer  - 00137A00000154FB: NWK Discovery add node 25C6
10:49:54.997 I/System.out: 227179 [NetworkManager-thread-5] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 0022A30000162B72: Node SVC Discovery: Update mesh
10:49:54.997 I/System.out: 227179 [NetworkManager-thread-5] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 0022A30000162B72: Node SVC Discovery: starting new tasks [ROUTES, NEIGHBORS]
10:49:54.998 I/System.out: 227180 [NetworkManager-thread-5] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 0022A30000162B72: Node SVC Discovery: scheduled [ROUTES, NEIGHBORS]
10:49:54.999 I/System.out: 227180 [NetworkManager-thread-5] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A000001D7BE: Node SVC Discovery: Update mesh not performed for possible end device
10:49:54.999 I/System.out: 227181 [NetworkManager-thread-5] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: Update mesh
10:49:55.000 I/System.out: 227182 [NetworkManager-thread-5] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: starting new tasks [ROUTES, NEIGHBORS]
10:49:55.000 I/System.out: 227182 [NetworkManager-thread-5] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: scheduled [ROUTES, NEIGHBORS]
10:49:55.006 I/System.out: 227188 [NetworkManager-thread-5] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A00000154FB: Node SVC Discovery: Update mesh not performed for possible end device
[... cut ...]
10:50:09.934 I/System.out: 242113 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: request NEIGHBORS successful. Advancing to null.
10:50:09.934 I/System.out: 242116 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: complete
10:50:09.937 I/System.out: 242119 [NotificationService-thread-11] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A0000065FD9: Node SVC Discovery: stopped
```

tejas-n, 2020-01-28:
We are using v1.3.0 but we've experienced this on v1.2.5 too.

tejas-n, 2020-01-28:
Another log file if it's helpful. The **00124B0007526E17** device is the repeater in this one.
```
11:30:06.578 I/System.out: 2638760 [NotificationService-thread-157] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer  - 00124B0007526E17: Device announce received. NWK=D88F
11:30:06.578 I/System.out: 2638760 [NotificationService-thread-157] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer  - 00124B0007526E17: NWK Discovery add node D88F
11:30:06.580 I/System.out: 2638762 [NotificationService-thread-150] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: created discoverer
11:30:06.581 I/System.out: 2638763 [NotificationService-thread-150] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: start discovery
11:30:06.581 I/System.out: 2638763 [NotificationService-thread-150] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: starting new tasks [NODE_DESCRIPTOR, NWK_ADDRESS, ACTIVE_ENDPOINTS, POWER_DESCRIPTOR]
11:30:06.582 I/System.out: 2638764 [NotificationService-thread-150] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: scheduled [NODE_DESCRIPTOR, NWK_ADDRESS, ACTIVE_ENDPOINTS, POWER_DESCRIPTOR]
11:30:06.589 I/System.out: 2638771 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 0022A30000162B72: Node SVC Discovery: Update mesh
11:30:06.590 I/System.out: 2638772 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 0022A30000162B72: Node SVC Discovery: starting new tasks [ROUTES, NEIGHBORS]
11:30:06.590 I/System.out: 2638772 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 0022A30000162B72: Node SVC Discovery: scheduled [ROUTES, NEIGHBORS]
11:30:06.591 I/System.out: 2638773 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A000001D7BE: Node SVC Discovery: Update mesh not performed for possible end device
11:30:06.591 I/System.out: 2638773 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: Update mesh not performed for possible end device
11:30:06.592 I/System.out: 2638774 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00137A00000154FB: Node SVC Discovery: Update mesh not performed for possible end device
11:30:07.069 I/System.out: 2639250 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: running NODE_DESCRIPTOR
11:30:07.118 I/System.out: 2639300 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 0022A30000162B72: Node SVC Discovery: running ROUTES
11:30:07.154 I/System.out: 2639334 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 0022A30000162B72: Node SVC Discovery: ManagementRoutingRequest returned CommandResult [SUCCESS, ManagementRoutingResponse [0000/0 -> 0000/0, cluster=8032, TID=90, status=SUCCESS, routingTableEntries=16, startIndex=0, routingTableList=[RoutingTable [destinationAddress=0, status=DISCOVERY_UNDERWAY, memoryConstrained=false, manyToOne=true, routeRecordRequired=true, nextHopAddress=10095], RoutingTable [destinationAddress=55439, status=DISCOVERY_UNDERWAY, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=10095], RoutingTable [destinationAddress=18003, status=ACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=10095], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=INACTIVE, memoryConstrained=false, manyToOne=false, routeRecordRequired=false, nextHopAddress=0], RoutingTable [destinationAddress=0, status=I [... cut ...]
11:44:41.213 I/System.out: 3513395 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: request NEIGHBORS failed after 13 attempts.
11:44:41.214 I/System.out: 3513395 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: complete
11:44:41.216 I/System.out: 3513398 [NotificationService-thread-195] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B0007526E17: Node SVC Discovery: stopped
```

tejas-n, 2020-02-19:
Any thoughts on this?

cdjackson, 2020-02-19:
Sorry - I have not had time to have a look at this. If you could debug it yourself I could possibly help with comments, or I could try and fit this in commercially if you wanted me to help, although I am quite busy.

tejas-n, 2020-02-19:
No worries, I'll have a look again and see if can find anything more.

tejas-n, 2020-02-21:
@cdjackson I think when the ZigBeeNodeServiceDiscoverer creates the `updatedNode`, it doesn't copy across the state of the node and so the `updatedNode` is created with state `UNKNOWN`.
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/ef8abce88cb3328052d44a987b29aa8e6a342e3f/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeNodeServiceDiscoverer.java#L232-L239
And when the discovery is finished and `networkManager.updateNode(updatedNode)` is called - which would notify all the listeners - this part of the `ZigBeeDiscoveryExtension` gets executed:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/ef8abce88cb3328052d44a987b29aa8e6a342e3f/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeDiscoveryExtension.java#L201-L215
And since the `updatedNode` was created with `UNKOWN` state, it stops the node's discovery.
I've committed changes to my fork https://github.com/tejas-n/com.zsmartsystems.zigbee/commit/f4177aabb4ab35db471d1e63840e908a8a7472c2 which seems to be working locally. What are your thoughts on this?

tejas-n, 2020-03-04:
@cdjackson After going through the codebase again, I think now have a slightly better understanding of what you were saying in your https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/992#pullrequestreview-366413533 comment

Do you think the following changes might be a better option?
1. We change this:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/4382465836fdaf2b814b82e98c32079e3eb8e634/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNode.java#L729-L734
to

    ```
    if (node.getNodeState() != null && (nodeState != node.getNodeState())) {
            nodeState = node.getNodeState();
            updated = true;
        }
    ```

2. We also remove `UNKNOWN` from enum `ZigBeeNodeState` and let null be the default value of `nodeState` i.e. change this line:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/4382465836fdaf2b814b82e98c32079e3eb8e634/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNode.java#L127
to

    `private ZigBeeNodeState nodeState;`

This will ensure that:
1. If anyone calls `updateNode(ZigBeeNode node)` method with a node whose `nodeState` is null, it wouldn't be copied across just like how it's done for other member variables like nodeDescriptor, powerDescriptor, etc.

2. Since the `nodeState` of the newly created `updatedNode` in `ZigBeeNodeServiceDiscoverer` now would be null and not `UNKNOWN`, calling `updateNode()` won't change the state of the node to `UNKNOWN`.

cdjackson, 2020-03-08:
Hi @tejas-n 

Yes, this is a more correct way to solve this.

However, I'm a little wary of removing ```UNKNOWN``` and using ```null```. I don't really think it's necessary, and it may break something and we'd need to be more careful with possible NPEs. I think you can achieve the same thing using ```UNKNOWN```.

To me, the ```UNKNOWN``` enum is simply a "better" definition of null that doesn't require null pointer checks everywhere to avoid possible NPE.

You can use the following code without removing ```UNKNOWN``` -:

```
if (node.getNodeState() != ZigBeeNodeState.UNKNOWN && nodeState != node.getNodeState()) {
        nodeState = node.getNodeState();
        updated = true;
}
```

What do you think?

tejas-n, 2020-03-08:
@cdjackson Agreed, that feels better. Do you want me to create a new PR and close the old one since it's totally different from what I started?

cdjackson, 2020-03-08:
Up to you - either update the current PR or a new one is fine if it’s easier for you.

Thanks.

PR #997 (2020-03-09): Fix updateNode(ZigBeeNode)
Change `updateNode(ZigBeeNode node)` to not update the state of the node if the state of the node passed to it is `UNKNOWN`

Fixes #975
closed 2020-03-09 by cdjackson