zsmartsystems/com.zsmartsystems.zigbee #977: ZdoCammand ignored in ZigBeeNode

dtodor, 2020-01-29:
Is there a particular reason why ZDO commands are ignored inside the ZigBeeNode:

```
    @Override
    public void commandReceived(ZigBeeCommand command) {
        // This gets called for all received commands
        // Check if it's our address
        if (!(command instanceof ZclCommand) || networkAddress == null
                || command.getSourceAddress().getAddress() != networkAddress) {
            return;
        }
...
    }
```

This could be helpful for processing delayed bind responses from sleepy end devices. Or is there a better way of accomplishing this? I guess it would be possible to register a command listener on the network manager but then the whole filtering and notifications have to be done all over again.

cdjackson, 2020-01-29:
Currently there is no need to process zdo messages here. One day I might change this to centrally process zdo messages but that’s not done now. 

If you are worried about processing a response, then you have a bigger problem since presumably the response is in response to a transaction. This transaction must have timed out or the transaction would handle the response. If the transaction has timed out then you probably should resend it.

dtodor, 2020-01-30:
There are plenty of the devices, which do not regularly poll their parents. Even if we send the bind requests upon receiving messages from the them, it often happens that no responses are sent. What we can see, however, is that such devices might answer queries when they come back online. Unfortunately those ZDO messages will be ignored since the transactions have expired.

Of course we retry the ZDO requests on a regular basis, but at this point it feels more like luck than skill :-) BTW, the same applies to the ZCL reporting configurations, but there we can register listeners on the respective clusters.

cdjackson, 2020-01-30:
Typically devices will not respond when they wake since this requires the parent to buffer messages (as per the spec) and the parent is only required to hold one message for 7.6 seconds. The transaction timers already account for this and if decided don’t respect the requirement to poll their parent within the required time then frames will simply be dropped. 

Maybe these devices are supporting poll control which provides application level knowledge about polling rates rather than relying on the Mac. Poll control is not implemented at this time but would be welcome if you wanted to implement it.

dtodor, 2020-01-31:
What we can see is that messages are received after the transactions have expired. I think it is all known that unfortunately there are plenty of devices, which do not conform to the ZigBee specification. The change would allow us to work around some of those quirks.

cdjackson, 2020-01-31:
I don’t really see it helping as this should not really occur.. As already mentioned, parents will not normally hold messages long enough for this to matter. I’m not sure what “messages” you are talking about either - it would really help if you could provide information about exactly what you are seeing (ie logs).

However, you also say “this change” would help - at the moment I’ve actually no idea what change you are proposing (actually, you are not proposing anything). Please feel free to propose something if you think it is useful and I can review or propose an alternative.

closed 2020-05-01 by stale[bot]