zsmartsystems/com.zsmartsystems.zigbee #981: Ember dual RF Frontend support

sraillard, 2020-02-05:
The new MGM210P modules from Silabs have two RF frontends. One frontend (chain 1) is connected to the antenna included in the module. The other frontend (chain 2) is exposed throught a pin where we can connect an external antenna having a better gain.

It still not very clear for me on how the RF frontend could be choose, if it's at compilation time using the hardware configurator of simplicity studio or at runtime using some EZSP commands. I have posted the question on the Silabs forum without answer for now:
https://www.silabs.com/community/wireless/zigbee-and-thread/forum.topic.html/mgm210p_rf_port_selection-3clZ

So, regarding this library, my question is more about if there will be an option to change the RF frontend at runtime? Maybe if we can change the value of EZSP_VALUE_ANTENNA_MODE using the library?

cdjackson, 2020-02-08:
I guess it's not possible to answer your question -:

>  if there will be an option to change the RF frontend at runtime?

until the first question is answered -:

> It still not very clear for me on how the RF frontend could be choose, if it's at compilation time using the hardware configurator of simplicity studio or at runtime using some EZSP commands.

Until we know if it's even possible to set at runtime, I can't answer. However, I would say that if it is possible to change this at runtime, then we will almost certainly find a way to do this in the library.

sraillard, 2020-02-08:
I agree, I'll try to get information from Silabs.

sraillard, 2020-02-12:
I have an answer from Silabs that confirm the answer in the forum (see link above).

There is no function now in the EZSP protocol to change the RF Path, it should be choosen at compilation time (ie. changing the value of a constant in a header file).

cdjackson, 2020-02-12:
> I have an answer from Silabs 

Just a small word of warning - the person who answered you does not work for Silabs and does not always provide correct answers despite providing a lot of answers ;)

If you really want to be sure, I would suggest to open a case directly with Silabs.

sraillard, 2020-02-12:
Yes, I have noticed! That why I have opened a ticket and I have got a similar answer from a Silabs employee. In fact, whith these new modules and their dual RF path, there is a special function RAIL_ConfigAntenna to set the RF path (and this function must be called before any TX/RX). The value to select the RF path is hardcoded to 1 in the code for building the NCP.

cdjackson, 2020-02-13:
Ok, so I guess there's no room for providing this feature in the library at this stage so I'll close this. If it changes in future I'm more than happy to look at the addition.

closed 2020-02-13 by cdjackson