zsmartsystems/com.zsmartsystems.zigbee #988: Transaction Manager: Wrong incoming ZCL command matched to outgoing request

hsudbrock, 2020-02-20:
The following situation led to a `ClassCastException` when reading an attribute via `ZclAttribute.readValue`:

1. A 'read attribute' command is sent out to a device, and is assigned transaction ID `20`.
2. Simultaneously, the device sends out a 'report attribute' command (for a different attribute and a different cluster than the read attribute command), and (by coincidence) also assigns it transaction ID `20`.
3. The 'report attribute' command is received, and, due to matching transaction IDs, the transaction manager considers the 'report attribute' command to be the response to the 'read attribute' command. This leads to a `ClassCastException` [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/ef8abce88cb3328052d44a987b29aa8e6a342e3f/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zcl/ZclCluster.java#L482) in `ZclCluster.readAttributeValue`, when trying to cast an instance of `ReportAttributesCommand` to a `ReadAttributesResponse`.
4. The actual response to the read attributes command is received from the device, but is no longer matched to the read attributes command.

I wonder how implementors of ZCL transaction management are expected to deal with such a situation. To me it sounds normal (though rare) that two devices, by coincidence, concurrently send out a ZCL command initiating a transaction with the same transaction ID (like, in the above example, a read attributes command sent from the one device, and an attribute report sent from the other device). Should the transaction manager deal with this situation by not only comparing transaction IDs, but taking more info into account? (E.g., by also matching cluster ID, or by never matching two commands that are always sent out as a request but never as a response.) Or should this be dealt with in another way?

I did not find much on that in the ZCL spec, only that 'The Transaction Sequence Number field can be used by a controlling device, which MAY have issued multiple
commands, so that it can match the incoming responses to the relevant command.', which is rather generic advice...

Chris, do you have any thoughts on this?

For completeness, here's the three log messages showing the three commands with transaction ID 20:

```
2020-02-18T08:44:44.960+0200   515  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: ReadAttributesCommand [Ota Upgrade: 0000/0 -> 8F54/1, cluster=0019, TID=20, identifiers=[2]]
2020-02-18T08:44:45.309+0200   515  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReportAttributesCommand [Thermostat: 8F54/1 -> 0000/1, cluster=0201, TID=20, reports=[AttributeReport [attributeDataType=SIGNED_16_BIT_INTEGER, attributeIdentifier=0, attributeValue=2560]]]
2020-02-18T08:44:45.498+0200   515  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ReadAttributesResponse [Ota Upgrade: 8F54/1 -> 0000/1, cluster=0019, TID=20, records=[ReadAttributeStatusRecord [status=UNSUPPORTED_ATTRIBUTE, attributeIdentifier=2]]]
```
And here's all the `zsmart` log messages from before that happening:
[logs_classcastexception.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/4229519/logs_classcastexception.txt)

cdjackson, 2020-02-20:
It's an interesting one.  I'm not really sure how best to handle this - it should (I think) be extremely rare that two devices happen to send a command to eachother with the same transaction id at the same time, but as you say, there's nothing to prevent this.

Matching on cluster ID might help resolve the issue - in this case, it would solve it, but of course if we extend the above statement that there is nothing to stop two devices sending a command to eachother with the same transaction id at the same time, to there is nothing to stop two devices sending a command to eachother with the same transaction id and the same cluster, then the problem still exists - but we are making it even less likely (although given that most devices probably only use 1, 2 or 3 clusters most of the time, it probably doesn't make a lot of difference.

I don't think there's any down side - I don't think there are any cases where you should send a command to one cluster and the response will come from a different cluster, so adding a check on the cluster should not have an adverse effect on transactions at least.

Do you want to provide a PR for this? 👍 

I'm not sure what else can be done to further reduce or eliminate the issue though.

hsudbrock, 2020-02-20:
I agree, an additional check for the cluster ID should reduce occurrences of this issue. I will attempt a PR (and maybe I get other ideas on how to improve this when doing this, in which case I would add a comment to this ticket here before implementing them).

wsowa, 2020-02-20:
Hi, I got here by accident but the issue seems interesting so I'll throw my $0.02...

If instead of OTA cluster you would be reading attribute from thermostat cluster then the cluster check wouldn't help. Assuming that reading thermostat attribute is equally probable to reading OTA attribute the probability of the collision get reduced a lot. 
The transaction ID on the devices are completely independent and wraps every 256 transactions so the chance for the collision is relatively high (comparing to 128 bit random number like uuid). So designing the system to operate correctly based on so high probability is flawed. 

What I would suggest as a more robust mechanism is to whitelist possible response types to every request And use this mapping when matching. So for ReadAttributeCommand only a ReadAttributeRequest and some error responses (or whatever the ZCL specs) would be matching. This way the collision would happen only if we send >256 times a request before receiving a response. Can't imagine a legitimate scenario for this to happen.

hsudbrock, 2020-02-20:
Thanks for your $0.02, @wsowa !

I agree that a mechanism that actually checks whether an incoming command could be a response to the command that we sent when initiating the transaction would be more robust. (My suggestion above to 'never match two commands that are always sent out as a request but never as a response' also goes into this direction.)

I have no feeling yet how much effort such a more robust solution would be (as we would then need to consider ZCL command semantics in the transaction manager). This is why I think that - as a first step and a first PR - we could already improve the situation by considering the cluster in the transaction manager; then, in a second step, one could think about an approach that considers the command type in the transaction manager.

Does that sound reasonable?

wsowa, 2020-02-20:
Well, I would assume that the logIc for that matching would sit in the abstract command class, mapping config for echa Command In the command class (or the XML for autocode) and the transaction manager would only call the match() method. 

I guess the biggest effort and also opportunity for introducing a bug would be in defining correct mapping. 

My suggestion that myself and Chris were discussing at the end of last year goes in a similar direction. There the idea was to refactor code in a way that the command class would have "response" methods in order to guarantee that we always send a response to the correct transaction, command and cluster. See it here:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/936#issuecomment-572274016

cdjackson, 2020-02-20:
> Well, I would assume that the logIc for that matching would sit in the abstract command class, mapping config for echa Command In the command class (or the XML for autocode) and the transaction manager would only call the match() method.

This is what is currently done for ZDO commands if I remember correctly.

I think that this is quite a lot of effort though and probably right now is not the right direction - simply due to time. I guess we could add the mechanism (similar to ZDO) and start to build up the response whitelist.

However in reality, the chances of this happening is pretty low (I think) - or do you disagree @wsowa?

In the short term I don't think the check on cluster hurts, and it will reduce the likelihood of this occurring, so it seems to still be worth doing I think?

wsowa, 2020-02-20:
Yes, adding extra check for cluster id won't hurt and is pretty low effort to implement. 

I agree the effort to build full matching solution is significant and a big bang approach is rather unrealistic but perhaps we could build the wireing and one command example now, implement all the dummy matchers to return true and then over the time they will transform to actual matchers.

hsudbrock, 2020-04-23:
I still plan to add something as discussed above, so maybe we could leave this issue open until then.

closed 2020-07-06 by stale[bot]