zsmartsystems/com.zsmartsystems.zigbee #996: Commands needed for testing JOIN

ImkereiWetzel, 2020-03-06:
Hi,

as I have problems using the provided cmdline tool to have a light join my network, may I kindly ask which commands are needed to have a light join my network?
Once it is joined, I have no problems reading/writing attributes, switching on/off ... 
Only the initial join is not working for me, and I don't know where to continue learning now.

BR
Marc

ImkereiWetzel, 2020-03-09:
OK, let me re-phrase my request.

If I use "join", my light is detected, but the stack is stuck at waiting for the tasks "NWK_ADDRESS, ACTIVE_ENDPOINTS, POWER_DESCRIPTOR, NODE_DESCRIPTOR" to get completed.

I waited for several 10 minutes, but they never complete.

If I run the same procedure before-hand with zcl (Z3GatewayHost/SimplicityStudio), and I form the network over there and then start the zsmartsystems.zigbee stack, the device is listed correctly.

What am I missing here?

cdjackson, 2020-03-09:
Really it’s hard to comment without visibility of what is going on. You provide no logs, no information on what device you are trying to join or what dongle you are using. 

Possibly the issue is with keys, but that’s pure speculation. This is a development library so I would suggest you try to work out what is happening using a debugger and/or a sniffer since the join isn’t working.

ImkereiWetzel, 2020-03-09:
I am using an Silicon labs Mighty Gecko with MGM13P02.
As I read (on the SiliconLabs Support Forum) you used a similar setup, so I tried the short cut - do I have to configure anything special for this ember module? You know, for the SiliconLabs Tool (Z3GatewayHost) nothing had to get configured, so I was thinking that you already know what might be missing.

cdjackson, 2020-03-09:
I don’t know your network but in general there shouldn’t need to be anything further configured other than network information.

ImkereiWetzel, 2020-03-09:
Thank you so far, I opened a support request at SiLabs.
Can you tell me about the stack and firmware you are using on your ember device and which device you are using for testing?
We are using a MGM13P02 with the latest stack, as for now this is: Gecko V2.7.2/EmberZNet SDK 6.7.2.0. We flashed the untouched ncp-uart-hw example code (running over USB)

TIA
Marc

cdjackson, 2020-03-09:
I've used different versions - from 5.8 up to 6.8. In openHAB people are using a wide variety, and we just performed ZigBee certification with a system using 6.4 and using this library.

Probably Silabs will not be so interested in helping - their normal stock answer is to repeat the issue using their tool. Really you probably need to get a sniffer to understand what is happening.

You also haven't said what the device is that you're trying to join?

ImkereiWetzel, 2020-03-09:
Ok, I will retry my setup with an older stack version just to see if it changes...
I am using one stock Paulmann RGBW light, as you can by them over here.
This device works perfectly with my Hue gateway... and also with the SiLabs CLI tool.

cdjackson, 2020-03-09:
Actually, I just checked, and 6.7 is the latest. It should work (I have tested it), but there were some changes to Silabs security settings and they made a large change to EZSP 8 in this version (which is why I thought it was 6.8).

An older version might be worth a try.

I have tested Paulman RGBW bulbs so it can work, but they are (IFRC) ZB3.0.

ImkereiWetzel, 2020-03-12:
Hi Chris,
i tried everything with openHab - attached you find my openhab.log
The relevant lines seem these here:
2020-03-12 17:35:36.070 [DEBUG] [bee.discovery.ZigBeeDiscoveryService] - Starting ZigBee scan for zigbee:coordinator_ember:b8f77723
2020-03-12 17:35:36.071 [DEBUG] [bee.handler.ZigBeeCoordinatorHandler] - ZigBee coordinator is offline - aborted scan for zigbee:coordinator_ember:b8f77723

ImkereiWetzel, 2020-03-12:
OK, I have problems adding log files, maybe I am not allowed to?

Maybe I added a problem, and I now don't know how I can fix it. I gave the command:
ncpstate offline
I had really trouble to get it online again. With the zss.cli tool it was not possible (or I made something really wrong), only with the SiLabs tool I could get it going again...
But now I can control lamps with the zss.cli tool, but openHAB tells me the coordinator is offline.
BTW: switching to older firmware didn't work out either - same problem here

cdjackson, 2020-03-12:
Sorry- I’ve no idea what you are doing, or what the issue is without some sort of information other than “it doesn’t work”.

If you’re using OH, then I’d really just suggest to use the binding and not worry about the CLI.

ImkereiWetzel, 2020-03-13:
Hi Chris,
may I contact you by email?

BR
Marc

cdjackson, 2020-03-13:
Hi Marc
Sure - if you can’t find an address please let me know, but I think it’s shown on GitHub. 

Chris

sorinpenteleiciuc, 2020-03-17:
Hello Chris Jackson,

 I contact you on behalf of P3-Group and would like to ask you if you can offer us paid support for the library. We would like a fix for this issue as soon as possible:https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/996. My colleague with the account ImkereiWetzel reported this issue. Could you please schedule a short call with my superior? 
My email is "sorin.penteleiciuc@p3-group.com"

Thank you and best regards,
Sorin

cdjackson, 2020-04-04:
@ImkereiWetzel I think this can be closed can't it?

closed 2020-04-15 by cdjackson