zsmartsystems/com.zsmartsystems.zigbee #999: NotificationService and multiple ZigBeeNetworkManagers

hsudbrock, 2020-03-11:
An instance of the `ZigBeeNetworkManager`, in most cases, does not share classes that is uses internally. One exception is the `NotificationService`, which is shared amongst all instances of the `ZigBeeNetworkManager`. 

Currently, this is an issue when using multiple instances of `ZigBeeNetworkManager`: If one instance is shut down via `ZigBeeNetworkManager.shutDown()`, then the executor service of the `NotificationService` is also shut down. In consequence, the other instances won't function correctly anymore, because notifications via the `NotificationService` no longer work.

I think that this is an issue in two cases:
(a) In sytems that handle multiple ZigBee chips in parallel, and
(b) in dynamically updatable systems (like, in my case, an ESH-based system) where a new `ZigBeeNetworkManager` might be already initialized when an old `ZigBeeNetworkManager` is still shutting down. (Unfortunately, ESH does not guarantee that a bridge handler was completely disposed before initializing a new handler for the same bridge.)

I see two possible solutions to improve this:
1) Do not share the `NotificationService` amongst different instances of `ZigBeeNetworkManager`.
2) Properly handle initialization/shutdown in the `NotificationService` even with multiple `ZigBeeNetworkManager`s.

I think that (1) has the advantage that the framework would stop sharing resources between `ZigBeeNetworkManager`s, but the disadvantage that one would have to pass the notification service from the network manager to many objects used by the network manager (transaction manager, clusters, command notifiers, etc.).

I think that (2) has the advantage that it would be a change only in `NotificationService`; my suggestion is to store within the notification service the information which network managers (or alternatively how many network managers) are currently using the notification service, and to shut down the executor service only once all using network managers asked it to shut down.

Do you have an opinion how this could/should be resolved, Chris?

cdjackson, 2020-03-11:
Thanks for pointing this out Henning - I agree - this needs to be fixed. I do however fear it might be a bit complex...

I prefer #1 - but it's not simple to pass this around I think, so it may make this a large change and #2 might therefore be preferred even if it's not as clean.

However, #2 I think has an added problem (working from memory). When a network manager closes, the closing of the ```NotificationService``` releases any threads which in turn releases any waiting threads around the wider system. Maybe it's possible to use separate executor service for each network (although I'm not sure how that is possible) - otherwise you'd potentially need to track what notifications are for what network so they can be released (again, I'm not sure how to do that since the link to the network is not available).

Hence why I kind of think that #1 might be a cleaner solution even if it feels more problematic at first glance...

hsudbrock, 2020-03-11:
This is a good point, Chris. I fear that you are right regarding #2 and that we need #1. But I also think that #1 will not be really complex (just a larger change), and might not make the code nicer because we need to pass the notification service instance into many different objects used by a single network manager.

I will think about it and probably play around with the two approaches to get a better feel.

hsudbrock, 2020-03-12:
I have created a pull request just so that one can see how solution #1 could look like. wdyt, Chris?

PR #1000 (2020-03-23): Use a separate notification service for each network manager
As discussed in #999, one option to resolve #999 is to use a separate non-static `NotificationService` for each instance of the `ZigBeeNetworkManager`.

This pull request illustrates how this could look like. The changes are smaller than I had initially thought. Basically, the `ZigBeeNetworkManager` now creates its own instance of the `NotificationService` and provides a method for retrieving the notification manager. Other objects managed by the network mangager (e.g., the transaction manager, the nodes, endpoints, clusters, some ap [... cut]
closed 2020-03-23 by cdjackson