zsmartsystems/com.zsmartsystems.zigbee #1005: Does the CluserMatcher needs to set the transaction id of the response?

tejas-n, 2020-03-19:
Hi, I've been struggling a bit enrolling an emergency button having an IAS Zone cluster to the coordinator. It was joining the network but the button presses weren't doing anything. I experimented by setting the `transactionId` of the `MatchDescriptorResponse` to be the same as that of `MatchDescriptorRequest`
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/5abf3804bc70f1d08602777a38fa7d9379e282bb/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/internal/ClusterMatcher.java#L157-L166

I added  `matchResponse.setTransactionId(matchRequest.getTransactionId());` to line 162 and the emergency button started working correctly.

Am I correct in assuming that the `transactionId` of the `MatchDescriptorResponse` needs to be the same as the `MatchDescriptorRequest`? Thank you.

cdjackson, 2020-03-19:
Hi Tejas,
Yes, this is correct - this should be set in this way. If you want to provide a PR, that would be great.

Cheers
Chris

tejas-n, 2020-03-19:
Thanks, Chris, I'll submit a PR shortly.

PR #1008 (2020-03-19): Set the transactionId of MatchDescriptorResponse
Use the transaction id of the incoming MatchDescriptorRequest when sending out a MatchDescriptorResponse in ClusterMatcher

Fixes #1005
closed 2020-03-19 by cdjackson