zsmartsystems/com.zsmartsystems.zigbee #1006: TelegesisEnergyScanCommand response prompt is broken in auto coder

triller-telekom, 2020-03-19:
According to the Telegesis AT specification the prompt for an `TelegesisEnergyScanCommand` is `AT+ESCAN`. This is correctly implemented in the auto coder.

But the response to the command has the prompt `+ESCAN`, so the "AT" is missing.

I will create a PR for this.

cdjackson, 2020-03-19:
Are you really sure this is correct? I just checked this, and it's wrong as far as I can see.

The manual shows the following -:

[image]

And when I typed in ```AT+ESCAN``` I get a response of ```+ESCAN``` - as per the manual.

[image]

There is no ```AT``` preceding this.

I should add that this is using the DTAG version of the firmware so it should be in line with what you're using?

triller-telekom, 2020-03-20:
> There is no AT preceding this.

This is exactly what I am talking about, right? :-)

Request HAS the the "AT", but response does NOT have the "AT".

triller-telekom, 2020-03-20:
Here is the output of a successful scan after I had removed the check for the string with "AT" in the response:

```
2020-03-19T16:35:18.850+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis: TelegesisEnergyScanCommand [scanResults=[]]
2020-03-19T16:35:18.850+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - TX Telegesis Data:AT+ESCAN
2020-03-19T16:35:21.500+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:+ESCAN:
2020-03-19T16:35:21.520+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:11:BD
2020-03-19T16:35:21.540+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:12:BE
2020-03-19T16:35:21.560+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:13:B2
2020-03-19T16:35:21.580+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:14:AB
2020-03-19T16:35:21.600+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:15:D0
2020-03-19T16:35:21.620+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:16:C1
2020-03-19T16:35:21.620+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:17:C1
2020-03-19T16:35:21.630+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:18:C7
2020-03-19T16:35:21.650+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:19:C5
2020-03-19T16:35:21.650+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:20:CF
2020-03-19T16:35:21.670+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:21:CA
2020-03-19T16:35:21.690+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:22:B9
2020-03-19T16:35:21.690+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:23:B3
2020-03-19T16:35:21.710+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:24:B0
2020-03-19T16:35:21.730+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:25:C7
2020-03-19T16:35:21.730+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:OK
2020-03-19T16:35:21.750+0100   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisEnergyScanCommand [scanResults=[ScanResult [channel=11, rssi=189], ScanResult [channel=12, rssi=190], ScanResult [channel=13, rssi=178], ScanResult [channel=14, rssi=171], ScanResult [channel=15, rssi=208], ScanResult [channel=16, rssi=193], ScanResult [channel=17, rssi=193], ScanResult [channel=18, rssi=199], ScanResult [channel=19, rssi=197], ScanResult [channel=20, rssi=207], ScanResult [channel=21, rssi=202], ScanResult [channel=22, rssi=185], ScanResult [channel=23, rssi=179], ScanResult [channel=24, rssi=176], ScanResult [channel=25, rssi=199]], status=SUCCESS]
```

cdjackson, 2020-03-20:
Sorry - I misread this. When you said "The ```AT``` is missing" I thought that you were adding it, but you mean the AT needs to be removed.

Ok, so I think then we are in agreement :) sorry for the confusion.

PR #1007 (2020-03-20): Fix TelegesisEnergyScanCommand response prompt in auto coder
Fixes #1006
commit 73355bbc (2020-03-20): Fix TelegesisEnergyScanCommand response prompt in auto coder (#1007)

Fixes #1006
closed 2020-03-20 by cdjackson