zsmartsystems/com.zsmartsystems.zigbee #1019: radioTxPower of EmberNetworkParameters is not deserialized correctly?

tejas-n, 2020-03-30:
I found out that the value that I am getting from `getRadioTxPower()` of `EmberNetworkParameters` is a signed value. For eg, I am getting 251 dBm which I assume should be -5 dBm.

I found out that the `radioTxPower` field is set by deserializing an `unint_8t` value
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/8ed1cc1536ccd89164cb7e6826085a996681b308/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ezsp/structure/EmberNetworkParameters.java#L289-L299

So I had a look at the [EZSP reference guide](https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf) and found out that even though it's defined as `uint8_t` in "Structure Definitions" on page 13, it is defined as `int8_t` in [EmberZNet Stack API Reference](https://docs.silabs.com/zigbee/latest/em35x/structEmberNetworkParameters)

Also, it is defined as `int8s`(which I assume is signed int) in `setRadioPower` frame on page 75 of the EZSP reference guide. Any thoughts on this?

tejas-n, 2020-03-30:
If it does need to be a signed integer, do you think the `serialize()` and `deserialize()` methods of `EmberNetworkParameters` need to have `serializer.serializeInt8S(radioTxPower)` and `radioTxPower = deserializer.deserializeInt8S()` respectively?

cdjackson, 2020-03-30:
Hi Tejas,
There is already a signed version of the serialisation - I know that it's used for RSSI measurement in the incoming message handler so it should just be as simple as changing the data type to ```int8s``` and re-running the autocoder?

tejas-n, 2020-03-30:
@cdjackson Thanks for the reply. I didn't know that those classes were generated by autocoder as well. I'll give it a try.

cdjackson, 2020-03-30:
It looks to me like the definition in setRadioPower is fine.

The XML file (as you say earlier) shows ```int8s``` which is signed.

In the code, this generates -:

```
        serializer.serializeInt8S(power);
```

I just created a test, and I think it is fine. I'll add this as a PR.

So I guess that EmberNetworkParameters is incorrectly defined. This is confirmed in the online Silabs docs -:

[image]

cdjackson, 2020-03-30:
I've created #1021 to update the definition - that should close this issue.

closed 2020-03-30 by cdjackson

tejas-n, 2020-03-30:
Looks good. Thanks, Chris.

PR #1021 (2020-03-30): Fix definition of radioTxPower in EmberNetworkParameters
Fixes #1019