zsmartsystems/com.zsmartsystems.zigbee #1026: Discovery of existing node doesn't work properly

triller-telekom, 2020-04-03:
I am currently facing an edge case, which unfortunately turned out to be done in practice :(

Imagine a user pairs a device , but right after the "DeviceAnnounce" the power of the device is cut off. The result will be that we will create a `ZigbeeNode` with nothing more than the IEEE and maybe network address.

If the user later wants to make use of the device and thus tries to re-pair the device again, he will run into this check:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c9d2152db28c79c45a24d29818b62f627147ebe7/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNetworkManager.java#L1711-L1714

This is because the `ZigbeeNetworkDiscoverer#addNode()` will call `ZigbeeNetworkmanager#updateNode()` and this will detect that the node already exists and thus call `refreshNode`, see above.

Since the `refreshNode` doesn't reach the part where it notifies its listeners, the `ZigBeeDiscoveryExtension#nodeUpdated()` will not be called so the Endpoints etc. won't be requested from the device.

I am also not sure about the semantic of the `sendNodeAdded`. I remember that in the past we had discussions on whether there should be 2x nodedAdded vs 1x nodeAdded and 1x nodeUpdated during pairing and the later was implemented and then reverted to the 2x nodeAdded solution.

So I leave this open for discussion on how to deal with partially paired devices.

triller-telekom, 2020-04-08:
I have investigated this a little further and found out that once the system is in the state that there is a `ZigBeeNode` with just IEEE and network address, there is no way to get the device with that IEEE address evere discovered/paired.

The reson is in the code part I cited above. The `ZigBeeNetworkManager.refreshNode` returns and does not announce the node update to its listeners.

One of the listeners is the `ZigBeeTransactionManager`. This one plays an importan role here because:

On system startup we load the `ZigBeeNode`s from the persistent storage and add it to the `ZigBeeNetworkManager`. This "adding" triggers a discovery scan in the `ZigBeeDiscoveryService`.

```
2020-04-08T14:15:07.813+0200   87 DEBUG [cpool-4: org.openhab.binding.zigbee.ember,com.zsmartsystems.zigbee.database.ZigBeeNetworkDatabaseManager] 00124B000EEA2880: Data store: Node was restored.
2020-04-08T14:15:07.813+0200   87 DEBUG [cpool-4: org.openhab.binding.zigbee.ember,com.zsmartsystems.zigbee.ZigBeeNetworkManager] 00124B000EEA2880: Updating node NWK=C7D4
2020-04-08T14:15:12.653+0200  282 DEBUG [NotificationService-thread-1,com.zsmartsystems.zigbee.ZigBeeNetworkManager] Network state ONLINE: Notifying node 00124B000EEA2880 [C7D4]
2020-04-08T14:15:12.653+0200  282 DEBUG [NotificationService-thread-2,com.zsmartsystems.zigbee.database.ZigBeeNetworkDatabaseManager] 00124B000EEA2880: Data store: Deferring write for 250ms.
2020-04-08T14:15:12.653+0200   55 DEBUG [NotificationService-thread-7,org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService] 00124B000EEA2880: Stefan: nodeAdded (discovered=false) node=ZigBeeNode [state=UNKNOWN, IEEE=00124B000EEA2880, NWK=C7D4, endpoints=[]]
2020-04-08T14:15:12.654+0200   55 DEBUG [NotificationService-thread-7,org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService] 00124B000EEA2880: Stefan: in if nodeAdded node=ZigBeeNode [state=UNKNOWN, IEEE=00124B000EEA2880, NWK=C7D4, endpoints=[]]
2020-04-08T14:15:12.655+0200  282 DEBUG [NotificationService-thread-8,com.zsmartsystems.zigbee.app.discovery.ZigBeeDiscoveryExtension] 00124B000EEA2880: DISCOVERY Extension: Adding discoverer for added node
2020-04-08T14:15:12.655+0200  282 DEBUG [NotificationService-thread-8,com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] 00124B000EEA2880: Node SVC Discovery: created discoverer
2020-04-08T14:15:12.655+0200  282 DEBUG [NotificationService-thread-8,com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] 00124B000EEA2880: Node SVC Discovery: start discovery
2020-04-08T14:15:12.655+0200  282 DEBUG [NotificationService-thread-8,com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] 00124B000EEA2880: Node SVC Discovery: starting new tasks [NWK_ADDRESS, NODE_DESCRIPTOR, POWER_DESCRIPTOR, ACTIVE_ENDPOINTS]
2020-04-08T14:15:12.655+0200  282 DEBUG [NotificationService-thread-8,com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] 00124B000EEA2880: Node SVC Discovery: scheduled [NWK_ADDRESS, NODE_DESCRIPTOR, POWER_DESCRIPTOR, ACTIVE_ENDPOINTS]
```

The tasks of that scan create transactions with the CURRENT network address of the node and trys to send them via the `ZigBeeTransactionManager` where they are placed into queues and then retried multiple times in case of failures.

If the device has been reset and wants to join the network now it sends its `EzspTrustCenterJoinHandler` and `DeviceAnnounce` messages containing its NEW network address, but we will not create a new discovery for it because the `ZigBeeDiscoveryService` is not notified.

However, there might be still one discovery running which has transactions in the queue from the `ZigBeeTransactionManager`. Unfortunately they all have the OLD net work address and thus will never succeed.

Once all transactions have been retried until their limit, they are dropped. Because the `ZigBeeDiscoveryService` isn't notified about the updated node, there is also no new discovery, and thus the node is left undiscovered.

---

If we change the `return null` from above to `sendNodeAdded = false` the situation is resolved. This holds for both cases:

- Case 1: If a discovery with transactions is still running: The tranactions will be rewritten to the new network address in `ZigBeeTransactionManager.nodeUpdated` and then eventually succeed.
- Case 2: If the discovery from startup has finished its failing transactions, a NEW discovery will be started because the ´ZigBeeDiscoveryExtension` is notified about the updated node and currently does not have a discovery running for this IEEE address.

I am just not 100% sure about the consequences yet. There are 7 listeners which will be notified even if the discovery is not yet complete, if we change the code as I suggested above:

```
2020-04-08T14:15:48.707+0200  282 DEBUG [NetworkManager-thread-1,com.zsmartsystems.zigbee.ZigBeeNetworkManager] 00124B000EEA2880: refreshNode. Listener.size()=7
2020-04-08T14:15:48.708+0200  282 DEBUG [NetworkManager-thread-1,com.zsmartsystems.zigbee.ZigBeeNetworkManager] 00124B000EEA2880: refreshNode. Calling listener: com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager@7c9e00b9
2020-04-08T14:15:48.708+0200  282 DEBUG [NetworkManager-thread-1,com.zsmartsystems.zigbee.ZigBeeNetworkManager] 00124B000EEA2880: refreshNode. Calling listener: org.openhab.binding.zigbee.ember.handler.EmberHandler@658e6dac
[... cut ...]
2020-04-08T14:15:48.708+0200  282 DEBUG [NetworkManager-thread-1,com.zsmartsystems.zigbee.ZigBeeNetworkManager] 00124B000EEA2880: refreshNode. Calling listener: org.openhab.binding.zigbee.discovery.ZigBeeDiscoveryService$1@930cb79
2020-04-08T14:15:48.708+0200  282 DEBUG [NetworkManager-thread-1,com.zsmartsystems.zigbee.ZigBeeNetworkManager] 00124B000EEA2880: refreshNode. Calling listener: com.zsmartsystems.zigbee.app.discovery.ZigBeeDiscoveryExtension@7e37f389
```

@cdjackson What is your opinion about this?

triller-telekom, 2020-04-09:
Unfortunately my ideas from the previous comments only fix case 1.

For case 2 this piece of code becomes a problem:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/b1b9dd9d7fbcf9aaa6ecf6e1462ee46904aa75e7/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeDiscoveryExtension.java#L201-L215

The `nodeDiscovery` map will still contained the timed-out discovery from the startup and hence it will NOT start a new discovery.

As far as I understand the code the only place where a discovery is removed from `nodeDiscovery` is in `stopDiscovery` which is only called once a node is removed or changed its state to be not `ONLINE`.

So I think there is some kind of callback mechanism missing between the `ZigBeeDiscoveryExtension` and the `ZigBeeNodeServiceDiscoverer` which are started per device. So whenever a `ZigBeeNodeServiceDiscoverer` has either successfully finished ALL its tasks, no matter if successful or timed out, the `ZigBeeDiscoveryExtension` should know about this to be able to remove it from the `nodeDiscovery` map.

Unfortunately this might become tricky because a `ZigBeeNodeServiceDiscoverer` might have finished SOME of its tasks, but not all of them when there is an incoming `nodeUpdated` which wants to trigger a new discovery.

triller-telekom, 2020-05-07:
After implementing a workaround to ignore nodes that to not have endpoints, powerdescriptor and nodedescriptor, I noticed another problem why I think this issue should be looked at.

The scenario is as follows: There is/was a device with network address 0x226F = 8815 connected with the zigbee controller having the ieee address 000D6F000DBD395B. I write "is/was" because it is unclear because the pairing failed and all that was left was a node in the ZigbeeNetworkManager with the IEEE address and network address set.

When the device now has its power supply cut off and er start the coordinator, the following happens:

```
2020-05-06T13:54:21.921+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: IeeeAddressRequest [0000/0 -> 0000/0, cluster=0001, TID=02, nwkAddrOfInterest=0000, requestType=1, startIndex=0]
2020-05-06T13:54:21.922+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=0000/0, profile=0000, cluster=0001, addressMode=DEVICE, radius=31, apsSecurity=false, ackRequest=true, apsCounter=05, rssi=--, lqi=--, payload=02 00 00 01 00]
2020-05-06T13:54:21.926+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeDiscoveryExtension] - DISCOVERY Extension: Starting mesh update
2020-05-06T13:54:21.940+0200   505  DEBUG          [symbolicName=org.openhab.binding.zigbee.core, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - 000D6F000DBD395B: networkStateUpdated called with state=ONLINE
2020-05-06T13:54:21.955+0200   467  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSendUnicastResponse [status=EMBER_SUCCESS, sequence=0B]
2020-05-06T13:54:21.971+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeDiscoveryExtension] - 000D6F000DBD395B: DISCOVERY Extension: Adding discoverer for added node
2020-05-06T13:54:21.972+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 000D6F000DBD395B: Node SVC Discovery: created discoverer
2020-05-06T13:54:21.972+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 000D6F000DBD395B: Node SVC Discovery: start discovery
2020-05-06T13:54:21.973+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 000D6F000DBD395B: Node SVC Discovery: starting new tasks [NODE_DESCRIPTOR, POWER_DESCRIPTOR]
2020-05-06T13:54:21.973+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 000D6F000DBD395B: Node SVC Discovery: scheduled [NODE_DESCRIPTOR, POWER_DESCRIPTOR]
2020-05-06T13:54:21.990+0200   467  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_BROADCAST_LOOPBACK, apsFrame=EmberApsFrame [profileId=0000, clusterId=0036, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=0A], lastHopLqi=255, lastHopRssi=0, sender=0000, bindingIndex=255, addressIndex=255, messageContents=00 00 01]
2020-05-06T13:54:21.990+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=0000/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=0A, rssi=0, lqi=FF, payload=00 00 01]
2020-05-06T13:54:21.991+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=00, permitDuration=0, tcSignificance=true]
2020-05-06T13:54:22.020+0200   467  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0000, clusterId=0036, sourceEndpoint=0, destinationEndpoint=0, options=[EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=0B], lastHopLqi=255, lastHopRssi=0, sender=0000, bindingIndex=255, addressIndex=255, messageContents=01 00 01]
2020-05-06T13:54:22.020+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=0000/0, destinationAddress=0000/0, profile=0000, cluster=0036, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=0B, rssi=0, lqi=FF, payload=01 00 01]
2020-05-06T13:54:22.021+0200   463  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: ManagementPermitJoiningRequest [0000/0 -> 0000/0, cluster=0036, TID=01, permitDuration=0, tcSignifi [... cut ...]
If you try to pair that device through the normal pairing process, this will fails because 2 discoverers will fight for the replies from the device and mess up their state machines which request descriptors, endpoints, etc.

At the end if all has timed out, there is still that "broken" node in the network manager which cannot be used to discover the rest of the device later on due to the reasons in my comments above.

@cdjackson Could you please have a brief look at this issue and provide your opinion on this?

triller-telekom, 2020-05-28:
I thought about this again and I will create a PR with a suggestion to fix this.

What I did is:

- I changed the `return null` from above to `sendNodeAdded = false` in the `ZigbeeNetworkManager.refreshNode` method so the `ZigBeeDiscoveryExtension` is informed about the update of a node.
- The `ZigBeeNodeServiceDiscoverer` now has a flag `finished` and a list of failed tasks
- Not the `ZigBeeDiscoveryExtension` can check for a node not only if there is a `ZigBeeNodeServiceDiscoverer` but also if it is finished and if so if tasks have failed and then act accordingly.

triller-telekom, 2020-06-02:
@cdjackson Hey Chris, just a short question: Since I haven't got a reply from you on this topic I just wanted to ask whether you are just too busy with other things (totally fine, of course!), or if you think this issue isn't important and thus you see no need to prioritize it?

cdjackson, 2020-06-02:
Sorry Stefan - I did start to look at it the other day but got pulled on to some other stuff. I will take a look over the next few days, or possibly at the weekend.

triller-telekom, 2020-06-02:
Thanks Chris, no worries.

I didn't mean to put any pressure on you, I was just interested if this is an issue where you also think its worth solving quickly or one that isn't worth dedicating a high priority to it.

Thank you for looking into it, but take your time.

Regarding testing of this issue: This can be done with the openhab binding and starting to pair a device and pulling its power right after it blinked, so the node is created in ZSS (and possibly on the chip). then you can restart the binding where the zss discovery will start for that empty node.

If you then wait for the 13 retries of the network_address request (i.e. the first task) to be passed before turning on power to for the device again, you won't be able to pair it, because there is already an existing discovery which is unsuccessful (or will be unsuccessful soon) and cannot be restarted.

Let me know if you have further questions.

cdjackson, 2020-06-06:
Just to answer to your question -:

> We send a IEEE address request to ourselves. Is this some requirement by the zigbee spec? because we should know our address, at latest after a first startup because then we have the coordinator node with network address 0.

No - it's not a requirement in the spec, but it's a way to discover devices that are in the network since there is no other way to do this. By this, I mean that there is no central list of devices that have joined - the only thing we can do is iterate through the nodes we know about and look at their neighbour tables to continue the discovery. 

Of course this is not required if the network is persisted, but if this is ever lost, then the network cannot be rediscovered.

PR #1065 (2020-06-10): Fix the partial discovery of zigbee nodes
Fixes #1026
commit 032716a9 (2020-06-10): Fix the partial discovery of zigbee nodes (#1065)

* Fix the partial discovery of zigbee nodes

Fixes #1026
closed 2020-06-10 by cdjackson