zsmartsystems/com.zsmartsystems.zigbee #1031: AshFrameHandler and NAKs

dtodor, 2020-04-06:
The `AshFrameHandler` is not handling NAKs according to the `UG101: UART-EZSP Gateway Protocol Reference`. Sections 5.9 and 5.10 state that a NAK should be sent only once when a frame error is detected (out of sequence, CRC error, etc.) and this should be tracked by setting the rejection condition.

The problem that we've observed with a particular installation was that often frames had CRC errors (due to an Ember firmware issue or due to problems with the serial communication, still unclear and under investigation), resulting every time in a NAK being sent to the NCP:

```
final AshFrame packet = AshFrame.createFromInput(packetData);
AshFrame responseFrame = null;
if (packet == null) {
  logger.debug("<-- RX ASH error: BAD PACKET {}", frameToString(packetData));

  // Send a NAK
  responseFrame = new AshFrameNak(ackNum);
} else {
   ...
}
```
This caused the NCP to go into an error state, eventually sending an ASH ERROR frame with the error code `0x51`.

Further, it would be helpful to log CRC errors and maybe add a diagnostics counter for those.

cdjackson, 2020-04-06:
Thanks @dtodor. This sounds useful - possibly something I missed when implementing this initially.

If you can propose a PR it would be appreciated?

Currently we count bad packets - or NAKs sent. It's hard I think to differentiate between different types of bad packet - if the CRC fails, then you can't trust anything, so the log entry that you provide above is effectively logging the CRC error (ie bad packet) and this is also counted as a NAK that is sent.

If any change here changes this (eg if we're not sending a NAK for every bad packet as you suggest here) then I've no problem with counting the number of bad packets as a separate counter.

dtodor, 2020-04-06:
I'll try to make the necessary changes and will post a merge request.

closed 2020-04-15 by cdjackson