zsmartsystems/com.zsmartsystems.zigbee #1034: RejectedExecutionException thrown by ZigBeeNetworkDatabaseManager

YordanDZhelev, 2020-04-14:
In `ZigBeeNetworkDatabaseManager` the state of the internal `ScheduledExecutorService` is not verified before submitting a new task in `saveNode(ZigBeeNode node)` method. 
This might lead to `RejectedExecutionException` if the `ScheduledExecutorService` is already shut down. 
I would suggest to either 
1. check that the `ScheduledExecutorService` is not shut down before trying to submit new task, or
2. provide a public method in `ZigBeeNetworkDatabaseManager` so in `ZigBeeNetworkManager` will be able to tell the status when node is updated.

cdjackson, 2020-04-14:
Can you explain what you mean by point 2 - there is already a public method in ```ZigBeeNetworkDatabaseManager``` to allow ```ZigBeeNetworkManager``` to advise that the node is updated - it is ```nodeUpdated```.

Can you also advise when you see this happening as it shouldn't really occur. I think it can only happen if ```nodeUpdated``` is called after ```shutdown``` which I don't think should be possible - although I've not checked ```ZigBeeNetworkManager``` code yet so wanted to see the conditions this happens under.

YordanDZhelev, 2020-04-14:
By "provide a public method" I mean a method to provide `ZigBeeNetworkManager` with a way to tell if the `ZigBeeNetworkDatabaseManager` has already been shut down. Something like `public boolean isShutDown()` or `public boolean isRunning()` in `ZigBeeNetworkDatabaseManager` class. 

This can happen by calling `ZigBeeNetworkManager.serializeNetworkDataStore()` after shut down of the `ZigBeeNetworkManager`.

cdjackson, 2020-04-14:
Firstly, if we added an "isShutdown" method to the database manager, that doesn't really help does it? At application level, you cannot access the database manager directly - so it still can't be used to stop you calling ```serializeNetworkDataStore```.

Then, what would you do with this "isShutdown" method? If the data can't be written - what do you intend to do? I don't think there's anything you can do.

> This can happen by calling ZigBeeNetworkManager.serializeNetworkDataStore() after shut down of the ZigBeeNetworkManager.

This is therefore a problem in your code then isn't it? If we stop the exception, then it should be fine - there is nothing that can be done anyway as the system is shutdown already, and the framework will have saved the state when it was shut down so all data will have been persisted.

PR #1038 (2020-04-22): Ensure executor is not shut down already
Fixes #1034 
@YordanDZhelev is this ok for you?
closed 2020-04-22 by cdjackson