zsmartsystems/com.zsmartsystems.zigbee #1035: Overlapping attribute ids

spudwebb, 2020-04-14:
Some cluster attributes ids are overlapping, see for example in ZclMeteringCluster.java

```
public static final int ATTR_PREVIOUSDAY16CONSUMPTIONRECEIVED = 0x0431;

public static final int ATTR_CURRENTWEEKCONSUMPTIONRECEIVED = 0x0431;
```
I believe the issue is here:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/759570d1975c25395c87c41507046e2340b608a2/com.zsmartsystems.zigbee.autocode/src/main/java/com/zsmartsystems/zigbee/autocode/ZigBeeZclClusterGenerator.java#L176-L177

I think the id shouldn't be generated as (attribute.code + arrayCount) but as (attribute.code + arrayCount - attribute.arrayStart)

cdjackson, 2020-04-14:
Thanks. It looks like you’re probably correct - if you’re able to provide a PR for this change it would be appreciated?

spudwebb, 2020-04-14:
sorry I can't. I found and fixed the problem in the ZigbeeNet project (https://github.com/Mr-Markus/ZigbeeNet/pull/112), and I thought I would mention it here because I checked there was the same issue here, but I don't have a java dev environment to make a PR for this project.

cdjackson, 2020-04-14:
Ok, thanks for reporting. I'll take a look.

PR #1037 (2020-04-15): Fix autocoder array count generator
Fixes #1035 

Thanks for reporting this @spudwebb - you probably should check this, and more to the point, you should check your implementation as from the quick look I had I suspect it's wrong as the enum/const looks to be wrong.

Eg -:

[image]

My fix is -:
[image]

Your is -:
[image]
closed 2020-04-15 by cdjackson