zsmartsystems/com.zsmartsystems.zigbee #1039: Whitelist for network joining process

mikomarrache, 2020-04-19:
Hi,

I'm looking for a way to allow ZED joining my network based on a preconfigured whitelist.

I'm using the library to configure the NCP, providing configuration similar to the console example project. After the network is started, I'm not able to make my ZED join the network until I enable network join. However, I would like to be able to instruct the NCP to respond to join request only for particular devices (without enabling network join).

Is it possible with ZigBee and if yes, is there such function provided by the library?

I'm using the Silicon Labs Ember dongle.

Thanks

cdjackson, 2020-04-19:
Yes, it should be possible if I understand you correctly…. You want to join a device to the network, without going through the join process?  I think you can do this, but you need to tell the ZED (or ZR) all the preconfigured security information. Take a look over the Silabs UG103 series - I think one of those documents is on security and I think it does mention how to do this (at top level). I’d suggest to have a read through that, then let me know what it states and we can discuss further about how to manage that in the library.

mikomarrache, 2020-04-19:
I'm not sure where to look at. Could you provide me the reference of the document?

Did you mean to look at the commissioning cluster?

cdjackson, 2020-04-19:
No - it's not the commissioning cluster. If I'm right, it's related to how you configure the initial network state when configuring the NCP for the first time.

I thought UG103 had one of the volumes about security, but the following looks a good reference -:

https://www.silabs.com/documents/public/application-notes/an1233-zigbee-security.pdf

That said, I can't see what I was looking for. However, I think that if you know both the link key, and the network key, then you should be able to configure these during the `SetInitialSecurityState` configuration. Take a look in the `EmberNetworkInitialisation` class - there are a number of methods there for joining and rejoining networks. 

I think if you know both keys it should work - I think I've tried this, but I'm not 100% sure so don't shoot me if I'm wrong ;). I'd suggest to hack around with these methods to see if you can make this work, then we can see if there's anything we need to change in the library to support this properly.

mikomarrache, 2020-04-19:
I see what you mean - however, I'm more interested about the case of ZEDs after factory reset. In such case, the ZEDs don't hold any network key I guess. They simply send a beacon request on every channel one after the other until they get a beacon. The idea would be to make the NCP reply with a beacon selectively based on some whitelist.

cdjackson, 2020-04-19:
I think if you factory reset, then there’s no easy way to recover. You could look to leave joining open at all times on you ZC and use the standard ZigBeeAlliance09 key. This should allow any device to join, but you would then need to manage security yourself as you already are thinking, but this isn’t something I’ve looked at.

Maybe I also misunderstood what you are doing - I thought you wanted to use the library as a ZED to join a network, but I think now you are running the library as a ZC and you want other devices to join and implement a whitelist in the library?

mikomarrache, 2020-04-19:
No - I'm using the library as a ZC and I would like as a ZC to be able to allow join selectively based on a whitelist. I know the list of ZEDs upfront so I would like the following:

- Always reply with a beacon - this is because the beacon request doesn't include the ZED extended address so I have no way to take the decision at this stage
- Control the association response based on the whitelist (at this point, I receive the ZED extended address so I'm able to take the decision if to accept/reject the association request)

mikomarrache, 2020-04-19:
> I think if you factory reset, then there’s no easy way to recover. You could look to leave joining open at all times on you ZC and use the standard ZigBeeAlliance09 key. This should allow any device [...]

As of ZigBee 3, is there any way to broadcast a permanent permit join? Looking at the library code, it seems that duration 255 is used for this purpose but is ignored by the NCP. By permanently leaving joining open, do you mean periodically enabling joining with duration 254 seconds?

cdjackson, 2020-04-19:
> Always reply with a beacon - this is because the beacon request doesn't include the ZED extended address so I have no way to take the decision at this stage

I don't think this is possible, but I have seen some new functions in the NCP (UG100) relating to beacons. I'm not sure this is needed though.

> Control the association response based on the whitelist (at this point, I receive the ZED extended address so I'm able to take the decision if to accept/reject the association request)

This is also not possible AFAIK. The join decision is made by the NCP based on the policy you set. You would have to allow all devices to join, then check your whitelist, and then make it leave if it was not one of your EDs.

> do you mean periodically enabling joining with duration 254 seconds?

Yes, exactly - you would need to periodically resend this - eg every 60 seconds or so send the joinEnable command. So long as it's done more often than every 254 seconds, it will be permanently enabled.

mikomarrache, 2020-04-19:
> I don't think this is possible, but I have seen some new functions in the NCP (UG100) relating to beacons. I'm not sure this is needed though.

I just checked - I can't see nothing related to what I'm trying to do.
 
> This is also not possible AFAIK. The join decision is made by the NCP based on the policy you set. You would have to allow all devices to join, then check your whitelist, and then make it leave if [...]

I saw this approach on the web - I'm not sure how that affects security since the network key would be known by everybody that would be physically in range. Unless there is a way to prevent sharing the network key at the library level right after a node join the network?

cdjackson, 2020-04-19:
> I'm not sure how that affects security since the network key would be known by everybody that would be physically in range.

Yes, this is the problem with using the well known key, but I'm not sure what else you can do. You could try something that tries to reprovision ZB3 install codes to get around this, but otherwise your requirements are heading down a route where it's not possible.

mikomarrache, 2020-05-05:
Thanks for your help! I'm closing the issue.

closed 2020-05-05 by mikomarrache