zsmartsystems/com.zsmartsystems.zigbee #1044: ZigBeeDongleEzsp should do formNetwork() if not joined

triller-telekom, 2020-05-04:
I just stumbled across line 478 in this code:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/5a91440b8e4774b9e81bd58e57a1da7cfbd87713/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ZigBeeDongleEzsp.java#L476-L502

Here we detect that the chip has not yet joined a network. However, we do line 496, i.e. the `formNetwork` call only if `reinitialize` has been set to `true`.

So I am wondering whether we should also set it to `true` if the chip reports us in line 478 that the chip isn't joined.

Is this a bug or is there a reason to only call `formNetwork` of the `reinitialize` flag has been set from outside?

cdjackson, 2020-05-04:
I don't think this is a bug. If you do a `formNetwork` then it will initialise the dongle which would completely wipe the network. If the network didn't come online for some transient reason, then I don't think we should just wipe everything and make people add all their devices to the network?

triller-telekom, 2020-05-04:
Right, that is why I am asking. So you are sure that the `formNetwork` wipes everything?

If so, would the command `EMBER_NO_FRAME_COUNTER_RESET=0x1000` be of any help? Because this states:

```
This denotes whether the device should NOT reset its
outgoing frame counters (both NWK and APS) when
::emberSetInitialSecurityState() is called. Normally it is
advised to reset the frame counter before joining a new
network. However in cases where a device is joining to the
same network again (but not using
::emberRejoinNetwork()) it should keep the NWK and APS
frame counters stored in its tokens
```

Also since the `deviceType` is `COORDINATOR` and not `ROUTER` we do not call line 501. Sou would it be of any help to rejoin the network if the chip reports we are not joined?

Actually I don't know how the chip gets into such situations, from one log I have:

```
2020-04-22T15:32:50.229+0200   193  DEBUG          [symbolicName=XXX, className=org.openhab.binding.zigbee.handler.ZigBeeCoordinatorHandler] - ZigBee Initialise: Previous device configuration was: channel=CHANNEL_11, PanID=50860, EPanId=000D6F000BEBC276
2020-04-22T15:32:50.235+0200   230  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - EzspSetConcentratorResponse [status=EZSP_SUCCESS]
2020-04-22T15:32:50.236+0200   192  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - ZigBeeNetworkManager startup: reinitialize=false, networkState=INITIALISING
2020-04-22T15:32:50.236+0200   230  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Dongle: Startup - reinitialize=false
2020-04-22T15:32:50.236+0200   230  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Adding Endpoint: ProfileID=0104, DeviceID=0050
2020-04-22T15:32:50.236+0200   230  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Adding Endpoint: Input Clusters   0000
2020-04-22T15:32:50.237+0200   230  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - EZSP Adding Endpoint: Output Clusters  0000
2020-04-22T15:32:50.253+0200   188  DEBUG          [symbolicName=com.qivicon.smarthome.config.system, className=com.qivicon.smarthome.config.system.internal.ServiceGatewayInfoImpl] - Event from topic smarthome/system/updated is coming in...
2020-04-22T15:32:50.266+0200   230  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.EmberNcp] - EzspAddEndpointResponse [status=EZSP_SUCCESS]
2020-04-22T15:32:50.297+0200   230  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.EmberNcp] - EzspNetworkInitResponse [status=EMBER_NOT_JOINED]
```

Unfortunately it happened on a device where I do not have access to, so I cannot reproduce.

triller-telekom, 2020-05-04:
One other thing that just came to my mind is: The gateway I mentioned in my previous post might have been a factory reseted one anyway.

So would it be an option that ZSS reports to the application, which uses ZSS, that the `startup` method was called with `reinitialise=false` but the chip has reported `EMBER_NOT_JOINED`?

Because in the case of a reseted gateway the application would know that there are no devices and it could then call `startup` with `reinitialise=true`. WDYT?

dtodor, 2020-05-04:
If the NCP reports ```EmberStatus.EMBER_NOT_JOINED```, then I would assume that there is no network. This is a status code, not an error code. At least this would be my assumption. I don't think that performing a network init after some time will result in a different status. WDYT?

cdjackson, 2020-05-04:
I know of at least one case where the NCP can return NOT_JOINED where it is joined and it will join later if tried again. I won't change the code in the library to reinitialise the network as in my opinion  it's a bad idea that could lead to systems being needlessly reset (and therefore unhappy users!).

We could probably do what you suggest Stefan - fail at that line if reinitialise is false and then you can reinitialise if you want by calling `startup` again.  How does that sound?

triller-telekom, 2020-05-04:
> We could probably do what you suggest Stefan - fail at that line if reinitialise is false and then you can reinitialise if you want by calling startup again. How does that sound?

Sounds good to me. At least this way the application can decide how to proceed.

So if there is really a case where a NOT_JOINED is returned but it will work later, we should definitely not create a new network automatically but leave it to the application.

PR #1045 (2020-05-05): Return error when Ember network not found
@triller-telekom please let me know if this is ok for you.

Fixes #1044
closed 2020-05-05 by cdjackson