zsmartsystems/com.zsmartsystems.zigbee #1049: Warning during network manager initialization

mikomarrache, 2020-05-05:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/43fc6ac240b846b817bccedc628bdbad633b1238/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNetworkManager.java#L403

Hi,

During network manager initialization, I see the warning above.

I understand it is caused by the fact the network address is not set but I'm not sure how the network address should be set at this stage.

Am I missing something? As far as I can see, the network address is set during startup, not during initialization.

Thanks,
Mickael

cdjackson, 2020-05-05:
When is this logged? 

There are 2 places where this can be set - you are probably right that it should not be set during the `initialize` method, but should be when the system is online (when this is also called).

mikomarrache, 2020-05-05:
The method addLocalNode() (which creates the log event) is also called during initialization.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/43fc6ac240b846b817bccedc628bdbad633b1238/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNetworkManager.java#L389

cdjackson, 2020-05-05:
Yes, that I know. 

Sorry if I wasn’t clear, but what I was asking is when is this logged? This method is called in multiple places so I wanted to understand which one is the one causing this error. I guess it is during the initialisation method, but I wanted to understand the issue before changing anything.

cdjackson, 2020-05-05:
Sorry - I just checked your link and see it references the initialize method - the email version doesn't show this so well.

I think it should be ok to remove this from the initialize method.

mikomarrache, 2020-05-05:
I tested the change - it causes an issue during startup.

A node is expected during startup (see the code below).

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/43fc6ac240b846b817bccedc628bdbad633b1238/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNetworkManager.java#L635

Therefore, the local node is expected at initialization but the error should not be logged because of the missing network address. I created a PR. #1050

cdjackson, 2020-07-04:
This was resolved already

closed 2020-07-04 by cdjackson