zsmartsystems/com.zsmartsystems.zigbee #1051: Setting channel mask for quietest channel selection

mikomarrache, 2020-05-05:
Hi,

We would like to be able to set a channel mask before forming the network so that the quietest channel will be selected (in case no channel is set before starting the network manager).

There is a field channels in the Ember network parameters, but I guess it hasn't been intended for this use, am I right?

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/4d678e66b4530b44072cb6ece35b12594118eb34/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ezsp/structure/EmberNetworkParameters.java#L84

Would it be possible to use it for this purpose?

Thanks,
Mickael

triller-telekom, 2020-05-08:
For Ember and Telegesis chips, this is already implemented: See

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/4d678e66b4530b44072cb6ece35b12594118eb34/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/internal/EmberNetworkInitialisation.java#L143-L145

and

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/4d678e66b4530b44072cb6ece35b12594118eb34/com.zsmartsystems.zigbee.dongle.telegesis/src/main/java/com/zsmartsystems/zigbee/dongle/telegesis/ZigBeeDongleTelegesis.java#L871-L876

Just pass the UNKNOWN channel when creating the network.

closed 2020-07-21 by stale[bot]

tejas-n, 2020-08-24:
@triller-telekom But I don't think you can set a custom channel mask, right? For eg: if I wanted the library to just scan channels 11, 14, 15, 19, 20, 24, and 25, is there a way to do that?
Another closely related question that I have is should would the library automatically scan and select the appropriate channel if the co-ordinator reboots and starts the network againg with `startup(false)`?

Is #958 related to these questions?

triller-telekom, 2020-08-25:
> But I don't think you can set a custom channel mask, right? For eg: if I wanted the library to just scan channels 11, 14, 15, 19, 20, 24, and 25, 

Correct.

> is there a way to do that?

No, this hasn't been implemented yet. Feel free to come up with a PR :-)

> Another closely related question that I have is should would the library automatically scan and select the appropriate channel if the co-ordinator reboots and starts the network again with `startup(false)`?

No.

> Is #958 related to these questions?

Yes.

tejas-n, 2020-08-25:
Thanks, @triller-telekom. I can give it a shot in my free time, I'll ask for further details on the other ticket 😄