zsmartsystems/com.zsmartsystems.zigbee #1053: Format `indexOrDestination`in `EzspMessageSentHandler.toString` as hex?

hsudbrock, 2020-05-08:
When checking logs for entries related to a particular network address, I sometimes miss the log entries for the `EzspMessageSentHandler`, because they contain the network address formatted as decimal, not as hexadecimal number.

I wonder whether we could change this to hexadecimal as well.

I am somewhat hesitant of just changing it, because the network address is in the field `indexOrDestination`, which might contain a network address, but also 'the address table or binding index' for non-direct unicasts.

Chris, would it be sensible to change this to hexadecimal formatting?

cdjackson, 2020-05-12:
Hi Henning. I don't see any issue changing this. It is only formatting in the `toString` method, so won't affect anything and in general it is clearer, and more standard (within the framework), to use hex.

I'll update...

hsudbrock, 2020-05-13:
Thanks Chris!

closed 2020-05-13 by hsudbrock