zsmartsystems/com.zsmartsystems.zigbee #1054: Transaction queue filling up

cdjackson, 2020-05-11:
An openHAB user has reported that commands stop being sent to a Centralite thermostat. From the logs, initially it does work ok, but at some stage the transaction queue length is increasing, and nothing is ever sent to the device again.

wtf911, 2020-07-10:
Just bumping this.

cdjackson, 2020-09-10:
I think that this may have been resolved with #1110 which increases the transaction timeout and #1113 which removes transactions from the queue if they time out.

Without these two changes, transactions could time out, but remain in the queue, and the application may also re-queue them, thus resulting in the queue increasing.

I'll close this and if there's further issues we will need to look at it again.

closed 2020-09-10 by cdjackson