zsmartsystems/com.zsmartsystems.zigbee #1056: ASH/EZSP initialization thread hangs forever if Ash reset frame is not acknowledged

hsudbrock, 2020-05-14:
In some error conditions, the `AshFrameRst` frame is not acknowledged with an `AshFrameRstAck` frame. (This happens, e.g., when the baud rate for the serial connection is wrongly configured.)

Currently, if this happens when connecting to the frame handler [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/80f99c1466158a552d2c39dd1be4abd5e716172e/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ZigBeeDongleEzsp.java#L1242) during the `ZigBeeDongleEzsp` initialization, the initialization hangs indefinitely in the `ncp.getVersion(..)` call [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/80f99c1466158a552d2c39dd1be4abd5e716172e/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/ZigBeeDongleEzsp.java#L1248).

It hangs indefinitely because `ncp.getVersion()` leads to calling `AshFrameHandler.sendEzspTransaction`, which waits for the EZSP transaction to finish [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/80f99c1466158a552d2c39dd1be4abd5e716172e/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/internal/ash/AshFrameHandler.java#L786), but the transaction won't finish before the `AshFrameHandler` has successfully connected (the frame handler sends out `AshFrameRst` frames again and again, but they get never acknowledged in the error condition mentioned above).

I would like to handle this more gracefully, so that the thread trying to initialize the EZSP protocol does not hang forever.

My suggestion is to fail the initialization after a timeout (say, 10 seconds?). I see two options for this:

1. Wait until the frame handler has connected before issuing the `getVersion` command (by inspecting the currently private variable `AshFrameHandler.stateConnected`).
2. Add a timeout when resolving the future in `AshFrameHandler.sendEzspTransaction` [here](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/80f99c1466158a552d2c39dd1be4abd5e716172e/com.zsmartsystems.zigbee.dongle.ember/src/main/java/com/zsmartsystems/zigbee/dongle/ember/internal/ash/AshFrameHandler.java#L786).

I would prefer variant 2, because it avoids the situation that we check that the frame handler is connected, but it disconnects for whatever reason before we can send out the getversion command.

I could provide a PR, but I think it would be good to align here first on how to improve this. Any opinions?

hsudbrock, 2020-05-14:
Just as info, another error condition where the `AshFrameRst` is not acknowledged is when the chip is in bootloader mode. I.e., also when the chip is in bootloader mode the thread trying to initialize the EZSP protocol does hang forever. (Tested with an Ember USB dongle.)

cdjackson, 2020-05-15:
Hi Henning. I don't think I see any issue with adding a timeout to the `get` as per your option 2. For most commands, there should never be a need to queue at this point for 10 seconds, so even if there is some queuing at this level (which there will be), a 10 second timeout should really never be hit.

Have you tested if it resolves your issue and allows the startup to continue/fail?

hsudbrock, 2020-05-15:
Thanks for your comment, Chris!

> Have you tested if it resolves your issue and allows the startup to continue/fail?

I have not tested it yet, but I will give it a try. If there are no issues, I will provide a PR.

hsudbrock, 2020-05-18:
> Have you tested if it resolves your issue and allows the startup to continue/fail?

I have tested this now, and everything worked normal when the chip acknowledged the reset frame, and the startup failed as expected after around 10 seconds when the chip did not acknowledge.

I could not test, however, with a larger setup with many ZigBee devices, so I cannot say for sure that a 10 second timeout is large enough.

When I suggested the 10 seconds, I only thought about the `getVersion` call during EZSP initialization, and not about the EZSP transactions sent during normal operations later. Do you think it is sensible to use a higher timeout to avoid too early cancellation of transactions, say, on a system under high load? We could even say "timeout after 10 seconds in the `getVersion` call, but only timeout after 1 minute later.

I am rambling a little here, because I am quite unsure about a good value for the timeout during "normal operation" on a more heavily loaded system.

cdjackson, 2020-05-18:
> I could not test, however, with a larger setup with many ZigBee devices

This shouldn't matter - the reset won't be affected, and all other commands should receive fast responses.

> When I suggested the 10 seconds, I only thought about the getVersion call during EZSP initialization, and not about the EZSP transactions sent during normal operations later.

That's fine - I did think about that when I was talking earlier about "most commands" in my response above. There shouldn't be a situation where commands are released to the ASH layer and queued for a long period - all command **should** get "immediate" responses and there should not be excessive queuing in this layer as the transaction manager should have already "smoothed things out". The queueing in the ASH layer is really just there to cope with the "immediate" delays caused by NCP interaction.

So, I think I'm happy with this timeout.

hsudbrock, 2020-05-18:
OK, thanks Chris. I will open the PR then.

PR #1061 (2020-05-18): Timeout sending EZSP transactions after 10 seconds
Resolves #1056 

I have added the timeout for both ASH and SPI frame handlers.
closed 2020-05-18 by cdjackson