zsmartsystems/com.zsmartsystems.zigbee #1077: Transaction handling vs zigbee routing

triller-telekom, 2020-07-01:
In our application we have implemented a feature that we define a device as not reachable if we send a command to it and this will timeout. In ZSS speech this means that we evaluate `CommandResult.isTimeout()` on the `CommandResult` returned by a `ZCLAttribute.writeValue(value)` call for example.

I noticed that we get a `true` (so it says it timed out) value in cases where the device is reachable and also reacting to our commands. Here is a corresponding log where we tried to set the target value of a bitron wall thermostat (902010/32) to 12 degrees Celsius:

```
2020-06-26T13:31:44.923+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.zcl.ZclCluster] - 000D6F000C2B93BB: Writing Server cluster 513, attribute 18, value 1200, as dataType SIGNED_16_BIT_INTEGER
2020-06-26T13:31:44.924+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: WriteAttributesCommand [Thermostat: 0000/0 -> EE91/1, cluster=0201, TID=2A, records=[WriteAttributeRecord [attributeDataType=SIGNED_16_BIT_INTEGER, attributeIdentifier=18, attributeValue=1200]]]
2020-06-26T13:31:44.925+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=CLIENT_TO_SERVER, disableDefaultResponse=false, manufacturerCode=0, sequenceNumber=42, commandId=2]
2020-06-26T13:31:44.927+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX APS: ZigBeeApsFrame [sourceAddress=0000/1, destinationAddress=EE91/1, profile=0104, cluster=0201, addressMode=DEVICE, radius=31, apsSecurity=false, ackRequest=true, apsCounter=67, rssi=--, lqi=--, payload=00 2A 02 12 00 29 B0 04]
2020-06-26T13:31:44.948+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspSendUnicastResponse [networkId=0, status=EMBER_SUCCESS, sequence=A4]
2020-06-26T13:31:44.952+0300   427  DEBUG          [symbolicName=de.telekom.smarthomeb2c.core, className=de.telekom.smarthome.b2cmodul.event.SmartHomeEventHandlerV2] - smartHomeEvent: {}
2020-06-26T13:31:44.979+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=E723, sourceEui=000D6F000A7BB67D, lastHopLqi=255, lastHopRssi=-43, relayList=1F7A]
2020-06-26T13:31:46.002+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=E723, sourceEui=000D6F000A7BB67D, lastHopLqi=255, lastHopRssi=-43, relayList=1F7A]
2020-06-26T13:31:46.033+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspMessageSentHandler [networkId=0, type=EMBER_OUTGOING_DIRECT, indexOrDestination=E723, apsFrame=EmberApsFrame [profileId=0104, clusterId=FC45, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=A3], messageTag=41, status=EMBER_SUCCESS, messageContents=]
2020-06-26T13:31:46.033+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=41 state=RX_ACK
2020-06-26T13:31:47.056+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=E723, sourceEui=000D6F000A7BB67D, lastHopLqi=255, lastHopRssi=-43, relayList=1F7A]
2020-06-26T13:31:49.667+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspMessageSentHandler [networkId=0, type=EMBER_OUTGOING_DIRECT, indexOrDestination=EE91, apsFrame=EmberApsFrame [profileId=0104, clusterId=0201, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=A4], messageTag=2A, status=EMBER_DELIVERY_FAILED, messageContents=]
2020-06-26T13:31:49.668+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=2A state=RX_NAK
2020-06-26T13:31:51.711+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=EE91, sourceEui=000D6F000C2B93BB, lastHopLqi=255, lastHopRssi=-39, relayList=1F7A]
2020-06-26T13:31:51.742+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0201, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=C9], lastHopLqi=255, lastHopRssi=-39, sender=EE91, bindingIndex=255, addressIndex=255, messageContents=18 2A 04 00]
2020-06-26T13:31:51.743+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX APS: ZigBeeApsFrame [sourceAddress=EE91/1, destinationAddress=0000/1, profile=0104, cluster=0201, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=C9, rssi=-39, lqi=FF, payload=18 2A 04 00]
2020-06-26T13:31:51.743+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX ZCL: ZclHeader [frameType=ENTIRE_PROFILE_COMMAND, manufacturerSpecific=false, direction=SERVER_TO_CLIENT, disableDefaultResponse=true, manufacturerCode=0, sequenceNumber=42, commandId=4]
2020-06-26T13:31:51.744+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: WriteAttributesResponse [Thermostat: EE91/1 -> 0000/1, cluster=0201, TID=2A, records=[WriteAttributeStatusRecord [status=SUCCESS, attributeIdentifier=0]]]
```

Note the transaction ID=2A of the `WriteAttributesCommand`. Apparently the chip does not know how to reach our thermostat device (with IEEE=000D6F000C2B93BB) because we receive a `EMBER_DELIVERY_FAILED' response for our transaction 2A:

```
2020-06-26T13:31:49.667+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspMessageSentHandler [networkId=0, type=EMBER_OUTGOING_DIRECT, indexOrDestination=EE91, apsFrame=EmberApsFrame [profileId=0104, clusterId=0201, sourceEndpoint=1, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY, EMBER_APS_OPTION_RETRY], groupId=0, sequence=A4], messageTag=2A, status=EMBER_DELIVERY_FAILED, messageContents=]
2020-06-26T13:31:49.668+0300   173  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX STA: msgTag=2A state=RX_NAK
```

Shortly afterwards we receive some routing information on how to reach our thermostat `000D6F000C2B93BB`: via the node `1F7A` (it is a Bitron 902010/25 smart plug capable of routing/meshing):

```
2020-06-26T13:31:51.711+0300   291  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.ember, className=com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] - RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=EE91, sourceEui=000D6F000C2B93BB, lastHopLqi=255, lastHopRssi=-39, relayList=1F7A]
```

And once the chip receives this information, it apparently sends out the `WriteAttributesCommand`, because what we see shortly afterwards is that the device replies to it!

```
[... cut ...] onse [Thermostat: EE91/1 -> 0000/1, cluster=0201, TID=2A, records=[WriteAttributeStatusRecord [status=SUCCESS, attributeIdentifier=0]]]
```

However the `CommandResult.isTimeout` seems to be filled with the first reply and this means, it is treated as timed out although it was sucessful.

No I am wondering whether this is a bug in the transaction handling of ZSS or a bug in the chip. Because the chip tells us that it cannot reach the device but then silently retries to send the command anyway...

cdjackson, 2020-07-11:
Interesting one this...

> And once the chip receives this information, it apparently sends out the WriteAttributesCommand, because what we see shortly afterwards is that the device replies to it!

I'm going to speculate on a different theory... It might be that the device was sleeping - the parent buffers the request, waiting for the thermostat to wake (potentially it will wake every 7 seconds?). For some reason the NCP has timed out after 5 seconds (in fact maybe even slightly less!). I don't think it should - by default the NCP is configured for 7.6 seconds -:

```
        stackConfiguration.put(EzspConfigId.EZSP_CONFIG_INDIRECT_TRANSMISSION_TIMEOUT, 7680);
```

At least I think this is the right timer... Have you changed this value? It seems to me that the possible cause of this could be that the NCP has timed out, but the parent is still got the message buffered - the thermostat wakes up after 7 seconds, receives the message and responds.

Some time back we added a call in the Ember NCP `setNodeDescriptor`. The purpose of this is to allow the dongle code to know what type of device a node is - eg if it is a ZED or ZR. If it's an End Device, then it should send the `setExtendedTimeout` command to tell the NCP that the device is a sleepy node - this extends this timeout since the NCP will only otherwise use the 7.6 second value if the sleepy device is a direct child of the coordinator (which it isn't in this case).

I assume you are running the code with this method? (see line ~800 of `ZigBeeDongleEzsp.java`. If so, you might want to see if this is being called for this device.

So, this aside, I can't see that the framework can do anything different? I'm open to being challenged on this, but here we see that the NCP has said the command was not delivered, so I don't think that ZSS should continue to wait. Of course, in this instance, it is incorrect - it was delivered later, so we do receive the response. I think therefore that the transaction management is ok, but it would be worth trying to understand if this timer is set correctly...

triller-telekom, 2020-07-13:
Thanks Chris for your feedback on this.

I think you might have spotted a bug in the line you are referring to:

In ~800 of `ZigBeeDongleEzsp.java` I see:
```
if (!nodeDescriptor.getMacCapabilities().contains(MacCapabilitiesType.RECEIVER_ON_WHEN_IDLE))
```

However the `ZigbeeNode` from this device has:

```
<ZigBeeNode>
  <ieeeAddress>000D6F000A7BB67D</ieeeAddress>
  <networkAddress>59171</networkAddress>
  <nodeDescriptor>
    <apsFlags>0</apsFlags>
    <bufferSize>82</bufferSize>
    <complexDescriptorAvailable>false</complexDescriptorAvailable>
    <manufacturerCode>4174</manufacturerCode>
    <logicalType>END_DEVICE</logicalType>
    <serverCapabilities class="sorted-set"/>
    <incomingTransferSize>82</incomingTransferSize>
    <outgoingTransferSize>82</outgoingTransferSize>
    <userDescriptorAvailable>false</userDescriptorAvailable>
    <frequencyBands>
      <FrequencyBandType>FREQ_2400_MHZ</FrequencyBandType>
    </frequencyBands>
    <macCapabilities class="sorted-set">
      <MacCapabilitiesType>REDUCED_FUNCTION_DEVICE</MacCapabilitiesType>
    </macCapabilities>
    <extendedEndpointListAvailable>false</extendedEndpointListAvailable>
    <extendedSimpleDescriptorListAvailable>false</extendedSimpleDescriptorListAvailable>
    <stackCompliance>0</stackCompliance>
  </nodeDescriptor>
  <powerDescriptor>
    <currentPowerMode>RECEIVER_ON_IDLE</currentPowerMode>
    <availablePowerSources>
      <PowerSourceType>DISPOSABLE_BATTERY</PowerSourceType>
    </availablePowerSources>
    <currentPowerSource>DISPOSABLE_BATTERY</currentPowerSource>
    <powerLevel>FULL</powerLevel>
  </powerDescriptor>
  <endpoints>
...
```

So the `RECEIVER_ON_WHEN_IDLE` information is actually only visible in the `PowerDescriptor` and not within the `NodeDescriptor`s mac capabilities set.

I have the full node attached here: [000D6F000A7BB67D.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/4912970/000D6F000A7BB67D.txt)

cdjackson, 2020-07-14:
Hi Stefan,
Do you mean a bug in the device? I guess that's possible. I would have preferred not to pass the `PowerDescriptor` down to the dongle layer as well as the `NodeDescriptor` though, but maybe we need to do this if devices are providing inconsistent information.

Another option might be to use the device type - ie if the device is an RFD then it is sleepy and should have the extended timeout.

To be clear - is this the same device that has the problem? The addresses seem to be different from the first log, but maybe it is the same device type? I'm just wondering how widespread this issue is.

triller-telekom, 2020-07-14:
Hi Chris,

you are right, i copy&pasted from the wrong xml file. However, the correct one is similar to the one I posted above:

```
<ZigBeeNode>
  <ieeeAddress>000D6F000C2B93BB</ieeeAddress>
  <networkAddress>61073</networkAddress>
  <nodeDescriptor>
    <apsFlags>0</apsFlags>
    <bufferSize>82</bufferSize>
    <complexDescriptorAvailable>false</complexDescriptorAvailable>
    <manufacturerCode>4209</manufacturerCode>
    <logicalType>END_DEVICE</logicalType>
    <serverCapabilities class="sorted-set"/>
    <incomingTransferSize>82</incomingTransferSize>
    <outgoingTransferSize>82</outgoingTransferSize>
    <userDescriptorAvailable>false</userDescriptorAvailable>
    <frequencyBands>
      <FrequencyBandType>FREQ_2400_MHZ</FrequencyBandType>
    </frequencyBands>
    <macCapabilities class="sorted-set">
      <MacCapabilitiesType>REDUCED_FUNCTION_DEVICE</MacCapabilitiesType>
    </macCapabilities>
    <extendedEndpointListAvailable>false</extendedEndpointListAvailable>
    <extendedSimpleDescriptorListAvailable>false</extendedSimpleDescriptorListAvailable>
    <stackCompliance>0</stackCompliance>
  </nodeDescriptor>
  <powerDescriptor>
    <currentPowerMode>RECEIVER_ON_IDLE</currentPowerMode>
    <availablePowerSources>
      <PowerSourceType>DISPOSABLE_BATTERY</PowerSourceType>
    </availablePowerSources>
    <currentPowerSource>DISPOSABLE_BATTERY</currentPowerSource>
    <powerLevel>FULL</powerLevel>
  </powerDescriptor>
  <endpoints>
...
```

First one was a Centralite 3310-D and the correct one is the Bitron Home 902010/32. See the full file here: [000D6F000C2B93BB.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/4917653/000D6F000C2B93BB.txt)

So to answer your question:

> To be clear - is this the same device that has the problem? The addresses seem to be different from the first log, but maybe it is the same device type? I'm just wondering how widespread this issue is.

There are at least 2 devices with this "problem".

> I would have preferred not to pass the `PowerDescriptor` down to the dongle layer as well as the `NodeDescriptor`

I am also not sure if we should pass down all kinds of data to the dongle layer...

Your other approach:

> if the device is an RFD then it is sleepy and should have the extended timeout

sounds more promising. Also in for both devices it would have worked, because both report `<MacCapabilitiesType>REDUCED_FUNCTION_DEVICE</MacCapabilitiesType>` in their `NodeDescriptor`.

I can provide a PR with the changes if you like.

cdjackson, 2020-07-14:
> There are at least 2 devices with this "problem".

Of course - they're all working to the standards :)

> I can provide a PR with the changes if you like.

Sure - that would be great - thanks. I guess it's probably best to `OR` the two - ie we extend the timeout if there's either `RECEIVER_ON_WHEN_IDLE` set, `OR` it's a `REDUCED_FUNCTION_DEVICE` - to cover as many "standards" as possible :)

triller-telekom, 2020-07-14:
> to cover as many "standards" as possible 

Yes, indeed I love the zigbee standard with all its "may"s instead of "must"s :-)

I have created the PR.

cdjackson, 2020-07-14:
Thanks Stefan. It would of course be interesting to know if this actually solves your issue you report above. Is this something that you see often, or I guess it's not so common?

triller-telekom, 2020-07-14:
Unfortunately it is hard to reproduce :( We were only able to catch it 2 times so far in a larger setup, i.e. >15 devices paired.

But I will certainly have an eye on it if we run into such a situation again, where we get a timeout from the chip and afterwards a response.

cdjackson, 2020-07-14:
That's what I expected. I'd suggest we close this for now at least - or maybe you can confirm that the `setExtendedTimeout` call is correctly made now when these devices start, and if it is then let's close this as I can't see too many things we can change - if it re-occurs, then we can re-discuss.

triller-telekom, 2020-07-17:
Indeed the call is now done. However, on startup there is a race condition so we sometimes run into this:

```
2020-07-17T14:12:06.028+0200  174 WARN  [NotificationService-thread-2,com.zsmartsystems.zigbee.ZigBeeExecutors] Uncaught exception in thread NotificationService-thread-2
java.lang.NullPointerException
	at com.zsmartsystems.zigbee.dongle.ember.EmberNcp.setExtendedTimeout(EmberNcp.java:991)
	at com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp.setNodeDescriptor(ZigBeeDongleEzsp.java:805)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager$6.run(ZigBeeNetworkManager.java:1324)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
```

This can easily be reproduced by pairing a RFD like a battery powered door/window contact and then restarting the system. During restart we read the nodes from storage and I think this is where the race condition might be. I haven't investigated this further.

Since you know the architecture of the ZSS stack better than me, maybe you know why sometimes the `protocolHandler` is `null` in `EmberNcp` during startup. I have seen it being `null` in the debugger, but I didn't know at which point of the initialization I was.

cdjackson, 2020-07-19:
This is strange, since this should be deferred until the network is `ONLINE`, and that is what we see...

At this line -:

```
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager$6.run(ZigBeeNetworkManager.java:1324)
```

the protocol handler should never be null. This call is in the middle of the `setNetworkStateOnline` method, so the `NetworkManager` has already got the NCP online, and is now starting everything.

I've not been able to replicate this issue here which I did by changing one of the XML files to remove the
 `<MacCapabilitiesType>RECEIVER_ON_WHEN_IDLE</MacCapabilitiesType>` line - this causes this code to be called, but the NCP is always `ONLINE` and it always works.

triller-telekom, 2020-07-20:
Hmm, since it only happened in my IDE and not with real devices, it might be related to that. Maybe I messed it up with some break points. Because I saw threads where it was `null` together with those where the protocol handler was correctly set.

So i guess we can close this issue for now.

And in case there is a timeout again before a correct reply is received we can open a new ticket, since then the extended timeout that we did here is not working.

closed 2020-07-20 by triller-telekom