zsmartsystems/com.zsmartsystems.zigbee #1082: console application exception with XBEE stick

sim732, 2020-07-17:
After upgrading to com.zsmartsystems.zigbee.console.main-1.3.6-jar-with-dependencies.jar
got:

ZigBee network state updated to INITIALISING
Exception in thread "main" com.thoughtworks.xstream.InitializationException: Could not instantiate mapper : com.thoughtworks.xstream.mapper.LambdaMapper
	at com.thoughtworks.xstream.XStream.buildMapperDynamically(XStream.java:645)
	at com.thoughtworks.xstream.XStream.buildMapper(XStream.java:622)
	at com.thoughtworks.xstream.XStream.<init>(XStream.java:586)
	at com.thoughtworks.xstream.XStream.<init>(XStream.java:514)
	at com.thoughtworks.xstream.XStream.<init>(XStream.java:483)
	at com.thoughtworks.xstream.XStream.<init>(XStream.java:429)
	at com.thoughtworks.xstream.XStream.<init>(XStream.java:396)
	at com.zsmartsystems.zigbee.console.main.ZigBeeDataStore.openStream(ZigBeeDataStore.java:65)
	at com.zsmartsystems.zigbee.console.main.ZigBeeDataStore.readNode(ZigBeeDataStore.java:118)
	at com.zsmartsystems.zigbee.database.ZigBeeNetworkDatabaseManager.startup(ZigBeeNetworkDatabaseManager.java:193)
	at com.zsmartsystems.zigbee.ZigBeeNetworkManager.initialize(ZigBeeNetworkManager.java:379)
	at com.zsmartsystems.zigbee.console.main.ZigBeeConsoleMain.main(ZigBeeConsoleMain.java:287)
Caused by: java.lang.ClassNotFoundException: com.thoughtworks.xstream.mapper.LambdaMapper
	at com.thoughtworks.xstream.core.util.CompositeClassLoader.loadClass(CompositeClassLoader.java:148)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:348)
	at com.thoughtworks.xstream.XStream.buildMapperDynamically(XStream.java:641)
	... 11 more

cdjackson, 2020-07-18:
I'm unable to produce this issue with an Ember dongle - it should not be dongle specific as shown in the stack trace. Please can you ensure that everything is up to date (ie all dependencies) and try to debug the problem.

cdjackson, 2020-07-18:
In the debugger this exception was being swallowed and not written to the console/log. It seems to be related to the version of XStream so I'll update that soon.

PR #1083 (2020-07-18): Update XStream dependency and improve logging
This updates the XStream version, and improves the logging of errors when reading and writing the XML database files in the console application.

Closes #1082
closed 2020-07-18 by cdjackson