zsmartsystems/com.zsmartsystems.zigbee #1087: ZclOtaUpgradeServer should not enforce querying the current file version after update

triller-telekom, 2020-07-20:
The `ZclOtaUpgradeServer` attempts to read the current file version from a device 10 times (in the logs there are even further attempts after the failure...):

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/6a2bf259f2062ca44c62eaec19166805512c0385/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/otaserver/ZclOtaUpgradeServer.java#L462-L478

if it cannot retrieve it, it marks the upgrade as failed.

According to the zigbee specifiactions the `CurrentFileVersion` (0x2) attribute from the `OTA Upgrade Cluster` (0x19) is **optional**, so the device must not support it.

Not having the option to always compare the firmware versions after an update makes me wonder whether there is another feature where one can determine that the upgrade was successful. I see the `DeviceAnnounce` in the logs, so the device has certainly rebooted. It also runs on the new firmware now.

Here is the log from an attempt for an update. it starts at the beginning of  `ZclOtaUpgradeServer#completeUpgrade()` where it queries the `ZclOtaUpgradeCluster.ATTR_IMAGEUPGRADESTATUS`:

```
2020-07-16T15:24:07.411+0200  101 DEBUG [TransactionManager-thread-3,com.zsmartsystems.zigbee.ZigBeeNetworkManager] TX CMD: ReadAttributesCommand [Ota Upgrade: 0000/0 -> 91EE/2, cluster=0019, TID=B7, identifiers=[6]]
2020-07-16T15:24:07.421+0200  101 DEBUG [TransactionManager-thread-3,com.zsmartsystems.zigbee.ZigBeeNetworkManager] TX APS: ZigBeeApsFrame [sourceAddress=0000/1, destinationAddress=91EE/2, profile=0104, cluster=0019, addressMode=DEVICE, radius=31, apsSecurity=false, ackRequest=true, apsCounter=EE, rssi=--, lqi=--, payload=08 B7 00 06 00]
2020-07-16T15:24:07.464+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=91EE, sourceEui=00124B000253FB62, lastHopLqi=255, lastHopRssi=-59, relayList=]
2020-07-16T15:24:07.477+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspMessageSentHandler [networkId=0, type=EMBER_OUTGOING_DIRECT, indexOrDestination=91EE, apsFrame=EmberApsFrame [profileId=0104, clusterId=0019, sourceEndpoint=1, destinationEndpoint=2, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=D8], messageTag=DB, status=EMBER_SUCCESS, messageContents=]
2020-07-16T15:24:07.523+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=91EE, sourceEui=00124B000253FB62, lastHopLqi=255, lastHopRssi=-63, relayList=]
2020-07-16T15:24:07.528+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspMessageSentHandler [networkId=0, type=EMBER_OUTGOING_DIRECT, indexOrDestination=91EE, apsFrame=EmberApsFrame [profileId=0104, clusterId=0019, sourceEndpoint=1, destinationEndpoint=2, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=D9], messageTag=B7, status=EMBER_SUCCESS, messageContents=]
2020-07-16T15:24:07.543+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=91EE, sourceEui=00124B000253FB62, lastHopLqi=255, lastHopRssi=-61, relayList=]
2020-07-16T15:24:07.550+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0019, sourceEndpoint=2, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=FC], lastHopLqi=255, lastHopRssi=-59, sender=91EE, bindingIndex=255, addressIndex=255, messageContents=18 B7 01 06 00 00 20 02]
2020-07-16T15:24:07.552+0200  101 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.ZigBeeNetworkManager] RX APS: ZigBeeApsFrame [sourceAddress=91EE/2, destinationAddress=0000/1, profile=0104, cluster=0019, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=FC, rssi=-59, lqi=FF, payload=18 B7 01 06 00 00 20 02]
2020-07-16T15:24:07.554+0200  101 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.ZigBeeNetworkManager] RX CMD: ReadAttributesResponse [Ota Upgrade: 91EE/2 -> 0000/1, cluster=0019, TID=B7, records=[ReadAttributeStatusRecord [status=SUCCESS, attributeIdentifier=6, attributeDataType=UNSIGNED_8_BIT_INTEGER, attributeValue=2]]]
2020-07-16T15:24:07.556+0200  101 DEBUG [NotificationService-thread-41,com.zsmartsystems.zigbee.zcl.ZclCluster] 91EE/2: Unknown Server attribute in 6 cluster 25
2020-07-16T15:24:07.556+0200  101 DEBUG [OtaCompleteUpgrade,com.zsmartsystems.zigbee.app.otaserver.ZclOtaUpgradeServer] 91EE/2 OTA status updated to OTA_UPGRADE_FIRMWARE_RESTARTING.
2020-07-16T15:24:07.559+0200  102 DEBUG [NotificationService-thread-54,org.openhab.binding.zigbee.handler.ZigBeeThingHandler] 00124B000253FB62: OTA transfer status update OTA_UPGRADE_FIRMWARE_RESTARTING, percent=100
2020-07-16T15:24:07.561+0200  101 DEBUG [OtaCompleteUpgrade,com.zsmartsystems.zigbee.ZigBeeNetworkManager] TX CMD: UpgradeEndResponse [Ota Upgrade: 0000/0 -> 91EE/2, cluster=0019, TID=B8, manufacturerCode=10132, imageType=1025, fileVersion=48, currentTime=0, upgradeTime=0]
2020-07-16T15:24:07.564+0200  101 DEBUG [OtaCompleteUpgrade,com.zsmartsystems.zigbee.ZigBeeNetworkManager] TX APS: ZigBeeApsFrame [sourceAddress=0000/1, destinationAddress=91EE/2, profile=0104, cluster=0019, addressMode=DEVICE, radius=31, apsSecurity=false, ackRequest=true, apsCounter=EF, rssi=--, lqi=--, payload=09 B8 07 94 27 01 04 30 00 00 00 00 00 00 00 00 00 00 00]
2020-07-16T15:24:07.609+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=91EE, sourceEui=00124B000253FB62, lastHopLqi=255, lastHopRssi=-61, relayList=]
2020-07-16T15:24:07.631+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspMessageSentHandler [networkId=0, type=EMBER_OUTGOING_DIRECT, indexOrDestination=91EE, apsFrame=EmberApsFrame [profileId=0104, clusterId=0019, sourceEndpoint=1, destinationEndpoint=2, options=[EMBER_APS_OPTION_ENABLE_ADDRESS_DISCOVERY, EMBER_APS_OPTION_RETRY, EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=DA], messageTag=B8, status=EMBER_SUCCESS, messageContents=]
2020-07-16T15:24:07.652+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspIncomingRouteRecordHandler [networkId=0, source=91EE, sourceEui=00124B000253FB62, lastHopLqi=255, lastHopRssi=-58, relayList=]
2020-07-16T15:24:07.659+0200  103 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.dongle.ember.ZigBeeDongleEzsp] RX EZSP: EzspIncomingMessageHandler [networkId=0, type=EMBER_INCOMING_UNICAST, apsFrame=EmberApsFrame [profileId=0104, clusterId=0019, sourceEndpoint=2, destinationEndpoint=1, options=[EMBER_APS_OPTION_ENABLE_ROUTE_DISCOVERY], groupId=0, sequence=FD], lastHopLqi=255, lastHopRssi=-59, sender=91EE, bindingIndex=255, addressIndex=255, messageContents=18 B8 0B 07 00]
2020-07-16T15:24:07.661+0200  101 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.ZigBeeNetworkManager] RX APS: ZigBeeApsFrame [sourceAddress=91EE/2, destinationAddress=0000/1, profile=0104, cluster=0019, addressMode=DEVICE, radius=0, apsSecurity=false, ackRequest=false, apsCounter=FD, rssi=-59, lqi=FF, payload=18 B8 0B 07 00]
2020-07-16T15:24:07.663+0200  101 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.ZigBeeNetworkManager] RX CMD: DefaultResponse [Ota Upgrade: 91EE/2 -> 0000/1, cluster=0019, TID=B8, commandIdentifier=7, statusCode=SUCCESS]
2020-07-16T15:24:17.668+0200  101 DEBUG [OtaCompleteUpgrade,com.zsmartsystems.zigbee.ZigBeeNetworkManager] TX CMD: ReadAttributesCommand [Ota Upgrade: 0000/0 -> 91EE/2, cluster=0019, TID=B9, identifiers=[2]]
2020-07-16T15:24:17.678+0200  101 DEBUG [OtaCompleteUpgrade,com.zsmartsystems.zigbee.ZigBeeNetworkManager] TX APS: ZigBeeApsFrame [sourceAddress=0000/1, destinationAddress=91EE/2, profile=0104, cluster=0019, addressMode=DEVICE, radius=31, apsSecurity=false, ackRequest=true, apsCounter=F0, rssi=--, lqi=--, payload=08 B9 00 02 00]
[... cut ...]
2020-07-16T15:28:14.682+0200  101 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.ZigBeeNetworkManager] 00124B000253FB62: Endpoint 4. Unknown remote endpoint
2020-07-16T15:28:14.683+0200  101 DEBUG [AshFrameHandler,com.zsmartsystems.zigbee.ZigBeeNetworkManager] Incoming message from node 91EE did not translate to command
2020-07-16T15:28:24.713+0200  102 DEBUG [NotificationService-thread-50,org.openhab.binding.zigbee.handler.ZigBeeThingHandler] 00124B000253FB62: OTA firmware request timeout (retry 2)
```

triller-telekom, 2020-07-20:
One additional "bug" that I just found in there is the reason for
```
00124B000253FB62: Endpoint 4. Unknown remote endpoint
```

Apparently AFTER the `DeviceAnnounce` the endpoint of the device has changed from 2 to 4 ! (maybe because of the new firmware). But the ZIgbeeNode does not reflect this.

cdjackson, 2020-07-20:
Thanks Stefan.

Can I suggest we address these two issues separately...

Let's address the querying of the attribute here - that should be simple. The change in endpoints is another question - that seems a bit more difficult as we probably need to effectively remove all current endpoints and rediscover the device. I suggest to open a new issue for this.

triller-telekom, 2020-07-20:
Agreed, I have opened https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/1088 for the endpoint change.

Regarding the unsupported attribute: Do you have an idea on what to ask the device in order to determine its working with the new firmware?

I mean if the attribute is not supported, I guess we could only try to send a command to see if it is back alive, but not compare the firmware versions...

cdjackson, 2020-07-20:
Ideally, we should still get a response even if it's unsupported, but the API won't return this directly through the attribute read API. It should be possible instead to use the `sendCommand` API as that ought to return the raw response rather than the processed attribute response - maybe in this case it's a good option?

In the event that the attribute is unsupported, you should expect a `DefaultResponse` with `UNSUPPORTED_ATTRIBUTE` status (or something - I'm going from memory). You could try this to see if it works and propose a PR?

triller-telekom, 2020-07-20:
>You could try this to see if it works and propose a PR?

The problem is that I cannot try this out unfortunately. It only happened in some cases fro some devices with some firmwares. And unfortunately our framework has tons of restrictions of firmware files, distributed over various systems, that its not as easy as "downgrading the a-forementioned plug and upgrading it again". So I fear I cannot directly test it...

However, the approach sounds plausible to me. So you suggest to use
```
Future<CommandResult> commandResult = cluster.sendCommand(new ReadAttributesCommand(identifiers));
CommandResult result = commandResult.get();

//pseudo code
if(result is ReadAttributeResponse) {
    //compare version numbers and if match -> update successful
}
else if ( result is DefaultResponse && unsupported attribute) {
{
   //firmware successful
}
else {
   //firmware update failed
}
```

to query the attribute and only take it as a failure if no response at all is received?

cdjackson, 2020-07-20:
Yep - that's exactly what I was thinking of :)

I think you can test this in two ways.

1) just testing that the normal response works - ie you get the `ReadAttributesResponse`
2) by changing the requested attribute from one that exists, to one that doesn't

triller-telekom, 2020-07-21:
One question about:
```
cluster.sendCommand(new ReadAttributesCommand(identifiers));
```

the `sendCommand` method for general commands from `ZclCluster` is protected and the `ZclOtaUpgradeCluster` is auto-generated any only offers a ` sendCommand(ZclOtaUpgradeCommand command)`.

Since you already said that we are somehow working around the general attributes API anyway, I am wondering how to solve this without offering an API that encourages people to work around the original API.

cdjackson, 2020-07-21:
Hmmm - I had in the back of my mind we might hit something like this...

I guess one option could be to add a method `ZclCluster.sendCommand(ZclGeneralCommand)` to allow sending all general commands via any cluster.

PR #1091 (2020-08-02): OTA Upgrade: Respect that CurrentFileVersion attribute is optional
Fixes #1087
commit dfd35fbc (2020-08-02): OTA Upgrade: Respect that CurrentFileVersion attribute is optional (#1091)

Fixes #1087
closed 2020-08-02 by cdjackson