zsmartsystems/com.zsmartsystems.zigbee #1088: ZigBeeNode endpoint change due to firmware update

triller-telekom, 2020-07-20:
As found in https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/1087#issuecomment-660905497 some devices might change their endpoints due to a firmware update.

Once the device announces itself via the `DeviceAnnounce` message, the corresponding `ZigBeeNode` has to be updated somehow.

cdjackson, 2020-07-20:
Throwing in a few thoughts here...

Ultimately we'd need to rediscover the node - this would need to remove all endpoints, and probably also the descriptors, from the `ZigBeeNode` and do a rediscovery. This would ensure that we account for any possible changes within the node (eg even changing it from a smoke sensor to a door sensor to a light bulb - if someone so desired).

I guess the big question is how to handle this - do we assume that since the firmware has changed, that the device could always have changed, and therefore after EVERY completed OTA that we should perform a rediscovery, or do we make this optional, or do we try and do something in-between. I'm tempted to say that we should always perform a rediscovery after every OTA completes - that ensures we know what the device has morphed into.

triller-telekom, 2020-07-21:
My first thought was that I could simply update the current node with "whatever comes in", but then I realized that we not not have much information about the device inside the `DeviceAnnounce`.

So yes, I agree, we would need to query everything from the device.

I haven't seen many zigbee devices that change during a firmware update, but I remember from a discussion in openHAB, that a firmware is basically what makes a devices. So if you change the firmware, you (potentially) get a different (type of) device, since features might have been added and/or removed.

Based on this, I would vote for re-discovering the node after each successful firmware update.