zsmartsystems/com.zsmartsystems.zigbee #1104: Mesh Discovery during pairing might prevent a successful pairing

triller-telekom, 2020-08-11:
I have a log from a pairing request for the device with IEEE `7CB03EAA00A3CED9` and current network address `482D`.

Apparently the device cannot be paired because the 

```
2020-08-10T12:02:21.207+0200   192  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 7CB03EAA00A3CED9: Node SVC Discovery: running ACTIVE_ENDPOINTS
```

task fails.

I see that `ActiveEndpointsResponse` is received just fine.

```
2020-08-10T12:02:21.843+0200   192  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 7CB03EAA00A3CED9: Node SVC Discovery: ActiveEndpointsResponse returned CommandResult [SUCCESS, ActiveEndpointsResponse [482D/0 -> 0000/0, cluster=8005, TID=8A, status=SUCCESS, nwkAddrOfInterest=482D, activeEpList=[3]]]
```

However, the `ACTIVE_ENDPOINTS` also requests the `SimpleDescriptors` and this is failing. Failing because it will be added to the transaction manager right after the log line above.

What I see is:

```
2020-08-10T12:02:33.845+0200   192  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 7CB03EAA00A3CED9: Node SVC Discovery: SimpleDescriptorResponse returned null
2020-08-10T12:02:33.845+0200   192  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 7CB03EAA00A3CED9: Node SVC Discovery: request ACTIVE_ENDPOINTS failed. Retry 1, wait 2146ms before retry.
2020-08-10T12:02:38.955+0200   192  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: SimpleDescriptorRequest [0000/0 -> 482D/0, cluster=0004, TID=99, nwkAddrOfInterest=482D, endpoint=3]
```

So the response returns `null`, whcih is due to the timeout of 12 seconds: 12:02.21 vs 12:02.33. Interestingly at 12:02:38.955 the request is send out to the device and it will be answered successfully:

```
2020-08-10T12:02:39.140+0200   192  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: SimpleDescriptorResponse [482D/0 -> 0000/0, cluster=8004, TID=99, status=SUCCESS, nwkAddrOfInterest=482D, length=26, simpleDescriptor=SimpleDescriptor [endpoint=3, profileId=C05E, deviceId=0010, deviceVersion=2, inputClusterList=[1000, 0000, 0003, 0004, 0005, 0006, 0B04, FC0F], outputClusterList=[0019]]]
```

Unfortunately, this is too late for the `ACTIVE_ENDPOINTS` and it will be treated as failed.

Why is this time out happening? According to the logs there are a lot of `ManagementLqiRequest`s running (and some are timing out) and I see a lot of `Node SVC Discovery: request NEIGHBORS failed` during the time where the `SimpleDescriptorRequest` is supposed to be send out.

I digged through the code to see where the `ManagementLqiRequest`s and `NEIGHBORS` tasks are coming from and found:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/e3a0844d9ccef30c5f284ca05586a7230c448646/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeDiscoveryExtension.java#L235-L240

So whenever a device joins the network, we do a full refresh of the mesh network information.

I think this is a bad idea to do, because on systems with many zigbee devices, this will lead to delays and possible failures like the one shown above. I am not even sure whether it's a good idea to scan once a device leaves the network. Because that case could also happen during paring if the device is reset and then we will end up with the same problem (our pairing requests get delayed due to the mesh update).

The system where it happend had "just" 29 zigbee devices paired, apparently some of them where also unreachable by the time the pairing attempt was made.

I have attached the full logs (grepped for "zsmartsystems" and "binding.zigbee").

Shall I create a PR to remove the `public void commandReceived(ZigBeeCommand command) {` method from `ZigBeeDiscoveryExtension`?

[bug-device-zigbee.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/5057286/bug-device-zigbee.txt)

triller-telekom, 2020-08-14:
@cdjackson Did you have a chance to have a look at this?

cdjackson, 2020-08-16:
> However, the ACTIVE_ENDPOINTS also requests the SimpleDescriptors and this is failing. Failing because it will be added to the transaction manager right after the log line above.

This should not be a problem. The transaction manager should queue this and send it when it can, so the fact that it is added to the transaction manager immediately at this point is in itself not a reason for any failure.

Just a quick look at the global view for reference...

[image]

I should say that there are really a lot of failed transactions in this log - something like 172 our of around 180 commands I think - so nearly everything is failing in the network. Each timeout delays things for a number of seconds and this is just choking the network.

[image]

I don't really see anything wrong here - ok, not "nothing" as such - but there isn't anything wrong with this request or the exact issue you raise...

What I think is happening is that there is a huge problem with this network - nearly every command is failing, and this is causing huge delays in the transaction manager. This is the reason that the `SimpleDescriptorRequest` / `SimpleDescriptorResponse` doesn't happen for such a long time after the command gets queued and when it is sent, it returns fine.

The only real problem I see here is with the transaction timeout. This timeout should not actually be in force, so there is a bug here I think. The issue is there is this default 12 second timeout in the `get` waiting for the future - this really should not be there IMHO. It causes the command to fail even though it is still waiting in the queue to be sent.

Increasing this timeout to a much longer time (in the minutes range) or solving the problem that is causing nearly every command to fail will resolve this problem I think.

cdjackson, 2020-08-16:
Thinking about this some more, this transaction timeout may be linked to #1054. I've seen in the past that the broadcast queue can fill up and this is probably due to this timer timing out, but the transactions staying in the queue and the higher layers are also adding another transaction to the queue. See also #844 which suggests we should remove transactions from the queue if they time out - that would also stop the command being sent after the application layer had already given up.

closed 2020-08-16 by cdjackson