zsmartsystems/com.zsmartsystems.zigbee #1109: Transactions should start their timeout timer only if they are send and not just enqueued

triller-telekom, 2020-08-17:
From the discussion of the closed ticket https://github.com/zsmartsystems/com.zsmartsystems.zigbee/issues/1104#issuecomment-674538471 I think that this is a bug too:

> The only real problem I see here is with the transaction timeout. This timeout should not actually be in force, so there is a bug here I think. The issue is there is this default 12 second timeout in the get waiting for the future - this really should not be there IMHO. It causes the command to fail even though it is still waiting in the queue to be sent.

I am just wondering where to start fixing this bug? I think the goal is to only start the timeout timer once we have tried to send out the transaction at least once. Before that, a timeout should not occur, since the transaction is still enqueued and will eventually be send out.

In `ZigbeeTransaction` I see 2 times outs already:

```
/**
     * The amount of time (in milliseconds) from when the command is sent to the transport, until when the transport
     * acknowledges it has been transmitted.
     * This must account for the total time from when the data is sent, until the response is expected, included any
     * queueing delay in the transport.
     */
    private final static int TRANSACTION_TIMER_1 = 10000;
    private int timeout1 = TRANSACTION_TIMER_1;
```

and

```
/**
     * The amount of time (in milliseconds) to wait for a response from the transport once the command has been
     * transmitted.
     */
    private final static int TRANSACTION_TIMER_2 = 8000;
    private int timeout2 = TRANSACTION_TIMER_2;
```

I think the first one is just what we need. However in #1104 the `ZigBeeTransation.getFuture().get()` returns too fast.

That's why I think the bug must be in here:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/878754782735f559d7d55a93127de21311e4a460/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/transaction/ZigBeeTransactionFuture.java#L74-L86

So I am wondering whether the timeout in `get` somehow has to consider the `TRANSACTION_TIMER_1` from `ZigbeeTransaction`.

Any suggestions?

cdjackson, 2020-08-17:
> Transactions should start their timeout timer only if they are send and not just enqueued

This is of course how it works now - except that the default time of 12 seconds prevents this from working. The transaction itself properly handles the transaction states and times - the problem is that when you use the default `get` method, it uses this 12 second time which means that the transaction timer now starts from the time it's queued.

As I think I suggested elsewhere - this timer should simply be made longer.

I think you are getting muddled here between two things.  The `ZigbeeTransaction`, which will properly handle the transaction, and the `Future` which is the time that the application waits for the future to complete - irrespective of the transaction.

So, I don't think there is a bug as such - the timer just needs to be increased to "a long time". Do you agree?

triller-telekom, 2020-08-18:
>I think you are getting muddled here between two things. The ZigbeeTransaction, which will properly handle the transaction,

Yes, right, that is `TRANSACTION_TIMER_1` in `ZigbeeTransaction`.

> and the Future which is the time that the application waits for the future to complete - irrespective of the transaction.

I got that, but I was wondering if we should just use a "long" timeout? What is "long" int hat case? 1 minute? 2 minutes?

Or we could let the timeout of the timeout for the future start only once the transactions is send. So instead of using a "long" timeout, we could also somehow add the queuing time to the existing time out. That is what I meant with "consider the `TRANSACTION_TIMER_1` from `ZigbeeTransaction`".

If you think the "long" timeout for the `get` is the better solution, we can certainly go that way. It is easier anyway :)

cdjackson, 2020-08-18:
> If you think the "long" timeout for the get is the better solution, we can certainly go that way.

Well, I think it's the only solution...

> Or we could let the timeout of the timeout for the future start only once the transactions is send. 

We can't do this - the timers are for different purposes and unrelated.

The application only wants / needs to know what is happening once the result of the transaction is known. This could account for some different issues - the queuing time, any retries, response time etc. The application doesn't need to know that the transaction is delayed for sending, or if it took 2 retries. If it's time critical that something happens in 30 seconds (for example) then the application should use the `future.get(30 seconds)` method. If it doesn't care about the time, it should use the `future.get()` method, and this should be set to "indefinite" response time (let's forget about what that value is right now).

Starting the 12 second timer when the transaction is sent doesn't solve the issue you saw - it will help of course, but it still doesn't allow the transaction manager to manage transactions as retries will then be ignored.

Consider the following diagram. The application needs to account for the various times in the queue, and transactions (I've only shown the ZSS stack times here - below this again is the NCP retries etc). The `future.get()` is at the application level, the transaction timer acts many times - ie once for each retry. The transaction manager already accounts for the timeout (I think this is 10 seconds).

[image]

So for me we should simply make this `.get` timeout "very long".

triller-telekom, 2020-08-18:
Thank you for the detailed graphic and explanation. I missed the "retry" part completely! That is why I thought it could help to some incorporate the queuing time in the future timeout...

Now, it makes perfect sense to me. If an application wants/expects a "quick" response, it can make use of the `future.get(something)` method and if it uses the `get()` method, I will set a default timeout of 5 minutes and we can monitor how this works.

I will provide a PR for this.

cdjackson, 2020-08-18:
Thanks Stefan.

For completeness, I will add to this diagram the alternative... The top diagram (and as presented above) is what happens when you call `Future.get()` and we just let the transaction manager "do its thing".

The bottom diagram I've just added is what should happen if you call `Future.get(timeout)`. Currently it's not the case and the `cancel` message is not sent to the transaction manager when the application gives up, so the transaction could continue to be retried (and of course this is what you saw in the first log where the descriptor request was sent a minute or so after the timeout). This is actually what #844 is about - I will reopen that and see if we can link this timeout (I'm not sure if it is possible).

[image]

PR #1110 (2020-08-18): Increase the default timeout for transaction futures
Fixes #1109
commit 266bc42d (2020-08-18): Increase the default timeout for transaction futures (#1110)

Fixes #1109
closed 2020-08-18 by cdjackson