zsmartsystems/com.zsmartsystems.zigbee #1114: ZigBeeNodeServiceDiscoverer network address request bugs

triller-telekom, 2020-08-19:
After https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1099 and https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1110 have been merged there are now some issues with the `ZigBeeNodeServiceDiscoverer`, especially within this method:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/5532d179e9c9f79e08fb1478d20b109a75d7415d/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeNodeServiceDiscoverer.java#L298-L322

There is at least one `return true` missing after

```
logger.debug("{}: Node SVC Discovery: NetworkAddressRequest confirmed by broadcast", node.getIeeeAddress()); 
```

However, I tried to pair a motion sensor device (Bitron Video 902010/22B) and it claims to not support a unicast networkaddress request:

```
2020-08-19_15:15:38:139	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: NetworkAddressResponse [D7B9/0 -> 0000/0, cluster=8000, TID=8C, status=NOT_SUPPORTED, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=NULL, startIndex=null, nwkAddrAssocDevList=[]]
```

This makes me wonder how a unicast can work at all, because why should I ask a device personally for its own network address? In that case I already know it, right?

However, since the unicast is failing, with https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1110 we wait for 5 minutes until we try via broadcast:

```
2020-08-19_15:15:37:366	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B001A015035: Node SVC Discovery: running NWK_ADDRESS
...
2020-08-19_15:20:37:379	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B001A015035: Node SVC Discovery: NetworkAddressRequest returned null
2020-08-19_15:20:37:380	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - Broadcast: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=null queueTime=0, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> FFFF/0, cluster=0000, TID=--, ieeeAddr=00124B001A015035, requestType=0, startIndex=0]]
2020-08-19_15:20:37:381	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - FFFF/0: Sending ZigBeeTransaction [ieeeAddress=null queueTime=1, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> FFFF/0, cluster=0000, TID=92, ieeeAddr=00124B001A015035, requestType=0, startIndex=0]]
2020-08-19_15:20:37:383	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0000/0 -> FFFF/0, cluster=0000, TID=92, ieeeAddr=00124B001A015035, requestType=0, startIndex=0]
2020-08-19_15:20:37:435	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: NetworkAddressRequest [0000/0 -> 0000/0, cluster=0000, TID=92, ieeeAddr=00124B001A015035, requestType=0, startIndex=0]
2020-08-19_15:20:37:436	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: NetworkAddressRequest [0000/0 -> 0000/0, cluster=0000, TID=92, ieeeAddr=00124B001A015035, requestType=0, startIndex=0] 
2020-08-19_15:20:37:468	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: NetworkAddressResponse [0000/0 -> 0000/0, cluster=8000, TID=92, status=SUCCESS, ieeeAddrRemoteDev=00124B001A015035, nwkAddrRemoteDev=D7B9, startIndex=null, nwkAddrAssocDevList=[]]
2020-08-19_15:20:37:474	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - notifyTransactionCommand: NetworkAddressResponse [0000/0 -> 0000/0, cluster=8000, TID=92, status=SUCCESS, ieeeAddrRemoteDev=00124B001A015035, nwkAddrRemoteDev=D7B9, startIndex=null, nwkAddrAssocDevList=[]] 
2020-08-19_15:20:37:484	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B001A015035: Node SVC Discovery: NetworkAddressRequest returned NetworkAddressResponse [0000/0 -> 0000/0, cluster=8000, TID=92, status=SUCCESS, ieeeAddrRemoteDev=00124B001A015035, nwkAddrRemoteDev=D7B9, startIndex=null, nwkAddrAssocDevList=[]]
2020-08-19_15:20:37:485	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B001A015035: Node SVC Discovery: NetworkAddressRequest confirmed by broadcast
2020-08-19_15:20:37:485	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B001A015035: Node SVC Discovery: NetworkAddressRequest failed after unicast and broadcast
```
(in the last line you notice the missing `return`).

cdjackson, 2020-08-19:
> why should I ask a device personally for its own network address? In that case I already know it, right?

Sure - you are confirming it. But that should not stop it being supported. This function is not just about getting the address, but also reading the associated devices. Now I guess this is a ZED, so it doesn't have associated devices.

Probably it also doesn't support the broadcast - it's not just a unicast issue here. The difference is that with a broadcast, the parent will respond. Maybe we need a different approach for SED.

> However, since the unicast is failing, with #1110 we wait for 5 minutes until we try via broadcast:

Ok, this is a problem with the transaction management then. If the transaction fails, the transaction should return and continue immediately. We need to understand why the transaction is not failing (or returning even) when it receives the response to this command.

Is this something you can look at?

> (in the last line you notice the missing return).

I'll fix this now.

triller-telekom, 2020-08-20:
I found the problem. it is here: https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c89a7ec90594f670603e36f0421fb33089cfc18e/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/transaction/ZigBeeTransaction.java#L363

This only matches the `ieeeAddrRemoteDev` attribute:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c89a7ec90594f670603e36f0421fb33089cfc18e/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zdo/command/NetworkAddressRequest.java#L175-L176

which in the `NOT_SUPPORTED` case is `null`:

```
2020-08-19_15:15:38:139	NA	DEBUG:	[symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: NetworkAddressResponse [D7B9/0 -> 0000/0, cluster=8000, TID=8C, status=NOT_SUPPORTED, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=NULL, startIndex=null, nwkAddrAssocDevList=[]]
```

Since that `NetworkAddressRequest.java` code is auto generated, I still have to find out where it is generated and then we have to fix it there.

triller-telekom, 2020-08-20:
Alright, the part in the code generator is here: https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c89a7ec90594f670603e36f0421fb33089cfc18e/com.zsmartsystems.zigbee.autocode/src/main/java/com/zsmartsystems/zigbee/autocode/ZigBeeZclCommandGenerator.java#L306-L316

However, the matchers are currently concatenated using "&&", so to also allow a "||" check, we would need to extend the matchers to have the and/or as an attribute somewhere.

then we could check if the status was "NOT_SUPPORTED, FAILED, etc" OR (in this case) the ieee address matches.

@cdjackson Do you have an idea for an elegant fix here?

triller-telekom, 2020-08-24:
Hi Chris,

I gave it a try with the matching transactions, and it doesn't look too bad. However I am running into the following 2 issues.

During startup where we request the IEEE address from the local node to look at the associated devices:

```
java.lang.ClassCastException: com.zsmartsystems.zigbee.zdo.command.IeeeAddressRequest cannot be cast to com.zsmartsystems.zigbee.zdo.command.IeeeAddressResponse
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer.getAssociatedNodes(ZigBeeNetworkDiscoverer.java:415)
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer.access$500(ZigBeeNetworkDiscoverer.java:48)
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNetworkDiscoverer$3.run(ZigBeeNetworkDiscoverer.java:346)
```

And during pairing:

```
2020-08-24T16:14:46.175+0200   432  ERROR          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B000EE99CDB: Node SVC Discovery: exception: 
java.lang.ClassCastException: com.zsmartsystems.zigbee.zdo.command.NetworkAddressRequest cannot be cast to com.zsmartsystems.zigbee.zdo.command.NetworkAddressResponse
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer.requestNetworkAddress(ZigBeeNodeServiceDiscoverer.java:331)
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer.requestNetworkAddress(ZigBeeNodeServiceDiscoverer.java:314)
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer.access$800(ZigBeeNodeServiceDiscoverer.java:75)
	at com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer$NodeServiceDiscoveryTask.run(ZigBeeNodeServiceDiscoverer.java:644)
```

Let's start with the first one, because the other is related, i.e. has a very similar reason:

We are sending the `IeeeAddressRequest` with `TID=02`:

```
2020-08-24T16:14:08.012+0200   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: IeeeAddressRequest [0000/0 -> 0000/0, cluster=0001, TID=02, nwkAddrOfInterest=0000, requestType=1, startIndex=0]
```

and the TransactionManager is now waiting for a message matching `TID=02` and the one which arrives first is:

```
2020-08-24T16:14:08.337+0200   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: IeeeAddressRequest [0000/0 -> 0000/0, cluster=0001, TID=02, nwkAddrOfInterest=0000, requestType=1, startIndex=0]
```

And of course, this cannot be casted to `NetworkAddressResponse`. Shortly afterwards the correct reply arrives:

```
2020-08-24T16:14:08.490+0200   432  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - RX CMD: IeeeAddressResponse [0000/0 -> 0000/0, cluster=8001, TID=02, status=SUCCESS, ieeeAddrRemoteDev=000D6F000F24146C, nwkAddrRemoteDev=0000, startIndex=null, nwkAddrAssocDevList=[]]
```

unfortunately, this is too late for the matching.

The same applies in a similar way to the `NetworkAddressRequest` vs `NetworkAddressResponse` problem. There we also receive another `NetworkAddressRequest` with our transaction ID and then the correct reply.

So far, I have only see this for `IeeeAddressRequest` and `NetworkAddressRequest`, for other request, like for example the neighbours task that queries the link quality etc. everything is working fine. Also previously paired devices (before my change in the transaction manager) are controllable and also report their states. So this is looking good at least.

---

I am now wondering if this is a bug in the `TransactionManager`, because it assigns the same `TID` again, or if the request that arrives is "somehow related" and thus receives the same `TID`, but should not be used for matching.

If this isn't a bug (assigning the same TID again), then I think we need to keep the `responseMatcher` (in addition to the TID check!) and generate for every `XXXRequest` command a method that checks whether the response is of type `XXXResponse`.

Any idea if this a bug or a feature of the TransactionManager? :-)

I have attached the full log for reference and cleaned it up from the start of the bundles until the pairing with only zss and the OH binding lines, so its just 1200 lines and easy to see where it breaks:

[zigbee-clean.txt](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/5118530/zigbee-clean.txt)

cdjackson, 2020-08-24:
Hi Stefan.

I think the issue you are hitting is caused by the NCP loopback. You can see below that we send TID2 request, and receive this again a little later -:

[image]

Have I interpreted this correctly? I don't see that the transaction manager is assigning the same TID again is it?

 know for broadcasts this can be disabled, but I'm less sure about unicasts.

triller-telekom, 2020-08-25:
For sure it can be a loopback of the NCP. So far I wasn't aware of such a loopback, but I have seen that we SOMETIMES receive our request that we have just sent out...

If this is a loopback, why do we only get it for SOME commands and not all?

But if we cannot disable it for all commands, what about my idea of checking the TID + checking whether the received message is a XXXreponse? So we would add the TID check to:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c89a7ec90594f670603e36f0421fb33089cfc18e/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/transaction/ZigBeeTransaction.java#L363

and keep this in the code generator:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/c89a7ec90594f670603e36f0421fb33089cfc18e/com.zsmartsystems.zigbee.autocode/src/main/java/com/zsmartsystems/zigbee/autocode/ZigBeeZclCommandGenerator.java#L299-L303

WDYT?

cdjackson, 2020-08-25:
> For sure it can be a loop back of the NCP. So far I wasn't aware of such a loop back, but I have seen that we SOMETIMES receive our request that we have just sent out...

I think is it clear that this is a loop back.  The NCP will loop back commands - for example all broadcast commands are always looped back (you can disable this from being passed to the application in the `EzspDongle`).

I believe you will also receive any commands sent to the local device - ie the coordinator - and that is why you are receiving this command. It was sent to address 0000, your application is on this address, so it receives the command.

I don't think we can assume that the framework will never receive its own transmissions.

I think the check of the expected response is probably a good one, so yes, I agree with what you've proposed :)

triller-telekom, 2020-08-25:
Thanks for the reply.

I have now provided the PR. It works really nice so far, i have to say. I was able to pair devices, control them and unpair them.

I used a couple of centralite smart plugs, bitron smart plugs, centralite keypad, bitron video door/window contact, bitron humidity and temp sensor, osram RGB bulbs.

Also I did not spot any hicups in the sniffer while doing all these experiments, so not doubling of requests/responses or so.

I even used the latest master that was containing the cancel transaction commit from you as well.

I would appreciate if you could give the PR a try, because we could get rid of quite some unnecessary stuff as it looks (At least that's what the removed vs added numbers saying :))

cdjackson, 2020-08-25:
Thanks Stefan - I was too quick on the PR asking if you'd tested it :)

I will try this with the 32 bulb network over the next few days and see how it looks.

triller-telekom, 2020-08-26:
One thing where I noticed a problem is this:

```
2020-08-26T11:06:41.145+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B000EE99CDB: Node SVC Discovery: running NWK_ADDRESS
2020-08-26T11:06:41.145+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - 00124B000EE99CDB: Creating new Transaction Queue
2020-08-26T11:06:41.145+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - 00124B000EE99CDB: Set profile to ZigBeeTransactionProfile [maxOutstandingTransactions=2, interTransactionDelay=50, maxRetries=2]
2020-08-26T11:06:41.145+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - 00124B000EE99CDB: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=00124B000EE99CDB queueTime=0, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> A847/0, cluster=0000, TID=--, ieeeAddr=00124B000EE99CDB, requestType=0, startIndex=0]]
2020-08-26T11:06:41.205+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - A847/0: Sending ZigBeeTransaction [ieeeAddress=00124B000EE99CDB queueTime=60, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> A847/0, cluster=0000, TID=86, ieeeAddr=00124B000EE99CDB, requestType=0, startIndex=0]]
2020-08-26T11:06:41.265+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0000/0 -> A847/0, cluster=0000, TID=86, ieeeAddr=00124B000EE99CDB, requestType=0, startIndex=0]
2020-08-26T11:06:42.505+0200   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:SR:01,00124B000EE99CDB,A847,E533
2020-08-26T11:06:42.535+0200   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisRouteRecordMessageEvent [hops=1, remoteAddress=00124B000EE99CDB, networkRoute=[43079, 58675]]
2020-08-26T11:06:42.655+0200   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis Data:SR:01,00124B000EE99CDB,A847,E533
2020-08-26T11:06:42.655+0200   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.internal.TelegesisFrameHandler] - RX Telegesis: TelegesisRouteRecordMessageEvent [hops=1, remoteAddress=00124B000EE99CDB, networkRoute=[43079, 58675]]
2020-08-26T11:06:42.715+0200   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisRouteRecordMessageEvent [hops=1, remoteAddress=00124B000EE99CDB, networkRoute=[43079, 58675]]
2020-08-26T11:06:42.715+0200   227  DEBUG          [symbolicName=com.zsmartsystems.zigbee.dongle.telegesis, className=com.zsmartsystems.zigbee.dongle.telegesis.ZigBeeDongleTelegesis] - Unhandled Telegesis Frame: TelegesisRouteRecordMessageEvent [hops=1, remoteAddress=00124B000EE99CDB, networkRoute=[43079, 58675]]
2020-08-26T11:06:43.105+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer] - 00124B000EE99CDB: Node SVC Discovery: NetworkAddressRequest returned NetworkAddressResponse [A847/0 -> 0000/0, cluster=8000, TID=86, status=NOT_SUPPORTED, ieeeAddrRemoteDev=null, nwkAddrRemoteDev=NULL, startIndex=null, nwkAddrAssocDevList=[]]
2020-08-26T11:06:43.105+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionQueue] - Broadcast: Added transaction to queue, len=1, transaction=ZigBeeTransaction [ieeeAddress=null queueTime=0, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> FFFF/0, cluster=0000, TID=--, ieeeAddr=00124B000EE99CDB, requestType=0, startIndex=0]]
2020-08-26T11:06:43.105+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - FFFF/0: Sending ZigBeeTransaction [ieeeAddress=null queueTime=0, state=WAITING, sendCnt=0, command=NetworkAddressRequest [0000/0 -> FFFF/0, cluster=0000, TID=88, ieeeAddr=00124B000EE99CDB, requestType=0, startIndex=0]]
2020-08-26T11:06:43.165+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.ZigBeeNetworkManager] - TX CMD: NetworkAddressRequest [0000/0 -> FFFF/0, cluster=0000, TID=88, ieeeAddr=00124B000EE99CDB, requestType=0, startIndex=0]
2020-08-26T11:06:43.165+0200   211  DEBUG          [symbolicName=com.zsmartsystems.zigbee, className=com.zsmartsystems.zigbee.transaction.ZigBeeTransactionManager] - Transaction complete: ZigBeeTransaction [ieeeAddress=00124B000EE99CDB queueTime=1950, state=COMPLETE, sendCnt=1, command=NetworkAddressRequest [0000/0 -> A847/0, cluster=0000, TID=86, i [... cut ...]
    public boolean isTransactionMatch(ZigBeeCommand request, ZigBeeCommand response) {
        return (response instanceof NetworkAddressResponse)
                && ((ZdoRequest) request).getDestinationAddress().equals(((NetworkAddressResponse) response).getSourceAddress());
    }
```

However, after some time the task magically succeeds and I get a discovery result.

But this needs some more investigation, that I can't do at the moment. I just wanted to leave this here in case you don't see any issues and want to merge this.

PR #1128 (2020-09-19): Match transactions by their TID and XXRequest to XXResponse
Fixes #1114
commit 24d9e1a6 (2020-09-19): Match transactions by their TID and XXRequest to XXResponse (#1128)

Match transactions by their TID and XXRequest to XXResponse

Fixes #1114
closed 2020-09-19 by cdjackson