zsmartsystems/com.zsmartsystems.zigbee #1130: com.zsmartsystems.zigbee.dongle.cc2531 for web service API

trunghtdev, 2020-08-28:
Hi, I am writing a web service for Zigbee Doggle CC5312 / EFR32, but I cannot open Port Serial , It notified with the message that is  "Port is not Opened"

And this is my code: 

`
package com.acexis.smh.zigbee;

import java.io.InputStream;
import java.io.OutputStream;
import javax.ws.rs.GET;
import javax.ws.rs.Path;
import javax.ws.rs.Produces;
import javax.ws.rs.core.Response;

import org.apache.commons.cli.ParseException;

import com.zsmartsystems.zigbee.ZigBeeStatus;
import com.zsmartsystems.zigbee.database.ZigBeeNetworkDataStore;
import com.zsmartsystems.zigbee.dongle.cc2531.ZigBeeDongleTiCc2531;
import com.zsmartsystems.zigbee.ZigBeeNetworkManager;
import com.zsmartsystems.zigbee.transport.TransportConfig;
import com.zsmartsystems.zigbee.transport.TransportConfigOption;
import com.zsmartsystems.zigbee.transport.ZigBeePort.FlowControl;
import com.zsmartsystems.zigbee.serial.ZigBeeSerialPort;
import com.zsmartsystems.zigbee.serialization.DefaultDeserializer;
import com.zsmartsystems.zigbee.serialization.DefaultSerializer;
import com.zsmartsystems.zigbee.console.main.ZigBeeDataStore;

@Path("/cc2531")

public class ZigbeeAPI {
	static private ZigBeeDongleTiCc2531 zigBeeDongleTiCc2531;
	static private ZigBeeSerialPort zigBeeSerialPort;
	static final String serialPortName = "ttyACM0";
	static final int serialBaud = 115200;

	public ZigbeeAPI() {

	}

	@GET
	@Path("")
	@Produces("text/html")
	public Response getStartingPage() {
		try {
			final TransportConfig transportOptions = new TransportConfig();

			ZigBeeSerialPort tempZigBeeSerialPort = new ZigBeeSerialPort(this.serialPortName, this.serialBaud,
					FlowControl.FLOWCONTROL_OUT_XONOFF);
			
			this.zigBeeDongleTiCc2531 = new ZigBeeDongleTiCc2531(tempZigBeeSerialPort);
			transportOptions.addOption(TransportConfigOption.RADIO_TX_POWER, 3);
			this.zigBeeDongleTiCc2531.startup(false);
			if (this.zigBeeDongleTiCc2531 != null) {
				ZigBeeNetworkManager networkManager = new ZigBeeNetworkManager(this.zigBeeDongleTiCc2531);

				ZigBeeNetworkDataStore dataStore = new ZigBeeDataStore("CC2531");

				networkManager.setNetworkDataStore(dataStore);
				networkManager.setSerializer(DefaultSerializer.class, DefaultDeserializer.class);
				this.zigBeeSerialPort = tempZigBeeSerialPort;
				ZigBeeStatus status = this.zigBeeDongleTiCc2531.initialize();

				String output = "<h1>Hello World!<h1>" + "<p>RESTful Service is running ... <br>Ping @ " + status.name()
						+ this.zigBeeDongleTiCc2531.getZigBeePanId() + " " + this.serialPortName + " " + this.serialBaud
						+ " " + "</p<br>";
				return Response.status(200).entity(output).build();
			} else {
				return Response.status(200).entity("Zigbee is null").build();
			}
		} catch (Error err) {
			System.err.println("Parsing command line failed.  Reason: " + err.getMessage());
			throw err;
		}
	}
}
`

trunghtdev, 2020-08-28:
@cdjackson please ! help me.

cdjackson, 2020-08-28:
I don't think there's a bug in the framework - I'd suggest that you use the console application to work out what is happening.

I'd also suggest to use the high level public methods that are no dongle specific so that it works with other dongles rather than the 2531 interface.

closed 2020-08-28 by cdjackson

trunghtdev, 2020-08-29:
Thank you @cdjackson 
Bug was resolved when I run my web service with sudo user <3