zsmartsystems/com.zsmartsystems.zigbee #1139: Exposing lqi and rssi for a received ZigBeeCommand

tejas-n, 2020-09-16:
I noticed that a feature for exposing rssi and lqi for a `ZigBeeNode` was released with https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1002
We had a use-case where we wanted to know rssi and lqi for a received command. Do you think it also makes sense to have those fields for `ZigBeeCommand`? 

**Describe the solution you'd like**
I'm assuming this would involve copying lqi and rssi from `ZigBeeApsFrame` to `ZigBeeCommand` in `receiveCommand(..)` of `ZigBeeNetworkManager`, right?

I can submit a PR if you think this is correct 😄

cdjackson, 2020-09-19:
I guess this is ok, although I would probably question why you can't use the current system - ie when a command notification is received in your application, you can simply call the `getLast` methods to get the last `Rssi` or `Lqi`. I guess there's the chance that another command comes in at the same time, but if they are received so close together, then they will almost certainly have the same RSSI/LQI, and if they don't, do to some sort of variable RF conditions, then this will obviously be unreliable anyway.

tejas-n, 2020-09-20:
@cdjackson I guess you are right and like you said, it's certainly possible to fetch that with the current implementation with a few extra steps. But I was just wondering why it wasn't included in `ZigBeeCommand` if the transport layer frame has it.

closed 2020-12-04 by stale[bot]