zsmartsystems/com.zsmartsystems.zigbee #1146: Static APS counter with multiple NCPs

mikomarrache, 2020-09-24:
Hi,

The network manager generates APS counter based on a static AtomicInteger instance.

```
    /**
     * {@link AtomicInteger} used to generate APS header counters
     */
    private final static AtomicInteger apsCounter = new AtomicInteger();
```

Does that behavior impact the ability to use the library with multiple NCPs? In our case, there would be one NetworkManager instance per NCP.

Thanks,
Mickael

cdjackson, 2020-09-24:
That's an interesting question - and timely as I'm working on a multinet NCP at the moment :)

Firstly, I think it will work fine - I don't think it matters that the counters are not contiguous - although I'm not 100% sure. 

Secondly though - it might not be a bad idea to fix this. I'm not sure why it needs to be static actually.

closed 2020-10-02 by cdjackson

sraillard, 2020-10-06:
Hi @cdjackson, what do you mean by `multinet`? To be able to use the Zigbee library with multiple coordinators?

cdjackson, 2020-10-08:
The multinet system allows a single NCP to participate in multiple networks - with certain limitations. See AN724 (from memory). Silabs docs are extremely poor though unfortunately so it’s been very low going with their support as support for this is limited.

sraillard, 2020-10-09:
Thank you for the clarification around the `multinet` term.
Exactly, this is AN724: https://www.silabs.com/documents/public/application-notes/an724-multi-network.pdf
There is an evolution where EmberZNet PRO 6.8 is adding the Multi-PAN option (two actives roles on the same channel).

cdjackson, 2020-10-09:
I’m aware of the different versions. I don’t believe it matters to the framework so long as you adhere to the constraints. It is very poorly documented and quite problematic to actually use for a number of reasons so I’m not sure if this will be published.

PR #1151 (2020-10-02): APS Counter does not need to be static
Closes #1146