zsmartsystems/com.zsmartsystems.zigbee #1157: Zigbee Console netstart command is broken

tunjid, 2020-11-08:
**Describe the bug**

The `netstart` console command always errors out.

Line 622 of the ZigeeNetworkManager contains

```java
        if (networkState != ZigBeeNetworkState.INITIALISING) {
            logger.error("ZigBeeNetworkManager startup: Can't be called when networkState={}", networkState);
            return ZigBeeStatus.INVALID_STATE;
        }
```
Since at the time the console is running the network is always online, netstart will always fail.

**To Reproduce**B
Attempt to use the `netstart` console command

**Expected behavior**
The `netstart` command is available.

**Additional context**
I can submit a PR for this, I believe a check for wether it is being reinitialized should suffice?

cdjackson, 2020-11-08:
There is a command called `netreinitialise` (or something close to this) that you must run first. This sets the states correctly.

tunjid, 2020-11-08:
I'm not sure what it is, but when I use the `reinitialize` command before the `netstart` command, ZigBee routers are never able to join the network. The same happens if I call `networkManager.initalize()` directly.

If  don't reinitialize the network and just call `netstart` directly having commented the above out,  routers can join the network.

Are there things that need to be reconfigured after a `reinitialize`? The source seems to just update a flag when `initialize`. is called.

cdjackson, 2020-11-08:
I'm not really sure what you are doing, so it's hard to know how to help.

Normally, you should not need to use the netstart command - just initialise the system through the command line args.  The netstart command was added for some specific use case when we were testing a system as a router, and needed to connect to different networks.

It sounds from your description that you are creating a network, so I would simply recommend to use the command line arguments to initialise the coordinator.

tunjid, 2020-11-08:
I was able to work around it it by invoking the methods on the ZigBeeTransport(dongle) directly. Pardon the Kotlin:

```kotlin
        networkManager.zigBeeTransport.apply {
            updateTransportConfig(TransportConfig().apply {
                addOption(TransportConfigOption.DEVICE_TYPE, DeviceType.COORDINATOR)
            })
            out.println("ZigBees status is: ${startup(true)}")
        }
```

Thanks so much for your help!

closed 2020-11-08 by tunjid

cdjackson, 2020-11-08:
Really this shouldn't be needed. If you have a suggestion for a change in the console, then please feel free to provide an update, but you should be able to initialise a network now through the console cli - this is how the system is designed to work and the netstart function is really only there for "special occasions".

tunjid, 2020-11-08:
I'm not sure if it's an Android peculiarity, but unless I have the network already `ONLINE`, and subsequently reinitialize the transport, I'm not quite able to get things connecting. My CLI args are the same in the readme:

`-dongle CC2531 -port /dev/tty.usbmodem144201 -baud 115200 -channel 11 -pan 0x2000 -epan 987654321 -nwkkey AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA -reset `

except the serial driver on Android is this: https://github.com/mik3y/usb-serial-for-android

I've been using an [Android based Home Automation hub](https://github.com/tunjid/Android-Home-Hub) using this library for the past 2 years, it's worked great, but for whatever reason I always had to invoke `netstart`. The issue is only coming up now since I bumped the version to where `reinitialize` is now needed. I've probably been using it wrong this whole time.

I'll be sure to update if I figure out what 'm doing wrong.

tunjid, 2021-03-23:
The issue was the join command with `enable` defaults to `255`:
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.console/src/main/java/com/zsmartsystems/zigbee/console/ZigBeeConsoleNetworkJoinCommand.java#L50

A value of 255 for `permitJoin` is not allowed as per https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/666a920bf606a24ad8befdc6a6bff9481c6bab82/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNetworkManager.java#L1418

cdjackson, 2021-03-23:
Thanks. If you can provide a PR, that would be great - maybe change to use 254?