zsmartsystems/com.zsmartsystems.zigbee #1158: error of function deserializeUInt32 from efr32 (efr32mg12p432f1024gl125)

trunghtdev, 2020-11-09:
**Describe the bug**
when I turn mode pairing of green power sensor - green peak, I has received error:
```
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at com.zsmartsystems.zigbee.dongle.ember.ezsp.EzspFrame.createHandler(EzspFrame.java:440)
	at com.zsmartsystems.zigbee.dongle.ember.internal.ash.AshFrameHandler$1.run(AshFrameHandler.java:204)
Caused by: java.lang.ArrayIndexOutOfBoundsException: 34
	at com.zsmartsystems.zigbee.dongle.ember.internal.serializer.EzspDeserializer.deserializeUInt32(EzspDeserializer.java:161)
	at com.zsmartsystems.zigbee.dongle.ember.ezsp.command.EzspGpepIncomingMessageHandler.<init>(EzspGpepIncomingMessageHandler.java:139)
```

data that my ember dongle received from green power sensor: 
```
 RX ASH frame: AshFrameData [frmNum=4, ackNum=0, reTx=false, data=98 90 FF 00 C5 7C D2 60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF E3 FF FF FF FF FF 01 CC]
```
data size is 34 byte, but the handler  - deserializeUInt32 accepted with 32 byte

Hoang Trong Trung

Thank you

Have a nice day !

cdjackson, 2020-11-09:
From a quick look, the definition of the frame appears consistent with the EZSP definition in UG100 (I found a naming error, but that won't impact the deserialisation as the data types look correct). 

It may well be that UG100 is incorrect (I know of at least 1 error that Silabs acknowledge as being incorrect in the docs). It might pay to check this against other documentation (eg header files in the SDK) to see if it's consistent.

It may also be that the deserialiser for some of the GP specific types is incorrectly reading the number of bytes - it would be good if you can debug this?

trunghtdev, 2020-11-10:
@cdjackson 

I have read UG100

This is not an error but because I use a zigbee device of qovor (green peak technologies). Programmers can develop on it. And it's not my device. It is possible that the code in the device has been changed

My hope is that there will be a module that reads chunks of data beyond 32 bytes

Thank you

Have a nice day!

trunghtdev, 2020-11-10:
@cdjackson 

This is a new idea , not a bug

Please close this bug

Thank you!

^^

cdjackson, 2020-11-10:
Sorry - I'm a bit confused... What is a "new idea"?

Here you raised a bug for an exception that seems to be caused by incorrectly processing a received frame - what do you mean when you say this exception is a "new idea"? The framework should process the frames received from the NCP without any exception.

Sorry - it's just a bit confusing what you mean. Or are you doing something outside of the standards?

trunghtdev, 2020-11-10:
Yes "outside of the standards" from some brand device sensor

Maybe, Device I have tested, It is example

It send one request with 34 byte data

cdjackson, 2020-11-10:
Ok, thanks. I'm still surprised that it has an error, but maybe there's a bug in the NCP code that is not correctly parsing the non-standard length frame and that's causing the EZSP frame to have an incorrect length...  If it works ok for standard length frames though, then let's leave it...

closed 2020-11-16 by cdjackson