zsmartsystems/com.zsmartsystems.zigbee #1182: Z3Light - EFR32 could not join into EFR32 dongle network: Z3LightSoc down when run "Update TC Link Key"

trunghtdev, 2020-12-10:
**Describe the bug**
`ZLL Commissioning Common: ZllCommStackStatus: status = 90, flags = 00
NWK Steering stack status 0x90
NWK Steering network joined.
Find and Bind Target: Start target: 0x00
Find and bind target start: 0x00
RX: ZDO, command 0x0013, status: 0x00
Device Announce: 0x5134
Update TC Link Key: Starting update trust center link key process: 0x00
RX: ZDO, command 0x8002, status: 0x00
RX: Node Desc Resp, Matches: 0x0000
Update TC Link Key: Error: 0x04
Partner: 00 00 00 00 00 00 00 00 
NWK Steering: Trust center link key update status: 0x04
NWK Steering Stop.  Cleaning up.
Join network complete: 0xAE
NWK Creator: Cannot start. State: 0x00
Form network start: 0x70
EMBER_NETWORK_DOWN
ZLL Commissioning Common: ZllCommStackStatus: status = 91, flags = 00
NWK Steering stack status 0x91`

**When I use Z3GatewayHost, It joined successfully**

`RX: ZDO, command 0x0013, status: 0x00
Device Announce: 0x6308
Update TC Link Key: Starting update trust center link key process: 0x00
RX: ZDO, command 0x8002, status: 0x00
RX: Node Desc Resp, Matches: 0x0000
Update TC Link Key: New key established: 0x03
Partner: 78 F0 D8 FE FF 57 0B 00 
NWK Steering: Trust center link key update status: 0x03
Update TC Link Key: New key established: 0x65
Partner: 78 F0 D8 FE FF 57 0B 00 
NWK Steering: Trust center link key update status: 0x65
pJoin for 180 sec: 0x00
NWK Steering: Broadcasting permit join: 0x00
NWK Steering Stop.  Cleaning up.
Join network complete: 0x00`

cdjackson, 2020-12-10:
I'm not sure what versions of anything you are using, or really what you're trying to do here. Probably you have something misconfigured for your system.  One thing you can try is to change the `EZSP_TC_KEY_REQUEST_POLICY` to use a new key.

cdjackson, 2020-12-10:
I'm not sure that this will work in the longer term as this will not allow devices to join the network. You might want to try `EZSP_ALLOW_TC_KEY_REQUEST_AND_GENERATE_NEW_KEY` to see if that helps.

trunghtdev, 2020-12-11:
It worked

Thank you

closed 2020-12-25 by cdjackson