zsmartsystems/com.zsmartsystems.zigbee #1183: [Z3Light With Color ClusterID: 0X0300]: not found command ID 0X0B00

trunghtdev, 2020-12-11:
**Describe the bug**
When I use Tuya App to connect Z3LightSoc and change color of Z3LightSoc, It worked but I could not change color Z3Light with Z3Light

**When Use Tuya App**

> T00000000:RX len 6, ep 01, clus 0x0008 (Level Control) FC 01 seq 00 cmd 04 payload[FE 02 00 ]
> 
> T00000000:RX len 7, ep 01, clus 0x0300 (Color Control) FC 01 seq 00 cmd 06 payload[2B FE 01 00 ]
> 
> T00000000:RX len 5, ep 01, clus 0x0300 (Color Control) FC 00 seq 18 cmd 0B payload[0A 00 ]

**When I send command:  [ClusterId: 0x0008] commandId = 0x0400, [ClusterId: 0x0300]: commandId = 0x0600**

> T00000000:RX len 6, ep 01, clus 0x0008 (Level Control) FC 01 seq 4C cmd 04 payload[0A 0A 00 ]
> 
> T00000000:RX len 7, ep 01, clus 0x0300 (Color Control) FC 01 seq 4D cmd 06 payload[00 01 0A 00 ]

**I can not find commandId 0X0B00 with clusterId 0x0300**

cdjackson, 2020-12-11:
I can't really work out what you are talking about here. Command IDs are only 8 bits long, so I can't understand when you talk about command ID 0x0600 or command ID 0xB00 etc.

If we are talking about command 0x0B (??) then this is not defined in the ZCL spec I have (which is not the latest, but is only about 12 months old).

[image]

It's also possible that devices use manufacturer specific commands that will not be in the cluster library (but again - they will be 8 bits long - not 16 as you say here).

Miguel077, 2020-12-11:
Found some information here: https://tasmota.github.io/docs/Zigbee/

cdjackson, 2020-12-11:
Is it part of the ZCL spec though? If so, please provide the ZCL reference - not Tasmota.

Miguel077, 2020-12-11:
Found a User Guide from NXP:
https://www.nxp.com/docs/en/user-guide/JN-UG-3115.pdf

cdjackson, 2020-12-11:
Please provide the information from the ZCL spec. If not possible, then please provide direct references - I'm unable to work out what you are referring to when you reference these links so please clearly provide the information from the spec for this attribute.

trunghtdev, 2020-12-14:
Here is log when I using Z3Gateway, I think I miss some thing
```
T00000000:RX len 6, ep 01, clus 0x0008 (Level Control) FC 00 seq 7F cmd 08 payload[00 00 00 ]

T00000000:RX len 5, ep 01, clus 0x0008 (Level Control) FC 00 seq 7F cmd 0B payload[09 00 ]
Z3LightSoc>
T00000000:RX len 5, ep 01, clus 0x0008 (Level Control) FC 00 seq 00 cmd 00 payload[00 00 ]
Z3LightSoc>
T00000000:RX len 5, ep 01, clus 0x0008 (Level Control) FC 00 seq 00 cmd 0B payload[01 00 ]
Z3LightSoc>
T00000000:RX len 6, ep 01, clus 0x0300 (Color Control) FC 00 seq 01 cmd 08 payload[00 00 00 ]
Find and Bind Initiator: Complete: 0x01

T00000000:RX len 5, ep 01, clus 0x0300 (Color Control) FC 00 seq 01 cmd 0B payload[09 00 ]

T00000000:RX len 5, ep 01, clus 0x0300 (Color Control) FC 00 seq 02 cmd 00 payload[00 00 ]

T00000000:RX len 5, ep 01, clus 0x0300 (Color Control) FC 00 seq 02 cmd 0B payload[01 00 ]

T00000000:RX len 6, ep 01, clus 0x0300 (Color Control) FC 00 seq 03 cmd 08 payload[00 00 00 ]

T00000000:RX len 5, ep 01, clus 0x0300 (Color Control) FC 00 seq 03 cmd 0B payload[09 00 ]

T00000000:RX len 5, ep 01, clus 0x0300 (Color Control) FC 00 seq 04 cmd 00 payload[00 00 ]

T00000000:RX len 5, ep 01, clus 0x0300 (Color Control) FC 00 seq 04 cmd 0B payload[01 00 ]

T00000000:RX len 5, ep 02, clus 0x0000 (Basic) FC 00 seq 05 cmd 00 payload[05 00 ]

T00000000:RX len 5, ep 02, clus 0x0000 (Basic) FC 00 seq 05 cmd 0B payload[01 00 ]

T00000000:RX len 5, ep 02, clus 0x0000 (Basic) FC 00 seq 06 cmd 00 payload[01 00 ]

T00000000:RX len 5, ep 02, clus 0x0000 (Basic) FC 00 seq 06 cmd 0B payload[01 00 ]
pJoin for 180 sec: 0x70
NWK Steering: Broadcasting permit join: 0x00
NWK Steering Stop.  Cleaning up.
Join network complete: 0x00

T00000000:RX len 6, ep 02, clus 0x0006 (On/off) FC 00 seq 07 cmd 08 payload[00 00 00 ]

T00000000:RX len 5, ep 02, clus 0x0006 (On/off) FC 00 seq 07 cmd 0B payload[09 00 ]

T00000000:RX len 5, ep 02, clus 0x0006 (On/off) FC 00 seq 08 cmd 00 payload[00 00 ]

T00000000:RX len 5, ep 02, clus 0x0006 (On/off) FC 00 seq 08 cmd 0B payload[01 00 ]

T00000000:RX len 6, ep 02, clus 0x0008 (Level Control) FC 00 seq 09 cmd 08 payload[00 00 00 ]

T00000000:RX len 5, ep 02, clus 0x0008 (Level Control) FC 00 seq 09 cmd 0B payload[09 00 ]

T00000000:RX len 5, ep 02, clus 0x0008 (Level Control) FC 00 seq 0A cmd 00 payload[00 00 ]

T00000000:RX len 5, ep 02, clus 0x0008 (Level Control) FC 00 seq 0A cmd 0B payload[01 00 ]

T00000000:RX len 6, ep 02, clus 0x0300 (Color Control) FC 00 seq 0B cmd 08 payload[00 00 00 ]

T00000000:RX len 5, ep 02, clus 0x0300 (Color Control) FC 00 seq 0B cmd 0B payload[09 00 ]

T00000000:RX len 5, ep 02, clus 0x0300 (Color Control) FC 00 seq 0C cmd 00 payload[00 00 ]

T00000000:RX len 5, ep 02, clus 0x0300 (Color Control) FC 00 seq 0C cmd 0B payload[01 00 ]

T00000000:RX len 6, ep 02, clus 0x0300 (Color Control) FC 00 seq 0D cmd 08 payload[00 00 00 ]

T00000000:RX len 5, ep 02, clus 0x0300 (Color Control) FC 00 seq 0D cmd 0B payload[09 00 ]

T00000000:RX len 5, ep 02, clus 0x0300 (Color Control) FC 00 seq 0E cmd 00 payload[00 00 ]

T00000000:RX len 5, ep 02, clus 0x0300 (Color Control) FC 00 seq 0E cmd 0B payload[01 00 ]

T00000000:RX len 4, ep F2, clus 0x0021 (Green Power) FC 19 seq 0F cmd 02 payload[00 ]

T00000000:RX len 4, ep F2, clus 0x0021 (Green Power) FC 19 seq 10 cmd 02 payload[05 ]

T00000000:RX len 5, ep 01, clus 0x0008 (Level Control) FC 00 seq 09 cmd 0B payload[0A 00 ]
```

trunghtdev, 2020-12-14:
> I can't really work out what you are talking about here. Command IDs are only 8 bits long, so I can't understand when you talk about command ID 0x0600 or command ID 0xB00 etc.
>
> If we are talking [...]

I am using Z3ColorLightSoc (Router) vs ncp-uart-hw (COODINATOR)

cdjackson, 2020-12-14:
> I am using Z3ColorLightSoc (Router) vs ncp-uart-hw (COODINATOR)

Maybe you need to discuss this on the Silabs forum - it doesn't look like you are using this library at all. You keep providing logs from Silabs, and it seems therefore that you're using the Silabs software?

Otherwise, please provide logs from this software. Also, as mentioned above, if we're talking about the attributes, I need to see that this attribute is defined in the ZCL - I do not add manufacturer defined attributes or commands in this library - you need to manage that yourself.

trunghtdev, 2020-12-14:
apparently, MoveToHueAndSaturationCommand and MoveToLevelWithOnOffCommand are related, but it not work in my case

cdjackson, 2020-12-14:
I'm sorry - I'm really not sure what the issue is that you're reporting. It looked from the initial report that the issue was with an unsupported attribute - even though you talked about commanding issues, the spec only supports commands that are 8 bits in length, so your initial report of command IDs of (for example) 0x0400 is not possible as that is 10 bits long (or 16 bits even).

There are links between the implementation of some commands - but again I'm just unsure what you're reporting here. I do not have any Tuya equipment on which to test, so my ability to understand the issue without a better explanation is limited.

I can look at this for you commercially if you prefer - then I can purchase the equipment to perform a test myself to replicate and resolve any potential issues.

trunghtdev, 2020-12-15:
I resolved, when I converte 
**hue value to range 0 -> 254**
**saturation value to range 0 -> 254**
**level value to range 0 -> 254**

trunghtdev, 2020-12-15:
Thank you for your support!

Have a nice day!

cdjackson, 2020-12-15:
Thanks. I'll close this then.

closed 2020-12-15 by cdjackson