zsmartsystems/com.zsmartsystems.zigbee #1184: ZigBeeDiscoveryExtension stops periodic discovery of routers

tejas-n, 2020-12-14:
**Describe the bug**
`ZigBeeDiscoveryExtension` stops the periodic discovery of routers.

**To Reproduce**
- Add a router to a ZigBee network
- Reboot the device/coordinator
- The `ZigBeeDiscoveryExtension` starts discovery
- When the `ZigBeeDiscoveryExtension` updates any information related to the router node, it stops the further periodic discovery if it.

**Relevant Logs**
```
2020-12-14 14:14:59.599 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeDiscoveryExtension  - DISCOVERY Extension: Starting mesh update
2020-12-14 14:14:59.611 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeDiscoveryExtension  - 00124B001C10409D: DISCOVERY Extension: Starting mesh update
2020-12-14 14:14:59.612 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: Update mesh
2020-12-14 14:14:59.613 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: starting new tasks [NEIGHBORS, NWK_ADDRESS, ROUTES]
2020-12-14 14:14:59.614 [NetworkManager-thread-2] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: scheduled [NWK_ADDRESS, NEIGHBORS, ROUTES]
2020-12-14 14:14:59.700 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: running NWK_ADDRESS
2020-12-14 14:14:59.778 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: NetworkAddressRequest returned NetworkAddressResponse [3CC5/0 -> 0000/0, cluster=8000, TID=2B, status=SUCCESS, ieeeAddrRemoteDev=00124B001C10409D, nwkAddrRemoteDev=3CC5, startIndex=null, nwkAddrAssocDevList=[]]
2020-12-14 14:14:59.779 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: NetworkAddressRequest confirmed by unicast
2020-12-14 14:14:59.780 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: request NWK_ADDRESS successful. Advancing to NEIGHBORS.
2020-12-14 14:14:59.782 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: running NEIGHBORS
2020-12-14 14:14:59.894 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: ManagementLqiRequest response CommandResult [SUCCESS, ManagementLqiResponse [3CC5/0 -> 0000/0, cluster=8031, TID=2C, status=SUCCESS, neighborTableEntries=6, startIndex=0, neighborTableList=[NeighborTable [extendedPanId=7CE6B13A0D1BD7DD, extendedAddress=000D6FFFFEE6C09F, networkAddress=0, deviceType=COORDINATOR, rxOnWhenIdle=UNKNOWN, relationship=PARENT, permitJoining=UNKNOWN, depth=0, lqi=83], NeighborTable [extendedPanId=7CE6B13A0D1BD7DD, extendedAddress=0015BC003900246E, networkAddress=25609, deviceType=END_DEVICE, rxOnWhenIdle=RX_OFF, relationship=CHILD, permitJoining=UNKNOWN, depth=2, lqi=170], NeighborTable [extendedPanId=7CE6B13A0D1BD7DD, extendedAddress=00137A0000016111, networkAddress=57074, deviceType=END_DEVICE, rxOnWhenIdle=RX_OFF, relationship=CHILD, permitJoining=UNKNOWN, depth=2, lqi=170]]]]
2020-12-14 14:14:59.993 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: ManagementLqiRequest response CommandResult [SUCCESS, ManagementLqiResponse [3CC5/0 -> 0000/0, cluster=8031, TID=2D, status=SUCCESS, neighborTableEntries=6, startIndex=3, neighborTableList=[NeighborTable [extendedPanId=7CE6B13A0D1BD7DD, extendedAddress=00137A000001D79A, networkAddress=54481, deviceType=END_DEVICE, rxOnWhenIdle=RX_OFF, relationship=CHILD, permitJoining=UNKNOWN, depth=2, lqi=170], NeighborTable [extendedPanId=7CE6B13A0D1BD7DD, extendedAddress=588E81FFFEBFD240, networkAddress=35579, deviceType=ROUTER, rxOnWhenIdle=RX_ON, relationship=CHILD, permitJoining=UNKNOWN, depth=2, lqi=117], NeighborTable [extendedPanId=7CE6B13A0D1BD7DD, extendedAddress=00124B001F68AA19, networkAddress=50289, deviceType=END_DEVICE, rxOnWhenIdle=RX_OFF, relationship=CHILD, permitJoining=UNKNOWN, depth=2, lqi=128]]]]
2020-12-14 14:14:59.995 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: ManagementLqiRequest complete [6 neighbors]
2020-12-14 14:15:00.006 [NetworkManager-thread-1] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: request NEIGHBORS successful. Advancing to ROUTES.
2020-12-14 14:15:00.007 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: running ROUTES
2020-12-14 14:15:00.079 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: ManagementRoutingRequest returned CommandResult [SUCCESS, ManagementRoutingResponse [3CC5/0 -> 0000/0, cluster=8032, TID=2E, status=SUCCESS, routingTableEntries=1, startIndex=0, routingTableList=[RoutingTable [destinationAddress=0, status=ACTIVE, memoryConstrained=false, manyToOne=true, routeRecordRequired=false, nextHopAddress=35579]]]]
2020-12-14 14:15:00.080 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: ManagementLqiRequest complete [1 routes]
2020-12-14 14:15:00.084 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: request ROUTES successful. Advancing to null.
2020-12-14 14:15:00.085 [NetworkManager-thread-6] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: complete
2020-12-14 14:15:00.093 [NotificationService-thread-33] DEBUG com.zsmartsystems.zigbee.app.discovery.ZigBeeNodeServiceDiscoverer  - 00124B001C10409D: Node SVC Discovery: stopped
```

**Additional context**
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/953772b97763c276964bfab584ef6905f29f3442/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeDiscoveryExtension.java#L235-L241
Whenever the `ZigBeeDiscoveryExtension` receives a `nodeUpdated(..)` for the router, it stops the discovery because the `nodeState` is `UNKNOWN`. This seems to be happening because we don't persist `nodeState` when using `ZigBeeNetworkDataStore`

What do you think the best solution might be here? Do you think we should update the `nodeState` to `ONLINE` when we receive *any* command(in addition to `DeviceAnnounce` and rejoin request) in the extension so that after the first discovery the `nodeState` is always updated to `ONLINE`?

cdjackson, 2020-12-14:
Let's address the easy question first :)

> Do you think we should update the nodeState to ONLINE when we receive any command(in addition to DeviceAnnounce and rejoin request)

Yes, I think we should.

Now to the more difficult question as I'm trying to understand the issue...

> When the ZigBeeDiscoveryExtension updates any information related to the router node, it stops the further periodic discovery if it.

If I look at the log, it seems that the discovery completed and everything worked fine? Is that correct? So your point is that the discovery has completed, but then doesn't do any further periodic rediscoveries? Or have I misunderstood?

If so, I think it's questionable if this is wise since it is quite a chatty thing to do (ie there's a lot going on - especially in large networks). There shouldn't really be a need for periodic rediscovery - any device that gets added to the network must be known to the coordinator (or more specifically, the TC, but that is nearly always the same as the coordinator).

Am I understanding this correctly or do I misunderstand something here?

tejas-n, 2020-12-14:
Thanks for the reply, Chris.

> If I look at the log, it seems that the discovery completed and everything worked fine? 

I assume when you use `ZigBeeDiscoveryExtension.setUpdatePeriod(..)`, the extension should be running the mesh update tasks periodically, is that correct? And it *is* doing that except for when you reboot the coordinator which causes the nodes' state set to `UNKNOWN` which makes the extension stop the periodic discovery. So am I correct in assuming that it's a bug?
<br>

> There shouldn't really be a need for periodic rediscovery

We did it for two reasons and maybe you could give some feedback about it:
1. We thought that it's necessary for the coordinator periodically run the `NodeDiscoveryTask.NEIGHBORS` and `NodeDiscoveryTask.ROUTES` on routers to keep itself updated. Do you think it's necessary to do that?
2. We have a use case were we want to check periodically if the router is alive on the network and report to our backend. We thought that since we are already using`ZigBeeDiscoveryExtension` we might as well use it to periodically elicit a response from the router. Do you think it's an intensive operation just to check if router is alive and receiving messages?

cdjackson, 2020-12-14:
>  I assume when you use ZigBeeDiscoveryExtension.setUpdatePeriod(..), the extension should be running the mesh update tasks periodically, is that correct?

Yes, that's correct...

> We thought that it's necessary for the coordinator periodically run the NodeDiscoveryTask.NEIGHBORS and NodeDiscoveryTask.ROUTES on routers to keep itself updated. Do you think it's necessary to do that?

No - this is not necessary.

> We have a use case were we want to check periodically if the router is alive on the network and report to our backend.
> Do you think it's an intensive operation just to check if router is alive and receiving messages?

Yes, I would suggest that you are using a sledgehammer to solve this...

---

Let's address a few issues... 

Firstly, the mesh update is only there to get mesh information so you can provide a chart (or whatever) of the mesh and signal strengths at that time between nodes. It doesn't have anything to do with routing etc. That is all managed directly in the coordinator.

Secondly, I'd suggest the following approach to deciding if a device is alive (it's what I use). Most routers do something - ie, they are a lighbulb, or a switch, etc. They aren't "just" an extender (although these exist, and maybe you are using them?). Therefore, the most efficient way to decide if a device is alive is to use reporting - you want to do this anyway (to know the state of the switch/light etc). If you configure reporting so that it reports every A minutes (let's say 10), and if you don't get a report for A minutes, times 2, plus 20 seconds, then you decide the device is dead (or, you then do a poll to confirm it's dead or alive).

The above formulae is what I use in the OH binding, and my recommendation for other commercial systems I work with. It allows for one lost report, plus a few seconds for timing mismatches etc, and if you implement a "last ditch poll" before marking it dead, then you have all cases covered.

Using the mesh network is a lot of additional traffic that is completely unnecessary - unless you're wanting to get an overview of the mesh and understand all the LQI values etc.

tejas-n, 2020-12-15:
> It doesn't have anything to do with routing etc. That is all managed directly in the coordinator.

Just out of curiosity, does that mean even if the library never sent out `ManagementLqiRequest` or `MangementRouteRequest` to gather information from the routers, the coordinator would still be able figure out how to send and route the packets? Is that handled on lower levels of the stack?

Using attribute reporting make sense and seems efficient. We use a few range extender as well so in that case, would you suggest sending any basic command like `NetworkAddressRequest` to see if it's alive?

cdjackson, 2020-12-15:
> Just out of curiosity, does that mean even if the library never sent out ManagementLqiRequest or MangementRouteRequest to gather information from the routers, the coordinator would still be able figure out how to send and route the packets? Is that handled on lower levels of the stack?

Yes, routing is handled by the coordinator - you don't need to send out these messages unless you are doing something with the data yourself?

> would you suggest sending any basic command like NetworkAddressRequest to see if it's alive?

If it really provides no reportable attributes, then you could use this, or simply poll an attribute from the basic cluster - I don't think it matters too much. If there's something reportable, then configuring the reporting would probably be best.

tejas-n, 2020-12-16:
Thanks for the answering the queries, looks like we won't be needing the periodic updates after all.

As for updating the `nodeState` to `ONLINE` when any command is received, would this be the right place to add it?
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/953772b97763c276964bfab584ef6905f29f3442/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/app/discovery/ZigBeeNetworkDiscoverer.java#L164-L175
Do you want me to go ahead and submit a PR?

cdjackson, 2020-12-16:
No problem.

Personally, I'd suggest that the `ZigBeeNetworkManager` is possibly the best place to put this. The reference you make is in the discovery code, which may not be used. I think it needs to be in a common place - possible somewhere as below -:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/1fa4dc10e6b9821358336c05ff90b02814ed10da/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/ZigBeeNetworkManager.java#L918

We just need to take care that we don't set it to `ONLINE` when the command is a device is leaving - I forget if that code is here, or possibly somewhere else. This method though receives all incoming commands, so somewhere here I think is probably a good spot if you want to come up with a PR (thanks).

tejas-n, 2020-12-17:
Okay, sounds good. I'll have a crack at it on the weekend :)

PR #1193 (2021-01-03): Mark a node as ONLINE when a command is received from it
Fixes #1184
closed 2021-01-03 by cdjackson