zsmartsystems/com.zsmartsystems.zigbee #1202: Snapshot artifacts

mikomarrache, 2021-01-27:
Hi,

I added the snapshot repository so that I can build against the master branch but I can't find there the last version (last version is 1.3.9-SNAPSHOT).

Is there another repository?

Thanks

cdjackson, 2021-01-27:
I think at the moment CI doesn't work, and in any case it doesn't deploy the snapshots. I've just built this manually so it should now be available, but ideally this would be automated through CI.

I'll try and take a look at this and will leave this open as a reminder (I'm moving house at the moment so have a lot on).

mikomarrache, 2021-01-27:
It looks like CI is working with Travis. But the Maven command is set as "mvn clean install".

I guess it should be changed to "mvn clean deploy", though it may require some configuration to deploy the artifacts.

cdjackson, 2021-01-27:
Yes, the deploy keys are not configured though - this needs to be set up when I get some time before it can deploy.

cdjackson, 2021-04-27:
Hopefully this is working now (although there might be an intermittent test failure!)

closed 2021-04-27 by cdjackson