zsmartsystems/com.zsmartsystems.zigbee #1207: Ember board configuration help

tunjid, 2021-02-13:
Hello, thanks again for such a great library.

I'm currently trying to move away from the Texas Instruments CC 2531 package to an Ember board as the Ember SDK is still being supported.

I've recently bought multiple Ember products off the Silicon labs website, and I'm having trouble communicating with them as easily as I did with the CC 2531.

I currently own the following EFR32 products:

1. EFR32MG Zigbee and Thread Starter Kit (SLWSTK6000B) for access to the Ember SDK
2. Thunderboard Sense 2 Sensor-to-Cloud Advanced IoT Kit (SLTB004A) for general use

I'm however however having issues with Simplicity Studio not being able to install any firmware for the former, and while I have success flashing the latter, I can't get the Java console app to communicate with it.

I understand this is open source software, and not an SiLabs/Ember support line. I however would appreciate if you could share the following if possible:

 1. The Ember hardware you use or testing if any
 2.  The name of the prepackaged project in simplicity studio that works with the hardware and this library
 3. The baud rate to use with the Java console to communicate with the hardware

Thanks for your help!

tunjid, 2021-02-27:
@cdjackson sorry for the ping, but any help on this is appreciated. 

I'm currently using the ncp-spi example with the `bootloader-spi-ezsp-combined.s37` bootloader under bootloader-spi-ezsp/efr32mg12p332f1024gl125-brd4162a which the starter kit board I have.

Is this the right sample and bootloader combination?

Thanks

cdjackson, 2021-02-27:
Normally you would not use an SPI interface, but it's hard to comment as I don't know what hardware you're using.

tunjid, 2021-02-27:
@cdjackson I've got the EFR32MG from this kit: https://www.silabs.com/development-tools/wireless/zigbee/efr32mg-zigbee-thread-starter-kit

cdjackson, 2021-02-27:
So you should not use the SPI - you should use the serial interface.

tunjid, 2021-02-27:
@cdjackson apologies if it's obvious, but of the following in the appended screenshot from simplicity studio, that would be which one?

Also the samples mention nothing about baud rate which is one of the CLI commands.

[image]

cdjackson, 2021-02-27:
You need to use a UART configuration as you are using a serial port (UART is a serial port).

Regarding baud rates - as it says in the readme -:

> one image supported hardware flow control with a baud rate of 115200 and the other image supported software flow control with a rate of 57600.

closed 2021-02-27 by tunjid