zsmartsystems/com.zsmartsystems.zigbee #1208: Code clarification on CC2531 ZdoXXXX classes

oblaise, 2021-02-24:
**Is your feature request related to a problem? Please describe.**
I would like to get an explanation for some classes in your code. It's neither a feature request or a bug report.

In the cc3531  ZDO handling code, the payload is constructed using half of the source address. The source address is located in `packet.getPacket()[4]` and `packet.getPacket()[5]` while the payload is constructed starting with byte 5 of the `packet.getPacket()`. As an example:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/666a920bf606a24ad8befdc6a6bff9481c6bab82/com.zsmartsystems.zigbee.dongle.cc2531/src/main/java/com/zsmartsystems/zigbee/dongle/cc2531/frame/ZdoNodeDescriptor.java#L21-L34

This is the same for the other sibling ZdoXXXXX classes. Why the payload of the `ZigBeeApsFrame` packet is constructed that way and not starting at byte 6 ?

Thank you very much for any clarification on this.

cdjackson, 2021-02-24:
It's been a long time since I looked at this. I have a feeling there is an extra byte in the packet that needed to be accounted for. I don't really use the 2531 as most users here (at least ones that I commercially support) are using the Silabs NCP.

oblaise, 2021-02-25:
When I looked into more detail, I see that the Payload is deserialized in the `com.zsmartsystems.zigbee.zdo.command.NodeDescriptorResponse` class that throw inheritance, uses the first byte as a transactionId : 
https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/666a920bf606a24ad8befdc6a6bff9481c6bab82/com.zsmartsystems.zigbee/src/main/java/com/zsmartsystems/zigbee/zdo/ZdoCommand.java#L20-L35

Is my understanding that the MSB of the address is used as transactionID in that case (because apparently the CC2531 doesn't return it through the API) seems OK ?

Thank you beforehand for your answer

cdjackson, 2021-02-25:
To be honest it has been a few years since I looked at the 2531 interface so I forget exactly what bytes are sent ;)

However the address and transaction ID must be different - the transaction ID can't be the MSB of the address or it would only allow 255 addresses/devices and this is not the case. Unfortunately I'm in the process of moving house so can't look at this in detail at the moment.

closed 2021-07-26 by stale[bot]