zsmartsystems/com.zsmartsystems.zigbee #1209: Publish artifacts to Maven Central

kaikreuzer, 2021-03-21:
In the light of Bintray+JCenter shutting down (see also https://github.com/openhab/openhab-distro/issues/1256), I'd like to ask whether it would be possible to publish the release artifacts to Maven Central - this would allow dependent projects to resolve them easily.

cdjackson, 2021-03-27:
The artefacts are already published to central so I think OH can already get them from there. Currently it is published to bintray first though, then promoted to central and I will take a look at publishing directly while I'm stuck in New Zealand quarantine :(

kaikreuzer, 2021-03-27:
Ah, excellent, I didn't notice, thanks!
You'll have to find a different way to publish it to Central in future, once Bintray closes our accounts in May...

Being stuck in NZ does not sound too back to me. Seems to be one of the few places on earth that is covid-free... Enjoy ;-)

closed 2021-03-27 by kaikreuzer

cdjackson, 2021-03-27:
I already have direct access to Sonatype but just never changed the Pom to use it. I updated this afternoon so it should be fine. 

Yes, I think NZ will be fine - just right now we are basically locked in a hotel for 13 days of quarantine...  but just 7 more days to go and life will be a bit more normal :)