zsmartsystems/com.zsmartsystems.zigbee #1234: About listeners execution order

mikomarrache, 2021-05-07:
Looking at multiple places in the library, the listeners are executer in separate threads. Therefore, there is no way to execute them in a particular order.

Is there a reason?

I understand the need to execute the listeners code in a separate thread to not block "library main threads", but why not to execute the listeners serially in one separate thread?

cdjackson, 2021-05-07:
Putting them in separate threads provides a level of isolation. If one listener takes a long time for some reason, it doesn't block other listeners. It is basically just to ensure that a rogue listener doesn't screw things up for others.

If they were all processed in a single thread, I'm still not sure that you can guarantee the order that they are called - you could ensure they are always called in the same order, and maybe this order could be the order the listeners were registered, but without a more complex system to (for example) set a priority, I'm not sure that order can really be achieved.

To reverse the question - why do you need to ensure the order?

mikomarrache, 2021-05-08:
> Putting them in separate threads provides a level of isolation. If one listener takes a long time for some reason, it doesn't block other listeners. It is basically just to ensure that a rogue listener doesn't screw things up for others.

I understand that. I think that should be done differently though. I think the responsibility of moving some processing to a different thread should be left to client code (i.e the actual listener implementation). I would agree that the library should only take care "library threads" are not stuck because client code and that can be done by running all listeners serially in one separate thread.

> If they were all processed in a single thread, I'm still not sure that you can guarantee the order that they are called - you could ensure they are always called in the same order, and maybe this [...]

ZigBeeNetworkNodeListener listeners are stored in a List - so they would be executed in the order they have been registered.
ZigBeeAnnounceListener listeners are stored in a Collection. I think it should also be declared as List to maintain the order.

> 
> To reverse the question - why do you need to ensure the order?

We would like to add extensions which will register listeners, and that would be a good thing to have control of which extension is notified first by controlling registration order.

cdjackson, 2021-05-08:
> I think the responsibility of moving some processing to a different thread should be left to client code

Hmm, well, I kind of disagree :). I wanted to ensure that the library code was not impacted by poorly written user code. This is why the library notifies the listeners in separate threads. This is similar to how I have seen other systems work and I don't want to have the library slowed down because the user code processing takes some time.

> ZigBeeNetworkNodeListener listeners are stored in a List - so they would be executed in the order they have been registered.

Sure - but as I think I mentioned earlier - this would ensure they are executed in the order registered, but that doesn't mean that it's the "right" order. Ie you still can't control the order they will be called if you can't control the order of registration and I'm not sure this can be guaranteed.

I'm still not really sure what problem you are trying to solve here. The library has been written to try and avoid any dependencies, so it should not matter which order the listeners are called.

> We would like to add extensions which will register listeners, and that would be a good thing to have control of which extension is notified first by controlling registration order.

I'd suggest to discuss this then. If this is your application code, then of course you also have the ability to register a single listener, and notify your code in whatever order you see fit.

closed 2021-07-23 by stale[bot]