zsmartsystems/com.zsmartsystems.zigbee #1239: EZSP Trust Center Link Keys Table setting in NCPs

MattWestb, 2021-05-15:
**Is your feature request related to a problem? Please describe.**
For the moment the SDK is limiting the TCLK table to 127 (both for EZSP 6.7.X and 6.9.X) and with high end MG1X devices and all MG2X is having enough Flash and Ram for extending it for our (the Zigbee community) use and the latest NCP with GP is using NVM3 so it shall not being any problems expanding the capacity.

ZHA is using hashed TCLK for going around this problems but i is not on great fan of that and like my ZB3 devices getting one new random TCLK for better security.

**Describe the solution you'd like**
I thinking its over 10 dvs and advanced user in the community that have reged WSTK and can putting support tickets in Sinabs support portal and i was thinking if you is also interesting trying getting Silabs putting the limit up to 256, 512 or 1024 tokens in the TCLKT ? 

**Describe alternatives you've considered**
I finding the 127 is being too low for on good NCP that all EM35X and EFR32MG i and like trying getting Solabs rising the limit.

I have using many IKEA ICC-A-1 modules that have one  MG1P and one 256 kB external flash (for OTA storage on real devices) but i think its not possible getting Silabs implanting storing NVM3 on the external flash for NCP use then the chips is  not longer active supported :-((

Is you interested of this approach ??

Thanks in advance.

Mattias 

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2021-07-14:
= no interest.

cdjackson, 2021-07-14:
= I'm not really sure what this has to do with the library. This is an issue for the Ember NCP.

MattWestb, 2021-07-14:
Its have not with the library direly only if you is interested getting Silabs putting the limit for the trust center link key higher then the 127 that is in the stack to day.

If like having high security its best force using updated link keys but its not possible today if having more then 127 devices in the system.

I can putting one support ticket but its useless if not more customers is asking for the same future.

cdjackson, 2021-07-14:
I think you need to raise this with Silabs - useless or not - it's the only way to get this included. I work a lot with Silabs and they will make their own decisions on the priorities, but if you don't raise the issue with them, it won't be included, and me also raising an issue will not change their mind.

MattWestb, 2021-07-14:
Thanks for constructive answer !

I was thinking Sirlabs seals force was only counting $ and if not being one large custom (like tuya or IKEA) thew was not paying any attention but if more small was asking for the same they can perhaps listen.

Thanks for your attention.

closed 2021-07-14 by MattWestb

cdjackson, 2021-07-14:
And do misunderstand - I agree that these sort of limitations should be removed as much as possible and if you raise a ticket with Silabs, I'm happy if you want to also point them at me as well.

I will canvas one of my larger users in Germany who I know are looking at larger networks (although it's generally only been around 75 lights I think).

MattWestb, 2021-07-14:
I finding it sad that EZSP have one so low "hard" limit and it is out of date with ZB3 and secure network. 
I wonder way not more company have complaining like tuya that is using EZSP 6.5.0 in the Zigbee GW (with EFR22MG1B) and new MG2x its no reason have it so low with NVM3 and plenty of flash.

I shall thinking little more and very like trying using little "seals force" ;-)