zsmartsystems/com.zsmartsystems.zigbee #1241: otaupgrade fails

deep-freeze, 2021-05-23:
**Describe the bug**
I tried to update the firmwares of Osram/Ledvance devices and was not successful except on the first try.

The first attempt was an update of a "PAR16 50 TW" device, which worked fine.

The "PAR16 50 TW" right next to it led to this output:
```
08:47:58.186 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA status updated to OTA_WAITING.
08:47:58.334 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA status updated to OTA_TRANSFER_IN_PROGRESS.
08:48:09.311 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Error: Invalid server state OTA_TRANSFER_IN_PROGRESS when handling QueryNextImageCommand.
08:48:18.327 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Data: Sending 60 bytes at offset 0
08:48:21.290 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Data: Sending 60 bytes at offset 0
...
08:50:33.312 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Data: Sending 60 bytes at offset 0
08:50:36.316 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Data: Sending 60 bytes at offset 0
08:50:38.706 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - F512/3: OTA Error: Timeout - aborting transfer.
08:50:38.711 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - F512/3 OTA status updated to OTA_UPGRADE_FAILED.
08:50:39.329 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Data: Sending 60 bytes at offset 0
08:50:42.327 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Data: Sending 60 bytes at offset 0
08:50:45.308 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Data: Sending 60 bytes at offset 0
08:50:54.875 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - 4B9A/3 OTA Error: Invalid server state OTA_TRANSFER_IN_PROGRESS when handling QueryNextImageCommand.
```
For a "PAR16 RGBW Z3" I just got this:
```
09:20:03.755 [DEBUG] [bee.app.otaserver.ZclOtaUpgradeServer] - CE1E/1 OTA status updated to OTA_WAITING.
```

**To Reproduce**
```
zigbee otaupgrade start 12345/3 firmware/ZLL_MK_0x01020510_PAR16_50_TW.ota
zigbee otaupgrade start 23456/1 firmware/PAR16_RGBW_Z3_IM0030_00103101-encrypted_11_27_2018_Tue_140612_79_withoutMF.ota
```

**Additional context**
OpenHab Version: 3.0.2-1
Zigbee Binding Version: 3.0.2
Coordinator:  Bitron Video AV2010/10 -  Ember NCP version 5.8.0.0, EZSP version 4

cdjackson, 2021-05-23:
Please provide a full logfile showing the transfer that succeeds and the one that fails. Also, please confirm that you are using the latest code.

deep-freeze, 2021-05-23:
I no longer have a log where the update process is successful (the offset was incremented there). I started investigating when the process failed on the second device. It is also not possible to get a new one from the device where it worked, because there is the same behavior as with the "PAR16 RGBW Z3". Nothing happens after `OTA status updated to OTA_WAITING.`.
The log for the failure case is complete, there is just an upgrade failure notification from another attempt (F512/3).

I am currently using version 1.3.8. but I can try the current master if you can show me how without messing up my setup. I was able to build the code and move some of the jar files to the addons folder, but openhab then shows both versions:
```
210 │ Active │  80 │ 1.3.8                   │ com.zsmartsystems.zigbee
211 │ Active │  80 │ 1.3.8                   │ com.zsmartsystems.zigbee.console
212 │ Active │  80 │ 1.3.8                   │ com.zsmartsystems.zigbee.console.ember
213 │ Active │  80 │ 1.3.8                   │ com.zsmartsystems.zigbee.console.telegesis
214 │ Active │  80 │ 1.3.8                   │ com.zsmartsystems.zigbee.dongle.cc2531
215 │ Active │  80 │ 1.3.8                   │ com.zsmartsystems.zigbee.dongle.ember
216 │ Active │  80 │ 1.3.8                   │ com.zsmartsystems.zigbee.dongle.telegesis
217 │ Active │  80 │ 1.3.8                   │ com.zsmartsystems.zigbee.dongle.xbee
241 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee
242 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee.cc2531
243 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee.console
244 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee.console.ember
245 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee.console.telegesis
246 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee.ember
247 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee.serial
248 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee.telegesis
249 │ Active │  80 │ 3.0.2                   │ org.openhab.binding.zigbee.xbee
265 │ Active │  80 │ 1.4.0.SNAPSHOT          │ com.zsmartsystems.zigbee.dongle.telegesis
266 │ Active │  80 │ 1.4.0.SNAPSHOT          │ com.zsmartsystems.zigbee.console.ember
267 │ Active │  80 │ 1.4.0.SNAPSHOT          │ com.zsmartsystems.zigbee.console
268 │ Active │  80 │ 1.4.0.SNAPSHOT          │ com.zsmartsystems.zigbee.dongle.cc2531
269 │ Active │  80 │ 1.4.0.SNAPSHOT          │ com.zsmartsystems.zigbee.dongle.ember
270 │ Active │  80 │ 1.4.0.SNAPSHOT          │ com.zsmartsystems.zigbee
271 │ Active │  80 │ 1.4.0.SNAPSHOT          │ com.zsmartsystems.zigbee.dongle.xbee
272 │ Active │  80 │ 1.4.0.SNAPSHOT          │ com.zsmartsystems.zigbee.console.telegesis
```

cdjackson, 2021-05-23:
I didn't know you were using openHAB. This is using an old library, but I'm really not sure what the issue is. It looked like the problem was with the server not completing an OTA. If so, a bug was recently fixed along these lines. However if you have reset the system and it still doesn't work, then that's strange.

It's not possible to use the latest libraries with openHAB at this time.

In any case, it would be good to see a log that shows the data frames - reduced logs are really difficult to work with.

deep-freeze, 2021-05-23:
Yeah, I have reset everything there is: LED, Zigbee coordinator and in the end the whole raspberry - all of them multiple times.

If that's the case, then I'll just wait for the openHAB folks to migrate to a newer version. The ota update is not critical for me.

If it still doesn't work then, how do I get a log that contains the needed information? The log level is set to debug, I have no idea how to get more logging output than this.

cdjackson, 2021-05-23:
Please read the OH binding documentation - this describes how to set the system to get the zigbee logs.

closed 2021-08-06 by stale[bot]