zsmartsystems/com.zsmartsystems.zigbee #1242: CC2531 always returns transaction ID 0

Ziver, 2021-06-06:
I'm using this project in a custom project with the CC2531 dongle, I'm able to pair devices but no descriptors or endpoints where not showing up including for the dongle itsself.

From my investigation it looks like the root cause of the issue is related to the pull request https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1128 where ZigBeeTransaction.java changed from:
`
if (responseMatcher.isTransactionMatch(command, receivedCommand)) {
`
To also verify the transaction ID:
`
if (Objects.equals(command.getTransactionId(), receivedCommand.getTransactionId())  && responseMatcher.isTransactionMatch(command, receivedCommand)) {
`
This change makes total sense but it looks like there is some issue with the CC2531 dongle where it always returns transaction ID as 0 for some responses which causes the transaction to not complete and instead timeout and fail.

**Some logs:**
Non working 1.3.11 version: [1.3.11_not_working.log](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/6603495/1.3.11_not_working.log)
Where the transaction fails (TID=04):

> 2021-06-06 03:41:05.059 [FINE]    ZigBeeNetworkManager        : RX CMD: NodeDescriptorResponse [0000/0 -> 0000/0, cluster=8002, TID=00, status=SUCCESS, nwkAddrOfInterest=0000, nodeDescriptor=NodeDescriptor [apsFlags=0, bufferSize=80, complexDescriptorAvailable=false, manufacturerCode=0000, logicalType=COORDINATOR, serverCapabilities=[PRIMARY_TRUST_CENTER], incomingTransferSize=160, outgoingTransferSize=160, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[ALTERNATIVE_PAN, FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=21]]
> 2021-06-06 03:41:05.060 [FINE]    ZigBeeTransactionManager    : notifyTransactionCommand: NodeDescriptorResponse [0000/0 -> 0000/0, cluster=8002, TID=00, status=SUCCESS, nwkAddrOfInterest=0000, nodeDescriptor=NodeDescriptor [apsFlags=0, bufferSize=80, complexDescriptorAvailable=false, manufacturerCode=0000, logicalType=COORDINATOR, serverCapabilities=[PRIMARY_TRUST_CENTER], incomingTransferSize=160, outgoingTransferSize=160, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[ALTERNATIVE_PAN, FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=21]] 
> 2021-06-06 03:41:13.054 [FINE]    ZigBeeTransactionManager    : Transaction complete: ZigBeeTransaction [ieeeAddress=00124B001CCE1B5F queueTime=16059, state=FAILED, sendCnt=2, command=NodeDescriptorRequest [0000/0 -> 0000/0, cluster=0002, TID=04, nwkAddrOfInterest=0000]]
> 2021-06-06 03:41:13.055 [FINE]    ZigBeeTransactionManager    : transactionListenerRemoved: 0 outstanding
> 2021-06-06 03:41:13.055 [FINE]    ZigBeeTransactionQueue      : 00124B001CCE1B5F: transactionComplete, state=FAILED, outstanding=0
> 2021-06-06 03:41:13.055 [FINE]    ZigBeeTransactionQueue      : 00124B001CCE1B5F: transactionComplete exceeded max retries 2
> 2021-06-06 03:41:13.056 [FINE]    ZigBeeTransaction           : Transaction cancelled: ZigBeeTransaction [ieeeAddress=00124B001CCE1B5F queueTime=16061, state=CANCELLED, sendCnt=2, command=NodeDescriptorRequest [0000/0 -> 0000/0, cluster=0002, TID=04, nwkAddrOfInterest=0000]]
> 2021-06-06 03:41:13.057 [FINE]    ZigBeeTransactionManager    : Transaction complete: ZigBeeTransaction [ieeeAddress=00124B001CCE1B5F queueTime=16061, state=CANCELLED, sendCnt=2, command=NodeDescriptorRequest [0000/0 -> 0000/0, cluster=0002, TID=04, nwkAddrOfInterest=0000]]
> 2021-06-06 03:41:13.057 [FINE]    ZigBeeNodeServiceDiscoverer : 00124B001CCE1B5F: Node SVC Discovery: NodeDescriptorResponse returned null
> 2021-06-06 03:41:13.057 [FINE]    ZigBeeTransactionManager    : transactionListenerRemoved: 0 outstanding
> 2021-06-06 03:41:13.057 [FINE]    ZigBeeTransactionQueue      : 00124B001CCE1B5F: transactionComplete but not outstanding, state=CANCELLED, outstanding=0
> 2021-06-06 03:41:13.057 [FINE]    NodeServiceDiscoveryTask    : 00124B001CCE1B5F: Node SVC Discovery: request NODE_DESCRIPTOR failed. Retry 2, wait 2301ms before retry.

 Working 1.3.8: [1.3.8_working.log](https://github.com/zsmartsystems/com.zsmartsystems.zigbee/files/6603496/1.3.8_working.log)

> 2021-06-06 03:42:56.923 [FINE]    ZigBeeNetworkManager        : RX CMD: NodeDescriptorResponse [0000/0 -> 0000/0, cluster=8002, TID=00, status=SUCCESS, nwkAddrOfInterest=0000, nodeDescriptor=NodeDescriptor [apsFlags=0, bufferSize=80, complexDescriptorAvailable=false, manufacturerCode=0000, logicalType=COORDINATOR, serverCapabilities=[PRIMARY_TRUST_CENTER], incomingTransferSize=160, outgoingTransferSize=160, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[ALTERNATIVE_PAN, FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=21]]
> 2021-06-06 03:42:56.924 [FINE]    ZigBeeTransactionManager    : notifyTransactionCommand: NodeDescriptorResponse [0000/0 -> 0000/0, cluster=8002, TID=00, status=SUCCESS, nwkAddrOfInterest=0000, nodeDescriptor=NodeDescriptor [apsFlags=0, bufferSize=80, complexDescriptorAvailable=false, manufacturerCode=0000, logicalType=COORDINATOR, serverCapabilities=[PRIMARY_TRUST_CENTER], incomingTransferSize=160, outgoingTransferSize=160, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[ALTERNATIVE_PAN, FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=21]] 
> 2021-06-06 03:42:56.925 [FINE]    ZigBeeTransactionManager    : Transaction complete: ZigBeeTransaction [ieeeAddress=00124B001CCE1B5F queueTime=33, state=COMPLETE, sendCnt=1, command=NodeDescriptorRequest [0000/0 -> 0000/0, cluster=0002, TID=03, nwkAddrOfInterest=0000]]
> 2021-06-06 03:42:56.925 [FINE]    ZigBeeNodeServiceDiscoverer : 00124B001CCE1B5F: Node SVC Discovery: NodeDescriptorResponse returned NodeDescriptorResponse [0000/0 -> 0000/0, cluster=8002, TID=00, status=SUCCESS, nwkAddrOfInterest=0000, nodeDescriptor=NodeDescriptor [apsFlags=0, bufferSize=80, complexDescriptorAvailable=false, manufacturerCode=0000, logicalType=COORDINATOR, serverCapabilities=[PRIMARY_TRUST_CENTER], incomingTransferSize=160, outgoingTransferSize=160, userDescriptorAvailable=false, frequencyBands=[FREQ_2400_MHZ], macCapabilities=[ALTERNATIVE_PAN, FULL_FUNCTION_DEVICE, MAINS_POWER, RECEIVER_ON_WHEN_IDLE], extendedEndpointListAvailable=false, extendedSimpleDescriptorListAvailable=false, stackCompliance=21]]
> 2021-06-06 03:42:56.925 [FINE]    ZigBeeTransactionManager    : transactionListenerRemoved: 0 outstanding
> 2021-06-06 03:42:56.925 [FINE]    NodeServiceDiscoveryTask    : 00124B001CCE1B5F: Node SVC Discovery: request NODE_DESCRIPTOR successful. Advancing to POWER_DESCRIPTOR.
> 2021-06-06 03:42:56.926 [FINE]    ZigBeeTransactionQueue      : 00124B001CCE1B5F: transactionComplete, state=COMPLETE, outstanding=0

What would be a good solution to solve this issue or maybe workaround it?

cdjackson, 2021-06-09:
I have some vague recollection of the 2531 driver always returning TID=0 in the past. I guess there are two options (in order of preference) -:

1. In the 2531 driver, try and restore the TID. I'm not sure if this is possible - I seem to recall that in the Telegesis driver there is also some trickery to do this, but I think it uses something else to perform the response match, and can then restore the TID. Maybe something like this is possible and if so it would probably be best.
2. If the 2531 always returns 0, then we could change this to return `null` and perform a check in the transaction manager to bypass the TID check.

Of course I'm also open to other ideas :). It's been a long time since I looked at the CC2531 code so I'm not very familiar with it these days.

Ziver, 2021-08-16:
Thanks for the suggestions, I agree that the your first suggestion would probably be the "cleanest" way to workaround this issue.

Will try to implement this and submit a PR.

einalex, 2021-08-22:
Thank you @Ziver and good luck :)
I have the same problem as you (while using openhab), and I suspect there is quite a number of other people suffering the same fate.

Best regards,
Alex

cdjackson, 2021-11-11:
Closing this by #1272 - thanks @Ziver

closed 2021-11-11 by cdjackson