zsmartsystems/com.zsmartsystems.zigbee #1252: IKEA Tradfri remote command error

dajudge, 2021-07-29:
I joined a IKEA TRADFRI remote (https://www.ikea.com/us/en/p/tradfri-remote-control-00443130/) via a ConBee USB Dongle. Commands from pressing buttons don't seem to be processed properly, though.

I also tried it w/ a CC2531 to basically the same results.

Now I'm well aware that the TRADFRI remote isn't the pinnacle of ZigBee compliance, but I didn't have any problems at least w/ the standard-compliant buttons on the remote w/ ZigPy.

Steps to reproduce the behavior:
* Join a IKEA TRADFRI remote
* Press the center button
* Observe error in log
* Observe no `commandReceived()` event on listener from `NetworkManager`

**Expected behavior**
* No error in log
* `commandReceived()` invocation on listener from `NetworkManager`

**Additional context**
Here's the (assumed) relevant part of the log:
```
08:31:10.488 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX: Data 17 32 00 21 00 1A 00 22 01 00 00 00 02 EF 70 01 04 01 06 00 03 00 01 2C 02 00 AF FF 44 D0 21 00 CF 08 FA
08:31:10.488 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX Frame: ReadReceivedDataResponse [sequence=50, status=SUCCESS, networkState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false], sourceAddress=NWK(28911), destinationAddress=GROUP(0), profileId=0104, clusterId=0006, lqi=255, rssi=207, data=01 2C 02]
08:31:10.489 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - RX APS: ZigBeeApsFrame [sourceAddress=70EF/1, destinationAddress=0000/0, profile=0104, cluster=0006, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=01 2C 02]
08:31:10.489 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - 588E81FFFE2314C4: Node update. NWK Address=70EF
08:31:10.489 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - 588E81FFFE2314C4: Node 70EF is not updated
08:31:10.489 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - Unknown local endpoint for APS frame ZigBeeApsFrame [sourceAddress=70EF/1, destinationAddress=0000/0, profile=0104, cluster=0006, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=01 2C 02]
08:31:10.490 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - Incoming message from node 70EF did not translate to command
08:31:10.490 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false]
08:31:10.490 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState-sync
08:31:10.490 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState-done
```
Here's the node's persisted data (I serialize to JSON):
```
{
    "ieee" : "c4:14:23:fe:ff:81:8e:58",
    "nodeDescriptor" : {
      "userDescriptorAvailable" : false,
      "incomingTransferSize" : 82,
      "manufacturerCode" : 4476,
      "complexDescriptorAvailable" : false,
      "logicalType" : "END_DEVICE",
      "macCapabilities" : [ "REDUCED_FUNCTION_DEVICE" ],
      "serverCapabilities" : [ ],
      "frequencyBands" : [ "FREQ_2400_MHZ" ],
      "apsFlags" : 0,
      "bufferSize" : 82
    },
    "bindingTable" : [ ],
    "endpoints" : [ {
      "endpointId" : 1,
      "profileId" : 260,
      "deviceId" : 2080,
      "deviceVersion" : 1,
      "inputClusterIds" : [ ],
      "outputClusterIds" : [ ],
      "inputClusters" : [ {
        "label" : null,
        "clusterId" : 0,
        "attributes" : {
          "0" : {
            "id" : 0,
            "name" : "ZCL Version",
            "dataType" : "UNSIGNED_8_BIT_INTEGER",
            "mandatory" : true,
            "implemented" : false,
            "readable" : true,
            "writable" : false,
            "reportable" : false,
            "minimumReportingPeriod" : 0,
            "maximumReportingPeriod" : 0,
            "reportingChange" : null,
[... 1933 lines cut ...]
08:31:10.485 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX: Data 0E 33 00 07 00 AA 00 0E FF
08:31:10.486 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX Frame: DeviceStateChanged [sequence=51, status=SUCCESS, state=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=true, dataRequest=true, configChanged=false]]
08:31:10.486 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX Frame but sentFrame is null: DeviceStateChanged [sequence=51, status=SUCCESS, state=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=true, dataRequest=true, configChanged=false]]
08:31:10.486 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=true, dataRequest=true, configChanged=false]
08:31:10.486 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState-sync
08:31:10.486 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState read
08:31:10.486 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE TX: ReadReceivedDataRequest [sequence=50]
08:31:10.486 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE TX: 17 32 00 08 00 01 00 00 AE FF
08:31:10.486 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState-done
08:31:10.488 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX: Data 17 32 00 21 00 1A 00 22 01 00 00 00 02 EF 70 01 04 01 06 00 03 00 01 2C 02 00 AF FF 44 D0 21 00 CF 08 FA
08:31:10.488 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX Frame: ReadReceivedDataResponse [sequence=50, status=SUCCESS, networkState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false], sourceAddress=NWK(28911), destinationAddress=GROUP(0), profileId=0104, clusterId=0006, lqi=255, rssi=207, data=01 2C 02]
08:31:10.489 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - RX APS: ZigBeeApsFrame [sourceAddress=70EF/1, destinationAddress=0000/0, profile=0104, cluster=0006, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=01 2C 02]
08:31:10.489 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - 588E81FFFE2314C4: Node update. NWK Address=70EF
08:31:10.489 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - 588E81FFFE2314C4: Node 70EF is not updated
08:31:10.489 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - Unknown local endpoint for APS frame ZigBeeApsFrame [sourceAddress=70EF/1, destinationAddress=0000/0, profile=0104, cluster=0006, addressMode=null, radius=0, apsSecurity=false, ackRequest=false, apsCounter=--, rssi=--, lqi=--, payload=01 2C 02]
08:31:10.490 [ConBeeReceiveHandler] DEBUG c.z.zigbee.ZigBeeNetworkManager - Incoming message from node 70EF did not translate to command
08:31:10.490 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState=DeviceState [networkState=NET_CONNECTED, dataConfirm=false, dataIndication=false, dataRequest=true, configChanged=false]
08:31:10.490 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState-sync
08:31:10.490 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - ConBeeDeviceState-done
08:31:11.508 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX: Data 1C 33 00 0C 00 05 00 02 EF 70 FF CF 71 FC
08:31:12.535 [ConBeeReceiveHandler] DEBUG c.z.z.d.c.i.ConBeeFrameHandler - CONBEE RX: Data 1C 34 00 0C 00 05 00 02 EF 70 FF CC 73 FC
08:31:13.491 [main] INFO  com.dajudge.jarvis.zigbee.Startup - Application exit
08:31:13.491 [main] DEBUG c.z.zigbee.ZigBeeNetworkManager - ZigBeeNetworkManager shutdown: networkState=ONLINE
08:31:13.491 [main] DEBUG c.z.z.d.ZigBeeNetworkDatabaseManager - Data store: Deferred Write Time set to 0ms
```

cdjackson, 2021-07-29:
I guess it's worth firstly pointing out that the ConBee is not currently supported.

> Expected behavior
>
> No error in log

I can't see any error at the moment (at least not from a search of the above)

> commandReceived() invocation on listener from NetworkManager

Please can you say what command you are expecting to see

dajudge, 2021-08-23:
Nevermind. Going for zigpy which works nicely.

Closing.

closed 2021-08-23 by dajudge