zsmartsystems/com.zsmartsystems.zigbee #1257: XBee driver not handling timeouts gracefully?

darrencocco, 2021-09-06:
Using Digi XU-Z11 with OpenHab one of the requests to `getZigBeeChannel` fails due to 500ms timeout which then throws a NullPointerException in `com.zsmartsystems.zigbbee.dongle.xbee.ZigBeeDongleXBee`

I believe this line is the cause of it: https://github.com/zsmartsystems/com.zsmartsystems.zigbee/blob/master/com.zsmartsystems.zigbee.dongle.xbee/src/main/java/com/zsmartsystems/zigbee/dongle/xbee/ZigBeeDongleXBee.java#L466

Because when the `frameHandler` fails it simply returns null which it is then trying call the method getChannel on.

Is throwing a runtime exception here expected behaviour?

darrencocco, 2021-09-07:
FYI if this is definitely not expected behaviour I can whip up a patch.

cdjackson, 2021-09-07:
Please feel free to propose a solution that you believe is better :). I've not analysed this myself yet so can't confirm if it should work this way, but it doesn't initially sound like the best option, so if you have a better solution, then submitting it would be appreciated.

cdjackson, 2021-09-11:
@darrencocco just an FYI - I'm planning to update the library version used in OH in the next week or so. If you are looking at this issue please let me know your timeframe so I can keep it in mind.

darrencocco, 2021-09-13:
@cdjackson I am working on this currently hopefully will have some progress in the next 2 days.

cdjackson, 2021-09-22:
Closed by #1263

closed 2021-09-22 by cdjackson