zsmartsystems/com.zsmartsystems.zigbee #1264: Question about supported Telegesis firmware versions

TomTravaglino, 2021-09-16:
In `ZigBeeDongleTelegesis.initialize()` there is this statement:

```
            if (!productInfo.getFirmwareRevision().startsWith("309")) {
                logger.error("Telegesis driver will not work with R{}, R309 is required",
                        productInfo.getFirmwareRevision());
                return ZigBeeStatus.FAILURE;
            }
```

Is there a reason to only support R309? I'm asking because there is also a R310 firmware available for the ETRX357. I have not tested it yet because I wanted to understand first if there is a reason not to install it.

cdjackson, 2021-09-16:
I didn’t know that there was an R310 - this must be new. When this was written, R309 was the most recent and R308 uses a different API - I think it’s similar, but there are some differences and it will not work. If there’s now an R310, and the API is the same, then this might work ok, and ou might be able to change the way this works so it works with 309 and 310.

Hedda, 2021-09-22:
FYI, there is an alternative EmberZNet 6.7.8 firmware with EZSP v8 (EmberZNet Serial Protocol version 8) interface available here:

https://github.com/walthowd/husbzb-firmware

There is some more information and discussion about that specific firmware image here -> https://github.com/walthowd/husbzb-firmware/issues/12

cdjackson, 2021-09-22:
The Telegesis dongle is not running standard Ember firmware - it is a completely different API.  I appreciate that it CAN be reprogrammed to do so, but that is not what this issue is about.

cdjackson, 2021-09-25:
@TomTravaglino I hope I answered your question? I will close this, but please let me know if you have any further questions on this. Bottom line is when we wrote this, it was written against the special DT firmware so one thing you should check is if the new version also introduces the extra commands that the DT version included for the network backup and restore.

closed 2021-09-25 by cdjackson