zsmartsystems/com.zsmartsystems.zigbee #1265: Add multicast table entry commands and data structure (Help support of IKEA TRÅDFRI remote contols and PIR sensor)

neo291, 2021-09-19:
**Is your feature request related to a problem? Please describe.**
Some IKEA TRÅDFRI devices like remote controls and PIR sensor is deisgned to use multicast groups in order to send messages to the related devices.
Multicast table commands is a Ember feature that is not implemented in this library making impossible to associate the coordinator to the correct group in the table and listen to the message send by that kind of devices.

**Describe the solution you'd like**
In order to manage it correctly and read the command sent by them without tedious workaround is neccessary to implement the multicast table entry support.
To be more specific is neccesary to implement the getMulticastTableEntry (0x63) and the setMulticastTableEntry (0x64) commands with the related data structure EmberMulticastTableEntry and EmberMulticastId.

**Describe alternatives you've considered**
There's no clean alternative solution to make the IKEA TRÅDFRI devices work in the correct way.
Some people use workaround like associate the remote/PIR with another device, for example a lamp, and read the lamp state to get the status of the command sent by remote/PIR, but for me is a dirty workaround that can be easily fixed.

**Additional context**
N/A

cdjackson, 2021-09-19:
From what I understand, the library ought to receive all messages - including multicast messages - without adding to the multicast table. I thought I'd tested this in the past, but possible not as mostly the multicast commands we use are sent out to other devices.

Have you actually tested this?

neo291, 2021-09-20:
Yes, I've tested and investigate on this.
I've an EMBER coordinator associated to some IKEA devices (remotes and PIR), then I set up another EMBER device to sniff the pakets.
In the end I've discovered that the packet is sent by the IKEA devices, but the EMBER coordinator discards it autonomously without sending nothing to the library.
This seems caused by the coordinator being not associated with the multicast group, then I've implemented the 2 multicast commands with their related data structure into the library and set up the multicast group for the coordinator during the start up, and it worked like a charm.
Also I've checked home assistant (that works perfectly with IKEA devices), being in python it use bellows as EMBER library and I see that it uses the setMulticastTableEntry during the start up to subscribe the coordinator to the neccessary group.

I've made a fork of the library where I've implemented this feature that you can find here: https://github.com/neo291/com.zsmartsystems.zigbee/tree/feat-multicast
I was near to make a PR, but I've seen that I need to accept the agreement for the contibution, but I'm not so convinced about some points that made me refuse to sign in (for example the point 9.3), so if you want you can integrate my work without any problem for me. But from what I understand I cannot make PR without acepting it, so I'm unable to send the PR directly.

cdjackson, 2021-09-20:
It is not the PR that is the issue - it is the work. You can create the PR, but I can't merge it. Likewise, I can't just take your work and add it to the framework if you do not agree to the CLA - sorry. The CLA is there to try and protect everyone so that this remains open source.

I don't really see the issue with 9.3. It simply states that in the event of your death, your heirs can't start selling this or revoke the CLA conditions that you signed up to.

PR #1276 (2021-11-12): Add Ember Multicast table management commands
This adds multicast table management features for Ember NCP and adds a `ncpmcast` console command to list the table, add and clear entries.

Note that the Ember console commands is now derived from `ZigBeeConsoleAbstractCommand` rather than just implementing `ZigBeeConsoleCommand` since the abstract class provides some additional features with parsing console arguments.

Closes #1265
closed 2021-11-12 by cdjackson