zsmartsystems/com.zsmartsystems.zigbee #1268: Is the Default Ember key exchange policy change a security issue?

triller-telekom, 2021-10-11:
Doing some security analysis of how we use ZigBee in our product made me stumble across a change of the default policy for network rejoins:

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/962/files

We had set this flag in the same way for our product due to some issues with I think Osram ZigBee 3 bulbs and I had left a comment on it:
```
// We have some ZigBee 3 bulbs that ask our coordinator which stackCompliance level it supports. Apparently our
// Ember dongles support ZigBee 3 features and thus the devices decide to request a trust center link
// key from them. Therefore we change the policy to allow sending out the link key to the device.
```

But now as I read the Silabs documentation about it I am worrying whether this setting is actually a good idea in the home automation world, were all (old) devices use the well-known Alliance09 link key...

The Silabs documentation https://www.silabs.com/documents/public/application-notes/an1233-zigbee-security.pdf on page 13 says:

> The recommendation to silently reject the Trust Center Rejoin as opposed to explicitly reject the Trust Center rejoin is necessary to maintain compatibility with pre-Zigbee 3.0 devices. Pre-Zigbee 3.0 devices may try Trust Center Rejoins without considering whether an updated link key has been obtained previously. If an explicit rejection is done, the device may drop off the network entirely while a silent rejection would allow the device to try a Secure Rejoin later and succeed.
>
> The Trust Center’s behavior will ultimately determine the security of the Trust Center Rejoin attempt. This can be controlled via the emberTrustCenterJoinHandler() callback on the SOC or via the ezspSetPolicy() API configuring the EZSP_TRUST_CENTER_POLICY for an NCP. To silently ignore the Trust Center Rejoin on an SOC device, the emberTrustCenterJoinHandler() must return EMBER_NO_ACTION. To silently ignore the Trust Center Rejoin on an NCP, the host must set the EZSP_TRUST_CENTER_POLICY to EZSP_IGNORE_TRUST_CENTER_REJOINS.

Before #962 the `EzspPolicyId.EZSP_TC_KEY_REQUEST_POLICY` was set to `EzspDecisionId.EZSP_DENY_TC_KEY_REQUESTS` which probably leads to the issue stated by the silabs documentation, i.e. `If an explicit rejection is done, the device may drop off the network entirely`.

Setting it to `EzspDecisionId.EZSP_ALLOW_TC_KEY_REQUESTS_AND_SEND_CURRENT_KEY` as the code is doing now means that we send the network key to any rejoining device if it presents us a valid link key (i.e. the Alliance09 one for example), right?

Shouldn't we instead set the value to `EzspPolicyId.EZSP_IGNORE_TRUST_CENTER_REJOINS` as stated in the documentation? As I understand from it, this would force the devices to try a secure join, i.e. they need to know the current network key before they are allowed to rejoin the network

cdjackson, 2021-10-12:
Hey Stefan,
So I think you are getting two things mixed up - the initial join, and rejoins.

For joins, I don't think the setting should matter for old devices. For old devices, the device will not request the link key, so the TC will not send it one - but it will still be happy. For newer devices, it MUST be set this way - otherwise the device will still request the updated key, and the TC will not send it, and the device will leave the network.

This issue is common - OH users running old firmware (eg the HUSBZB dongle comes with v5.8) cannot use new devices for this exact reason - the old firmware doesn't support the R21 join, so the devices leave the network.

The ZigbeeAlliance09 key isn't really relevant the the subsequent key exchange - this is only used for joining - after this, the network is secured by the network key. That said, of course it's a well-known key, and the best (ie most secure) way to have a device join is not to use this key, but to use the install code - then it's impossible for devices to join - but the TCLK is still updated after the join, so the above setting is still required.

There is another option - the more secure way is to use a setting that generates a new key for each device - the problem with this is that it fills up the key table, and for larger networks (where the word "large" might not be uncommon) this will fail. This is required for some devices (eg Smart Energy requires a new and authenticated key to be exchanged using the CBKE protocol).

Then there's the issue of rejoins, and yes, you are correct that ideally rejoins should not be enabled - requiring devices to use the network key that they (hopefully) already know. This is more secure, but may mean that devices drop off the network. This can happen if the network key was to change while they weren't listening (eg battery devices might be sleeping). 

So, the most secure option is to not use the ZB09 key, and use the install codes for joining. You still have to update the keys after joining, and then you should not accept rejoins. This should give you a bullet proof network from the security perspective, but of course you might have other issues like loosing devices :)

I think what I've said above is correct - but it's quite a complex area of the protocol, and one that is also often difficult to understand the settings (and I think Silabs also changed some of these policy options in the recent past, so there's a bodge in the code to use two different policy settings depending on the firmware (sigh!) ). As I say though - I THINK I'm correct - if you think I'm not - then - well, I might not be and I'm happy to discuss further :)

cdjackson, 2021-10-12:
> If EZSP_TC_REJOINS_USING_WELL_KNOWN_KEY_POLICY is not disabled, an attacker can fake a rejoin of a device and obtain the network key while asking for it using the well-known ZigbeeAlliance09 link-key.

I agree - this is the well known attack from a few years back that Zigbee 3 install codes resolved. As always, it's a trade of security and reliability though - especially for older devices.

> Nevertheless I think I would vote for option 1, because as far as I understand, option 2 might also prohibit zigbee 3 devices from rejoining the network, although they might have their own link-key due to an install code.

It's worth noting that install codes will not help with rejoins since the install code is only valid for the join - the NCP will remove the install code after a few minutes (I think you can set this time).

> I would therefore propose the following changes to the output of ncpsecuritystate:

Regarding the `ncpsecuritystate` cli, before moving forward, I'd like to understand what version of NCP code you are using, as things change with the definition of 'EZSP_TRUST_CENTER_POLICY` with EZSP8. After EZSP8 it is now a bitmap? Don't ask me why Silabs changed this - it's annoying! You will find that at this point I changed to printing the value rather than the enum as it is now a bitmap.

https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pull/1013

See also -:

https://community.silabs.com/s/question/0D51M00007xeRaTSAU/ember-67-trust-centre-decision-policy

I'm happy if you can propose something better - the enums are nicer, but we need to handle both versions - pre and post EZSP8 changes. At the time, changing to print the hex value seemed safest.

Regarding `EZSP_TC_REJOINS_USING_WELL_KNOWN_KEY_POLICY` I'm not against changing (or setting) this if you want to propose a PR - I might not merge it immediately as I need to ensure that it's possible to change this through a UI, and we should also think about how this is handled though `TransportConfigOption` and `TrustCentreJoinMode` which IIRC were defined initially for the pre-EZSP8 definitions. Maybe we just need a separate option here? I'm happy if you want to suggest something :)

PR #1269 (2021-11-12): Explicitly disallow rejoins using a well-known key
Fixes #1268
commit 1b5cf7c3 (2021-11-12): Explicitly disallow rejoins using a well-known key (#1269)

Fixes #1268
closed 2021-11-12 by cdjackson