zsmartsystems/com.zsmartsystems.zigbee #1278: Quirky Xiaomi Mijia button device stuck in detection (Unknown remote endpoint)

elfman03, 2021-11-15:
**Describe the bug**
Using Ember and the current 1.4.2 and earlier versions used by OpenHab2 and OpenHab3, the Xiaomi Mijia button (the round ones not the square Aquara).  It will get stuck staying offline failing to complete node detection.

**Expected behavior**
Button should detect as a switch.

**Additional context**
Enabling debug logging, whenever I press the button, I get this message logger.debug("{}: Endpoint {}. Unknown remote endpoint", 
It appears that it gets stuck in
ZigBeeNetworkManager.java -- receiveZclCommand()
        ZigBeeEndpoint endpoint = node.getEndpoint(apsFrame.getSourceEndpoint());
        if (endpoint == null) {
            logger.debug("{}: Endpoint {}. Unknown remote endpoint", node.getIeeeAddress(),
                    apsFrame.getSourceEndpoint());
            return null;
        }

The following block of code after where it gets stuck seems to show that receiveZclCommand() is resilient to new channels coming into endpoints and adds those channels dynamically.  

In my fork tried a similar trick in this offending block of code to dynamically add the endpoint to the node at this point and it seems to resolve my problem.  (The button now comes online and detects as a switch and button presses work).

ZigBeeEndpoint endpoint = node.getEndpoint(apsFrame.getSourceEndpoint());
        if (endpoint == null) {
	    // ELFY CUSTOMIZATION BEGIN
	    //
	    // Some devices (e.g., Xiaomi Meija buttons dont comply with detection but do come thru with data on an endpoint.
	    // Similar to how the below block of code will update channels dynamically when a new channel comes in on a known endpoint
	    // This block attempts to add a newly discovered endpoint to a known known.
	    // 
            logger.info("{}: Endpoint {}. Unknown remote endpoint.  Attempting to add...  ELFY...", node.getIeeeAddress(),
                    apsFrame.getSourceEndpoint());
            ZigBeeEndpoint newEndpoint = new ZigBeeEndpoint(node, apsFrame.getSourceEndpoint());
            node.addEndpoint(newEndpoint);
            refreshNode(node);
	    // ELFY CUSTOMIZATION END
	    endpoint=node.getEndpoint(apsFrame.getSourceEndpoint());
            if(endpoint==null) {
		    logger.debug("{}: Endpoint {}. Unknown remote endpoint", node.getIeeeAddress(),
                    apsFrame.getSourceEndpoint());
              return null;
	    }
        }

Is this something you would be interested in seeing cleaned a bit and submitted as a PR?

cdjackson, 2021-11-15:
If you are getting an unknown remote endpoint, then you need to add the endpoint to the node. The framework cannot process commands from endpoints that it doesn't know - this is against the standards and will cause a certification failure. So what you are proposing cannot be included here as it will fail certification - sorry.

Without knowing details of the device, I can't comment further, but you can simply add the endpoint/cluster and it should work fine.

elfman03, 2021-11-15:
Thanks,!  You guys know way way more than I ever will about the standard and what it requires.  Now that I know the endpoint and channel in question, I'll try playing with the jsondb a bit to add the endpoint/cluster with a stock build.  That would be preferable to maintaining a hacked branch.

cdjackson, 2021-11-15:
This code was specifically added to allow us to get a product through zigbee certification.

I’m not really sure what database you mean here, but again you shouldn’t have to manipulate any database. I maintain the openHAB binding and there is no need so far to have to manipulate any database so maybe you can look at openHAB as an example for your own product.

elfman03, 2022-04-16:
Resolved with change to the openhab binding instead.  Thanks!

closed 2022-04-16 by elfman03